evcc-io/evcc #5911: Security: Docker - run as nonroot / open ports / latest alpine

isarrider, 2023-01-27:
### Describe the bug

Hi,

regarding attack surface, I would propose 3 things:

docker: open only the ports that are needed
- run as non root (is there any need to do so...?)
- means change of docu and for those with KEBA or SMA Hardware an additional "-p ..." in the docker run cmd
- use the latest alpine (security fixes...)

I would prepare 3 PR if agreed...
2 & 3 run already fine on my SBC btw...

BR,
Alex

### Steps to reproduce

n/a

### Configuration details

```yaml
n/a
```

### Log details

```text
n/a
```

### What type of operating system are you running?

Docker container

### Version

latest

DerAndereAndi, 2023-01-27:
Some thoughts:

- There already is a PR for non root user in Docker: https://github.com/evcc-io/evcc/pull/4878
- Not using host mode makes the configuration very complicated for many users, and systems that e.g. rely on mDNS won't work. Like all EEBUS based wallboxes
- Having an advanced documentation for people who know what they are doing may make sense, but someone also has to maintain for which cases this will not work. Here is someone who made a documentation to make sure that host mode is used as people (himself included) had issues setting it up otherwise (as it won't work when using EEBUS in this case): https://github.com/evcc-io/evcc/discussions/5691
- What kind of attack surface do you see in the local network for not using host mode?

andig, 2023-01-27:
As Andi wrote:
- non-root has an open PR
- latest Alpine: we're using 3.16 which is still quite new plus under maintenance
- ports: imho the Doku is ok; if not I'd appreciate PRs for docs.evcc.io

That said: seems we have most pieces in place and should work on the open PR?

isarrider, 2023-01-27:
ok - all valid points...
Just wanted to help as I have a lil more time atm...

andig, 2023-01-27:
@isarrider please check the docs- I have some time ago simplified and corrected the Docker setup, please check what's missing.

isarrider, 2023-01-27:
@andig from what I can see this looks good
(considering that the dockerfile that we just discussed about already exposes the ports,
they would not need to be exposed again... ;)
This is what I meant btw - advanced users can the just leave the ports they dont use...

-> This is what the docu states:
sudo docker run \
-d \
--name evcc \
-v $(pwd)/evcc.yaml:/etc/evcc.yaml \
-v $(pwd)/.evcc:/root/.evcc \
-p 7070:7070 \
-p 8887:8887 \
-p 7090:7090/udp \
-p 9522:9522/udp \
evcc/evcc:latest

->This is what the Dockerfile contains:
# mDNS
EXPOSE 5353/udp
# UI and /api
EXPOSE 7070/tcp
# KEBA charger
EXPOSE 7090/udp
# OCPP charger
EXPOSE 8887/tcp
# SMA Energy Manager
EXPOSE 9522/udp

so in my POV the Dockerfile could loose the ports?

andig, 2023-01-27:
`EXPOSE` in the Dockerfile- afaik- really does *nothing*  except documentation. It doesn't open any ports.

isarrider, 2023-01-27:
https://docs.docker.com/engine/reference/builder/#expose
Yes, you are right... Somehow I remembered that it was then accessible (at least from inside the docker network)
But nice to have learned sth before the weekend... ;)

closed 2023-01-27 by andig