evcc-io/evcc #5913: manueller refresh des UI notwendig i.V.m. Zielladen

VolkerK62, 2023-01-27:
Mit ist folgendes aufgefallen (getestet mit Firefox und Chrome):

Ich stelle maxcurrent per Api auf 32A.
Klicke im UI auf Zielzeit. Angezeigt wird eine Dauer von 5:11 Std.
Ich aktiviere es nicht, sondern schließe das Fenster.
Dann ändere ich maxcurrent per api auf 6A.
Klicke im UI auf Zielzeit . Angezeigt wird eine Dauer von 5:11 Std.
Ich schließe das Fenster, drücke F5 (refresh) und klicke erneut auf Zielzeit.
Jetzt wird eine Dauer von 8:22 Std angezeigt.

Hier geht es nur um den notwendigen refresh.
Für die unrealistische Dauer bei ca. 10 kWh mache ich eine neue Diskussion auf.

_Originally posted by @VolkerK62 in https://github.com/evcc-io/evcc/discussions/5889_

naltatis, 2023-01-27:
Du bist im Nightly? Ich vermute das hat mir einem Fehler im Releaseprozess zu tun. Siehe #5908

VolkerK62, 2023-01-27:
Das war aber auch schon so, vor dem Nightly, dass die Einsparung weggenommen hat.

andig, 2023-01-27:
@naltatis ich glaub das ist anders. Der aktuelle Plan wird anscheinend beim Laden der UI abgerufen und nicht beim laden des Dialoges? Wenn sich zwischendurch die Ströme ändern fehlt der Roundtrip zum Backend. Zumindest sieht mir das in der Browserkonsole danach aus.

naltatis, 2023-01-29:
Hab das Ladeverhalten nun ein bisschen cleverer gemacht. Daten werden beim Öffnen des Dialogs und nicht mehr beim Öffnen der Seite aktualisiert.

PR #5931 (2023-01-29): Improve plan updating in UI
fixes #5913

- update plan data every time the dialog is opened
- don't fetch target charge plan on page load
closed 2023-01-29 by andig