evcc-io/evcc #5915: EVCC startet nicht wenn Charger nicht verbunden ist in Version 0.111.1

Steven-2023, 2023-01-27:
### Describe the bug

Wenn in der EVCC Konfiguration ein Charger eingetragen ist, der zum Zeitpunk des Starts von EVCC nicht erreichbar ist, startet die Anwendung nicht.

Dies betrifft EVCC 0.111.1. (Docker evcc/evcc:0.111.1)
Mit Version 0.110.1 startet die Anwendung. (Docker evcc/evcc:0.110.1)

### Steps to reproduce

1. use same configration in docker-compose
2. start container with tag 0.111.1 or 0.110.1
3. power charger on or off
...

### Configuration details

```yaml
chargers:
  - type: template
    template: pmcc 
    ski: 'xxxxxxxxxxx'  
    ip: 192.168.xx.yyyy  
    name: wallbox5
```

### Log details

```text
******* Start Up Sequenz EVCC 0.110.1 wenn Charger offline ist
evcc    | [main  ] INFO 2023/01/27 13:52:43 evcc 0.110.1
evcc    | [main  ] INFO 2023/01/27 13:52:43 using config file: /etc/evcc.yaml
evcc    | [main  ] INFO 2023/01/27 13:52:43 starting ui and api at :7070
evcc    | [db    ] INFO 2023/01/27 13:52:43 using sqlite database: /root/.evcc/evcc.db
evcc    | [mqtt  ] INFO 2023/01/27 13:52:43 connecting evcc at tcp://192.168.xx.xx:1883
evcc    | [eebus ] INFO 2023/01/27 13:52:43 Local SKI:  xxxxxxxxxxxxxxxxxxxx
evcc    | [site  ] INFO 2023/01/27 13:52:47 site config:
evcc    | [site  ] INFO 2023/01/27 13:52:47   meters:      grid ✓ pv ✓ battery ✓
evcc    | [site  ] INFO 2023/01/27 13:52:47     grid:      power ✓ energy ✗ currents ✓
evcc    | [site  ] INFO 2023/01/27 13:52:47     pv 1:      power ✓ energy ✗ currents ✗
evcc    | [site  ] INFO 2023/01/27 13:52:47     battery 1: power ✓ energy ✗ currents ✗ soc ✓
evcc    | [site  ] INFO 2023/01/27 13:52:47   vehicles:
evcc    | [site  ] INFO 2023/01/27 13:52:47     vehicle 1: range ✓ finish ✓ status ✓ climate ✓ wakeup ✗
evcc    | [lp-1  ] INFO 2023/01/27 13:52:47 loadpoint 1:
evcc    | [lp-1  ] INFO 2023/01/27 13:52:47   mode:        off
evcc    | [lp-1  ] INFO 2023/01/27 13:52:47   charger:     power ✓ energy ✗ currents ✓ phases ✗ wakeup ✗
evcc    | [lp-1  ] INFO 2023/01/27 13:52:47   meters:      charge ✓
evcc    | [lp-1  ] INFO 2023/01/27 13:52:47     charge:    power ✓ energy ✗ currents ✓
evcc    | [lp-1  ] INFO 2023/01/27 13:52:47 vehicle updated: unknown -> MyCar
evcc    | [lp-1  ] ERROR 2023/01/27 13:52:47 charger: YYYYYY-YYYYY-YYYYYY charger reported as disconnected

########################
******* Start Up Sequenz EVCC 0.111.1 wenn Charger offline ist

evcc    | [main  ] INFO 2023/01/27 13:48:12 evcc 0.111.1
evcc    | [main  ] INFO 2023/01/27 13:48:12 using config file: /etc/evcc.yaml
evcc    | [main  ] INFO 2023/01/27 13:48:12 starting ui and api at :7070
evcc    | [db    ] INFO 2023/01/27 13:48:12 using sqlite database: /root/.evcc/evcc.db
evcc    | [mqtt  ] INFO 2023/01/27 13:48:12 connecting evcc at tcp://192.168.xx.xx:1883
evcc    | [eebus ] INFO 2023/01/27 13:48:12 Local SKI:  xxxxxxxxxxxxxxxx
evcc    | [main  ] FATAL 2023/01/27 13:48:22 cannot create charger 'wallbox5': cannot create charger 'template': cannot create charger 'eebus': i/o timeout
evcc    | [main  ] FATAL 2023/01/27 13:48:22 will attempt restart in: 5m0s
evcc    | [main  ] FATAL 2023/01/27 13:49:07 evcc was stopped. OS should restart the service. Or restart manually.

########################
******* Start Up Sequenz EVCC 0.111.1 wenn Charger online ist
evcc    | [main  ] INFO 2023/01/27 15:11:10 evcc 0.111.1
evcc    | [main  ] INFO 2023/01/27 15:11:10 using config file: /etc/evcc.yaml
evcc    | [main  ] INFO 2023/01/27 15:11:10 starting ui and api at :7070
evcc    | [db    ] INFO 2023/01/27 15:11:10 using sqlite database: /root/.evcc/evcc.db
evcc    | [mqtt  ] INFO 2023/01/27 15:11:10 connecting evcc at tcp://192.168.xx.xx:1883
evcc    | [eebus ] INFO 2023/01/27 15:11:10 Local SKI:  xxxxxxxxxxxxxxxxxxxxxx
evcc    | [site  ] INFO 2023/01/27 15:11:28 site config:
evcc    | [site  ] INFO 2023/01/27 15:11:28   meters:      grid ✓ pv ✓ battery ✓
evcc    | [site  ] INFO 2023/01/27 15:11:28     grid:      power ✓ energy ✗ currents ✓
evcc    | [site  ] INFO 2023/01/27 15:11:28     pv 1:      power ✓ energy ✗ currents ✗
evcc    | [site  ] INFO 2023/01/27 15:11:28     battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✗
evcc    | [site  ] INFO 2023/01/27 15:11:28   vehicles:
evcc    | [site  ] INFO 2023/01/27 15:11:28     vehicle 1: range ✓ finish ✓ status ✓ climate ✓ wakeup ✗
evcc    | [lp-1  ] INFO 2023/01/27 15:11:28 loadpoint 1:
evcc    | [lp-1  ] INFO 2023/01/27 15:11:28   mode:        off
evcc    | [lp-1  ] INFO 2023/01/27 15:11:28   charger:     power ✓ energy ✗ currents ✓ phases ✗ wakeup ✗
evcc    | [lp-1  ] INFO 2023/01/27 15:11:28   meters:      charge ✓
evcc    | [lp-1  ] INFO 2023/01/27 15:11:28     charge:    power ✓ energy ✗ currents ✓
```

### What type of operating system are you running?

Docker container

### Version

0.111.1

DerAndereAndi, 2023-01-27:
This is wanted. In a future release the timeout for EEBUS wallboxes will be increased. evcc doesn't start if any (!) configured device can not be reached.

The Elli wallbox has some issues, as the EEBUS service is often crashing, also on startup, and then needs some to restart. The increased timeout will help with this, but the proper fix is for Elli to fix the software in their wallbox.

For now you either need to restart or wait until the service restarts itself.

closed 2023-01-27 by andig

nase30, 2023-03-03:
I have the same Problem. It worked and since yesterday, I do not get any connection.
Is it possible to prove it somehow, that Elle has an issue? I'd like to get in Touch with them, but I have no idea how an EEBUS works and what is wrong...
Can somebody describe what the problem with EEBUS is?
Regards
nase30