evcc-io/evcc #5918: Anzeige Kapazität Batterie

StevieC121176, 2023-01-27:
Wäre es möglich (evtl. auch als Einstellung wie km/miles) anstelle bzw gleichzeitig auch die aktuelle Batteriekapazität anzuzeigen?
(12.8 kWh/100%)` capacity` kann ja in der Config angegeben und daraus der aktuelle Wert errechnet werden.

[image]

andig, 2023-01-27:
Zu welchem Zweck? Du kennst die doch?

StevieC121176, 2023-01-27:
Ich meine nicht die Kapazität bei 100%, die ist mir bekannt (12,8 kWh), sondern die errechnete aus den aktuellen %. z.Bsp. bei 68% Akku sind noch 8,7 kWh im Akku
Mein Vater z.Bsp. findet diesen Wert aussagekräftiger als die % Anzeige.

andig, 2023-01-27:
Verstehe, das ist sicher sehr individuell. Ich wüsste allerdings auch nicht, was ich als Anwender mit dem Wert anfangen soll. Ist 1,8kWh gut? Bei 10% weiss ich sofort, dass sie fast leer ist. Ich glaube hier brauchts weiteres Feedback ob/dass wir das ändern wollen.

StevieC121176, 2023-01-27:
wählbar wäre vielleicht eine Alternative

andig, 2023-01-27:
Wenn wir eins nicht wollen, dann weitere Konfigurationsoptionen...

StevieC121176, 2023-01-30:
ein weiterer Vorschlag wäre, wie beim Anzeigen des Fahrzeuglimits, beim drüber bewegen mit der Maus so eine Sprechblase mit dem Wert anzuzeigen (sh. Foto)

[image]

naltatis, 2023-02-04:
@StevieC121176 Der Vorschlag mit dem Tooltip war gut. Ich hab das jetzt zusammen mit der Option CO2/Preisdaten einblenden zu lassen in den PR hier 👆 eingebaut. `capacity` muss dafür natürlich konfiguriert sein und wir unterstützen gleich den Fall von mehreren Akkus.

PR #6048 (2023-02-06): Option to show current price and co2 info
Introducing an option to show current price and co2 data in the expanded energy flow. 

**Changes**
- 💵🌱 New UI option to add `price` or `co2` data to the energy flow. (fixes #5488)
  - The option is off by default since we consider it a more enthusiast feature
  - Options are only available of the datasources (`tariffs`) are configured.
  - Tooltip shows all available data (price, co2) including their full units. For brevity we've opted to show short units in the overview (`g` instead of `gCO2e/kWh` and `ct` instead of `ct/kWh`).
- 🔋 Added additional  [... cut]
closed 2023-02-06 by andig