evcc-io/evcc #5919: Server crasht wenn Sonne unter geht  /Kostal Piko CI

GeorgFreiberger, 2023-01-27:
### Describe the bug

Hi,
nach dem hier ja schon ausführlich beschriebenen Drama mit den ABB Terra Wallboxen läuft inzwischen alles soweit:

Shelly Zähler ( WLAN)
Raspberry Pi 2B mit EVCC 0.111  (LAN) 
3x Terra AC Wallbox auf einem Modbus RTU Strang 
1x TCP - RTU Converter ZLAN 5143D(LAN)  (knapp 20€ Aliexpress)  -  funktionierende Einstellungen kann ich nachliefern bei Interesse
1x Kostal Piko CI 30 Wechselrichter (LAN)

Geprüft und funktioniert:  Der Piko CI wie einen Plenticore Hybrid behandeln (hält sich zumindest für den hier relevanten Teil an die Sunspec Protokolle), ohne Batterie natürlich.

das Problem: 
Die Kommunikationsplatine des Piko CI wird ausschließlich über die PV Module versorgt. Wenn die Sonne also weg ist, gibts keine Kommunikation mehr zum PV Zähler und EVCC stürzt ab.
Meine Ideen (die ich aber leider nicht umsetzen kann): 
- Kommunikationstimeout vom Zähler auf 15Std. setzen. 
- oder den letzten gesendeten Wert einfach weiter verwenden bis wieder ein neuer kommt (wieder bis zu ca. 15Std.) 
- ??

### Steps to reproduce

1. Sonne geht unter
2.Wechselrichter schaltet ab und kommuniziert nicht mehr
3.EVCC schaltet mit Kommunikationsfehler ab.
...

### Configuration details

```yaml
wird nachgereicht (mir sind vorhin die Finger abgefroren)
```

### Log details

```text
wird nachgereicht (mir sind vorhin die Finger abgefroren)
```

### What type of operating system are you running?

Linux

### Version

0.111

premultiply, 2023-01-28:
Lösungsvorschlag:
Externer Erzeugungszähler hinter den WR setzen (sowas wie SDM72DMv2) und ansonsten den WR völlig aus der Konfig nehmen.

Wer konstruiert sowas???

GeorgFreiberger, 2023-01-28:
Im konkreten Fall: Kostal
Aber ganz allgemein habe ich letztes Jahr so viel mit den Entwicklungsabteilungen von Herstellern gesprochen wie noch nie - 
und die Hälfte gab dann irgendwann zu dass das Produkt(bzw. der relevante Teil davon) nur ein Zukaufteil ist und sie da nix machen können - (Kostal, SMA, Spelsberg, ABB, Jung, und noch ein paar mehr )

ja, vermutlich wirds darauf raus laufen.... (zusätzlichen Zähler einbauen)

andig, 2023-01-28:
Gibts hier etwas in evcc zu tun? Wäre "WR-Fehler ignorieren" eine sinnvolle Lösung? Woher wissen wir, dass genau der Fehler ignoriert werden darf?

GeorgFreiberger, 2023-01-29:
Wäre natürlich cool, ja, weil dadurch ein weiterer Zähler(Hardware) eingespart werden könnte.
Dies scheint bei beiden Piko CI Wechselrichtern so zu sein (30 und 50 kW). Ist es viel aufwand für diese ein neues Template zu erstellen (quasi Copy-Paste vom Plenticore Hybrid, nur eben ohne Batterie und mit dem Fehler ignorieren)  ?

premultiply, 2023-01-29:
Wiso Plenticore?
Das sind doch stinknormale SunSpec-PV-WR oder?

GeorgFreiberger, 2023-01-30:
Nein, laut Kostal nicht. In keiner Anleitung steht "Sunspec" Was genau differiert konnte aber niemand sagen (und ich hatte auch noch keine Zeit die Tabellen genau zu vergleichen) Mit dem Plenticore template funktioniert es auf jeden Fall (Tagsüber) - von daher. das war dann für mich schon i.O.
Vielleicht gehts auch mit der Standard Sunspec Template, aber auch der WR hat richtig ärger gemacht (über 3x das selbe Update neu eingespielt bis es endlich akzeptiert wurde und so kram...) von daher war ich einfach froh als es dann mal soweit lief.

GeorgFreiberger, 2023-02-01:
ok, werde das nächste Woche mal ausprobieren!

andig, 2023-02-05:
@GeorgFreiberger ich mache mal zu. Gerne mit Testergebnissen melden und dann gehts weiter. Du kannst auch mal mit kurzem `timeout` spielen und schauen ob das hilft- dann kämen zwar Fehler, es würde aber recht schnell weiter gehen im Programmfluss.

closed 2023-02-05 by andig

GeorgFreiberger, 2023-02-13:
Update:
ich war jetzt nochmal vor Ort "grundsätzlich" geht alles.
1. Der Wechselrichter wird definitiv NUR mit der Plenticore Template ausgelesen - sonst kommen immer fehler. ist aber egal - geht ja.
2. Nur zur Info zu den ABB Wallboxen: Wenn man diese einphasig anschließt muss in der Terra Config App auf "Einphasen-3-Leiter netz gestellt werden". das klingt jetzt hier logisch, im Kontext mit den anderen Einstellmöglichkeiten aber nicht mehr.

Habe aktuell jetzt noch ein Problem welches aber meines Erachtens am Fahrzeug hängt (EQS 580, neu geliefert Mitte 2022 - weitere Details unbekannt (Softwarestände etc.)) - werde das mal mit anderen Fahrzeugen testen, aber hier habe ich rekonstruierbar:

Fahrzeug wird eingesteckt (egal ob 1phasige Wallbox (1,5-3,6kW) oder 3 phasige Wallbox (4,5-22kW - Fahrzeug kann nur 11kW)
Ladevorgang beginnt (egal welcher Modus) :-)
Ladevorgang stoppt (egal ob von Hand oder Solarbedingt)
Ladevorgang startet wieder (egal ob von Hand oder Solarbedingt)
 - soweit alles super! 

jetzt gehe ich ca. 15m vom Fahrzeug weg (Keyless Go Schlüssel Reichweite), 
Fahrzeug schließt ab.
Ladevorgang geht ganz normal weiter. :-)
Ladevorgang stoppt (egal ob von Hand oder Solarbedingt)
Ladevorgang startet NICHT wieder selbstständig (egal ob von Hand oder Solarbedingt) ! 
 "...warte auf Fahrzeug..."
Fahrzeug weiß weiterhin das es angesteckt ist, und lässt den Stecker auch nicht los - aber nimmt keine Leistung auf.

Wenn ich mich jetzt wieder dem Fahrzeug nähere, beginnt die Ladung wieder! (ich drücke nichts, ich sehe nur dass der EQS die Griffe ausfährt und Lichtspielereien beginnt, weil er den Schlüssel in Reichweite erkennt.)

Morgens beginnt das Laden also quasi nie von alleine :-s (ob man hier durch Spielen an der Fahrzeugapp was machen kann weiß ich noch nicht.)  soviel zum aktuellen Update :-)

MarkusThur, 2023-03-28:
Auch wenn geschlossen..

@GeorgFreiberger

Der Kostal Piko CI versteht (oder spricht) SunSpec. Findet sich auch so in der Schnittstellenbeschreibung... Der SunSpec 103 (3 phasen Umrichter) block findet sich ab Adresse 40070 (bzw 40071 Registernummer) er hat auch noch ein paar andere SunSpec blocks implementiert. Findet sich im Interface Dokument bei den Downloads zum Piko CI auf der Kostalseite.

Und das mit dem Abschalten, so bin ich auf den thread hier aufmerksam geworden ist mir heute beim CI 50 aufgefallen.. Wollte nach Sonnenuntergang nochwas umstellen und dann war auch das kommunikationsmodul aus... Ich dachte schon das Teil sei im A.... 😂

Auf jedenfall zeigt meine Protkolldatei um 20:22:57 einen Statuseechsel im entsprechenden sunspec register von Einspeisen auf Standby an und dann gibt es nurnoch connection time outs.. War dann am WR, alle lampen aus, kein wlan hotspot, keine chance mit der APP dran zu kommen.. Und dann das hier gefunden und gedacht.. Strom sparen ist ja fein, aber das ist Bul.shi. Werde morgen erstmal den Kostalsupport etwas rund machen.. 😂