evcc-io/evcc #5922: invalid zipcode bei grünstromindex

StevieC121176, 2023-01-28:
### Describe the bug

Bei PLZ die mit 0 beginnen, z.Bsp. Leipzig 04103,  kommt bei mir der Fehler:

```
planner:
[gsi   ] ERROR 2023/01/28 09:28:09 invalid zipcode
[main  ] FATAL 2023/01/28 09:28:09 cannot create tariff 'grünstromindex': invalid zipcode
pi@raspberrypi:~ $ 
```

ich habe ander PLZ die nicht mit 0 beginnen probiert, diese funktionieren.

### Steps to reproduce

evcc -tariff

### Configuration details

```yaml
planner:
    type: grünstromindex
    zip: 04103
```

### Log details

```text
[gsi   ] ERROR 2023/01/28 09:37:01 invalid zipcode
[main  ] ERROR 2023/01/28 09:37:01 failed configuring planner tariff: cannot create tariff 'grünstromindex': invalid zipcode
```

### What type of operating system are you running?

Linux

### Version

0.111.1 (253e05da)

VolkerK62, 2023-01-28:
Haste mal `"04103"` versucht

StevieC121176, 2023-01-28:
Ich hatte mich an die Schreibweise aus dem Blog Artikel gehalten, aber ja du hast recht, mit "04103" funktioniert es. Komisch nur das z.bsp. 13607 auch ohne "" funktioniert.

closed 2023-01-28 by StevieC121176

VolkerK62, 2023-01-28:
das liegt an der führenden Null.

VolkerK62, 2023-01-28:
@andig  obwohl `zip` als string definiert ist, scheinen führende Nullen ein Problem zu sein. 
Ich habe Grünstromindex mal in die Doku aufgenommen und dort einen entsprechenden Hinweis plaziert.