evcc-io/evcc #5943: Cannot create vehicle: updated ToS: form not found

ammann-ruchti, 2023-01-30:
### Describe the bug

Seit gestern kann evcc das Fahrzeug nicht mehr einbinden (ungefähr dasselbe wie auch am 29. Januar 2022!) Hier die Info aus dem "debug vehicle":

[main  ] ERROR 2023/01/30 09:36:43 creating vehicle ev4 failed: cannot create vehicle 'template': cannot create vehicle 'enyaq': updated ToS: form not found
.Soc:     vehicle not available: cannot create vehicle 'template': cannot create vehicle 'enyaq': updated ToS: form not found

Bin ich der Einzige mit diesem Problem? Wie kann dieses behoben werden?

Letzte Nightly version hat Problem auch.

### Steps to reproduce

N/A

### Configuration details

```yaml
N/A
```

### Log details

```text
pi-evcc@evcc-raspi:~ $ evcc --log debug
[main  ] INFO 2023/01/30 09:42:29 evcc 0.111.1
[main  ] INFO 2023/01/30 09:42:29 using config file: /home/pi-evcc/evcc.yaml
[main  ] INFO 2023/01/30 09:42:29 starting ui and api at :7070
[db    ] INFO 2023/01/30 09:42:29 using sqlite database: /home/pi-evcc/.evcc/evcc.db
[main  ] ERROR 2023/01/30 09:42:31 creating vehicle ev4 failed: cannot create vehicle 'template': cannot create vehicle 'enyaq': updated ToS: form not found
[site  ] INFO 2023/01/30 09:42:31 site config:
[site  ] INFO 2023/01/30 09:42:31   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2023/01/30 09:42:31     grid:      power ✓ energy ✗ currents ✗
[site  ] INFO 2023/01/30 09:42:31     pv 1:      power ✓ energy ✗ currents ✗
[site  ] INFO 2023/01/30 09:42:31     battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✗
[site  ] INFO 2023/01/30 09:42:31   vehicles:
[site  ] INFO 2023/01/30 09:42:31     vehicle 1: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
[lp-1  ] INFO 2023/01/30 09:42:31 loadpoint 1:
[lp-1  ] INFO 2023/01/30 09:42:31   mode:        pv
[lp-1  ] INFO 2023/01/30 09:42:31   charger:     power ✗ energy ✗ currents ✗ phases ✗ wakeup ✗
[lp-1  ] INFO 2023/01/30 09:42:31   meters:      charge ✗
[lp-1  ] DEBUG 2023/01/30 09:42:31 phase timer inactive
[lp-1  ] DEBUG 2023/01/30 09:42:31 pv timer inactive
[main  ] FATAL 2023/01/30 09:42:31 listen tcp :7070: bind: address already in use
```

### What type of operating system are you running?

Linux

### Version

0.111.1

StevieC121176, 2023-01-30:
Einfach die neuen AGB auf der Webseite oder in der connect App akzeptieren. Das Problem hatte ich auch. Liegt aber am VW Konzern, nicht an evcc

ammann-ruchti, 2023-01-30:
Danke für den Hinweis. Leider habe ich aktuell neu ein Problem mit der Fahrzeugregistrierung bei Skoda und muss dies gemäss telefonischer Kontaktnahme mit Skoda-Connect neu aufsetzen. Der Zugriff von evcc funktioniert dagegen jetzt wieder.

closed 2023-01-30 by ammann-ruchti

emign, 2023-01-31:
> Einfach die neuen AGB auf der Webseite oder in der connect App akzeptieren. Das Problem hatte ich auch. Liegt aber am VW Konzern, nicht an evcc

Bei mir wird keine neue TOS/AGB angezeigt auf der Seite/App bzw. ich habe diese schon bestätigt. Der Fehler erscheint trotzdem weiterhin. 

@StevieC121176 Hast du evtl noch andere Schritte unternommen, welche die eigentliche Ursache für das Fixing verantwortlich sein könnten?

ammann-ruchti, 2023-01-31:
Hast du mal probiert, dich via Webbrowser bei Skoda-connect.com anzumelden? Bei mir ging das nicht. Ich hatte dann mit dem Support von Skoda Kontakt aufgenommen. Mein Fahrzeug war mehrfach registriert. Ich habe es in der App dann gelöscht und neu angemeldet (mit QR-Code von der neuen Benutzeranmeldung im Fahrzeug). Seit dann läuft es bei mir wieder.

ammann-ruchti, 2023-01-31:
Du kannst ja trotzdem mal probieren, das Fahrzeug aus der App zu löschen und dann neu anzumelden. Vielleicht hilft das schon. Verlieren tust du dabei nichts.

ammann-ruchti, 2023-01-31:
Und ja, danach evcc neu starten.