evcc-io/evcc #5950: influx ERROR due to phases configured is empty

VolkerK62, 2023-01-30:
Nach einem Neustart kommt immer ein influx ERROR.
Negative Auswirkungen konnte ich bisher nicht feststellen.

```
Jan 28 14:26:49 raspberrypi systemd[1]: Started evcc.
Jan 28 14:26:49 raspberrypi evcc[12898]: [main  ] INFO 2023/01/28 14:26:49 evcc 0.111.1 (253e05da)
Jan 28 14:26:49 raspberrypi evcc[12898]: [main  ] INFO 2023/01/28 14:26:49 using config file: /etc/evcc.yaml
Jan 28 14:26:49 raspberrypi evcc[12898]: [main  ] INFO 2023/01/28 14:26:49 starting ui and api at :7070
Jan 28 14:26:49 raspberrypi evcc[12898]: [db    ] INFO 2023/01/28 14:26:49 using sqlite database: /root/.evcc/evcc.db
Jan 28 14:26:49 raspberrypi evcc[12898]: [mqtt  ] INFO 2023/01/28 14:26:49 connecting evcc-804384411 at tcp://localhost:1883
Jan 28 14:26:49 raspberrypi evcc[12898]: [mqtt  ] DEBUG 2023/01/28 14:26:49 tcp://localhost:1883 connected
Jan 28 14:26:49 raspberrypi evcc[12898]: [proxy-5200] DEBUG 2023/01/28 14:26:49 modbus proxy for 192.168.178.26:502 listening at :5200
Jan 28 14:26:52 raspberrypi evcc[12898]: [lp-1  ] WARN 2023/01/28 14:26:52 poll mode '{always 4h0m0s}' may deplete your battery or lead to API misuse. USE AT YOUR OWN RISK.
Jan 28 14:26:53 raspberrypi evcc[12898]: [site  ] INFO 2023/01/28 14:26:53 site config:
Jan 28 14:26:53 raspberrypi evcc[12898]: [site  ] INFO 2023/01/28 14:26:53   meters:      grid ✓ pv ✓ battery ✓
Jan 28 14:26:53 raspberrypi evcc[12898]: [site  ] INFO 2023/01/28 14:26:53     grid:      power ✓ energy ✓ currents ✓
Jan 28 14:26:53 raspberrypi evcc[12898]: [site  ] INFO 2023/01/28 14:26:53     pv 1:      power ✓ energy ✗ currents ✗
Jan 28 14:26:53 raspberrypi evcc[12898]: [site  ] INFO 2023/01/28 14:26:53     battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✓
Jan 28 14:26:53 raspberrypi evcc[12898]: [site  ] INFO 2023/01/28 14:26:53   vehicles:
Jan 28 14:26:53 raspberrypi evcc[12898]: [site  ] INFO 2023/01/28 14:26:53     vehicle 1: range ✓ finish ✓ status ✓ climate ✗ wakeup ✗
Jan 28 14:26:53 raspberrypi evcc[12898]: [lp-1  ] INFO 2023/01/28 14:26:53 loadpoint 1:
Jan 28 14:26:53 raspberrypi evcc[12898]: [lp-1  ] INFO 2023/01/28 14:26:53   mode:        off
Jan 28 14:26:53 raspberrypi evcc[12898]: [lp-1  ] INFO 2023/01/28 14:26:53   charger:     power ✓ energy ✓ currents ✓ phases ✗ wakeup ✗
Jan 28 14:26:53 raspberrypi evcc[12898]: [lp-1  ] INFO 2023/01/28 14:26:53   meters:      charge ✓
Jan 28 14:26:53 raspberrypi evcc[12898]: [lp-1  ] INFO 2023/01/28 14:26:53     charge:    power ✓ energy ✓ currents ✓
Jan 28 14:26:53 raspberrypi evcc[12898]: [influx] TRACE 2023/01/28 14:26:53 write bufferSoc=map[value:0] (map[])
Jan 28 14:26:53 raspberrypi evcc[12898]: [influx] TRACE 2023/01/28 14:26:53 write prioritySoc=map[value:0] (map[])
Jan 28 14:26:53 raspberrypi evcc[12898]: [influx] TRACE 2023/01/28 14:26:53 write residualPower=map[value:50] (map[])
Jan 28 14:26:53 raspberrypi evcc[12898]: [influx] TRACE 2023/01/28 14:26:53 write minCurrent=map[value:6] (map[loadpoint:Carport vehicle:])
Jan 28 14:26:53 raspberrypi evcc[12898]: [influx] TRACE 2023/01/28 14:26:53 write maxCurrent=map[value:20] (map[loadpoint:Carport vehicle:])
Jan 28 14:26:53 raspberrypi evcc[12898]: [influx] TRACE 2023/01/28 14:26:53 write phasesConfigured=map[] (map[loadpoint:Carport vehicle:])
Jan 28 14:26:53 raspberrypi evcc[12898]: [influx] ERROR 2023/01/28 14:26:53 no serializable fields
Jan 28 14:26:53 raspberrypi evcc[12898]: [influx] TRACE 2023/01/28 14:26:53 write phasesEnabled=map[value:1] (map[loadpoint:Carport vehicle:])
Jan 28 14:26:53 raspberrypi evcc[12898]: [influx] TRACE 2023/01/28 14:26:53 write phasesActive=map[value:1] (map[loadpoint:Carport vehicle:])
Jan 28 14:26:53 raspberrypi evcc[12898]: [lp-1  ] DEBUG 2023/01/28 14:26:53 phase timer inactive
Jan 28 14:26:53 raspberrypi evcc[12898]: [lp-1  ] DEBUG 2023/01/28 14:26:53 pv timer inactive
Jan 28 14:26:53 raspberrypi evcc[12898]: [lp-1  ] INFO 2023/01/28 14:26:53 vehicle updated: unknown -> (T)Raumschiff
Jan 28 14:26:53 raspberrypi evcc[12898]: [influx] TRACE 2023/01/28 14:26:53 write minSoc=map[value:0] (map[loadpoint:Carport vehicle:])
Jan 28 14:26:53 raspberrypi evcc[12898]: [influx] TRACE 2023/01/28 14:26:53 write targetSoc=map[value:100] (map[loadpoint:Carport vehicle:])
Jan 28 14:26:53 raspberrypi evcc[12898]: [lp-1  ] DEBUG 2023/01/28 14:26:53 set charge mode: pv
Jan 28 14:26:53 raspberrypi evcc[12898]: [lp-1  ] DEBUG 2023/01/28 14:26:53 set target soc: 100
Jan 28 14:26:53 raspberrypi evcc[12898]: [influx] TRACE 2023/01/28 14:26:53 write vehicleCapacity=map[value:72] (map[loadpoint:Carport vehicle:(T)Raumschiff])
Jan 28 14:26:53 raspberrypi evcc[12898]: [influx] TRACE 2023/01/28 14:26:53 write targetEnergy=map[value:0] (map[loadpoint:Carport vehicle:(T)Raumschiff])
Jan 28 14:26:53 raspberrypi evcc[12898]: [influx] TRACE 2023/01/28 14:26:53 write phasesActive=map[value:1] (map[loadpoint:Carport vehicle:(T)Raumschiff])
Jan 28 14:26:53 raspberrypi evcc[12898]: [influx] TRACE 2023/01/28 14:26:53 write vehicleSoc=map[value:0] (map[loadpoint:Carport vehicle:(T)Raumschiff])
Jan 28 14:26:53 raspberrypi evcc[12898]: [influx] TRACE 2023/01/28 14:26:53 write vehicleRange=map[value:0] (map[loadpoint:Carport vehicle:(T)Raumschiff])
Jan 28 14:26:53 raspberrypi evcc[12898]: [influx] TRACE 2023/01/28 14:26:53 write vehicleTargetSoc=map[value:0] (map[loadpoint:Carport vehicle:(T)Raumschiff])
Jan 28 14:26:53 raspberrypi evcc[12898]: [influx] TRACE 2023/01/28 14:26:53 write targetSoc=map[value:100] (map[loadpoint:Carport vehicle:(T)Raumschiff])
Jan 28 14:26:53 raspberrypi evcc[12898]: [influx] TRACE 2023/01/28 14:26:53 write minSoc=map[value:0] (map[loadpoint:Carport vehicle:(T)Raumschiff])
Jan 28 14:26:53 raspberrypi evcc[12898]: [site  ] DEBUG 2023/01/28 14:26:53 ----
```

_Originally posted by @VolkerK62 in https://github.com/evcc-io/evcc/discussions/5925_

VolkerK62, 2023-01-30:
Es liegt wohl daran, das phasesConfigured leer ist.
Der ERROR taucht zusätzlich immer dann auf, wenn man per API den Wert für phases ändert.

```
Jan 28 14:26:53 raspberrypi evcc[12898]: [influx] TRACE 2023/01/28 14:26:53 write phasesConfigured=map[] (map[loadpoint:Carport vehicle:])
Jan 28 14:26:53 raspberrypi evcc[12898]: [influx] ERROR 2023/01/28 14:26:53 no serializable fields
```

andig, 2023-01-30:
Ist das im aktuellen Release auch der Fall?

VolkerK62, 2023-01-31:
Downgrade auf 0.111.1
kein Fehler mehr.

`Jan 31 04:59:07 raspberrypi evcc[13524]: [influx] TRACE 2023/01/31 04:59:07 write phasesConfigured=<nil> (map[loadpoint:Carport vehicle:(T)Raumschiff])`

VolkerK62, 2023-02-01:
Da scheint noch mehr rausgefallen zu sein.  z.B. `chargecurrets` der einzelnen Phasen.
Ich habe mich gewundert, dass ich da keine Werte bekommen und gerade zufällig gesehen, dass das gestern mit dem letzten Release noch funktioniert hat. Seit das Nightly wieder drauf ist, kommt da nix mehr.

andig, 2023-02-03:
Sollte in 15min wieder gehen, Tests ergänzt damit das nicht wieder passiert. Danke für Eure Geduld.

closed 2023-02-03 by andig