evcc-io/evcc #5961: Ladeplaner - Zielzeit festlegen

nuiler, 2023-01-31:
Die 2 Varianten bringen keine Besserung anbei dazu Screenshots.

```
tariffs:
  currency: EUR # (default EUR)
  grid:
    type: fixed
    price: 0.4447 # [currency]/kWh
    zones:
      - days: Mo-So
        hours: 21:30-0,0-5:30
        price: 0.3912 # EUR/kWh
  feedin:
    type: fixed
    price: 0.17 # [currency]/kWh
```
[image]

```
tariffs:
  currency: EUR # (default EUR)
  grid:
    type: fixed
    price: 0.4447 # [currency]/kWh
    zones:
      - days: Mo-So
        hours: 21:30-5:30
        price: 0.3912 # EUR/kWh
  feedin:
    type: fixed
    price: 0.17 # [currency]/kWh
```
[image]

_Originally posted by @nuiler in https://github.com/evcc-io/evcc/discussions/5639#discussioncomment-4823289_

nuiler, 2023-01-31:
Das ist Die Ausgabe bei

```
tariffs:
  currency: EUR # (default EUR)
  grid:
    type: fixed
    price: 0.4447 # [currency]/kWh
    zones:
      - days: Mo-So
        hours: 21:30-0,0-5:30
        price: 0.3912 # EUR/kWh
  feedin:
    type: fixed
    price: 0.17 # [currency]/kWh
```

pi@evcc:~ $ evcc tariff
[main  ] INFO 2023/01/31 10:47:07 evcc 0.111.1 (c496ed80)
[main  ] INFO 2023/01/31 10:47:07 using config file: /etc/evcc.yaml
[db    ] INFO 2023/01/31 10:47:08 using sqlite database: /home/pi/.evcc/evcc.db
[mqtt  ] INFO 2023/01/31 10:47:08 connecting evcc-1015445303 at tcp://192.168.178.6:1883
grid:
From                   To                     Price/Cost
2023-01-31 00:00:00    2023-01-31 05:30:00    0.445
2023-01-31 05:30:00    2023-01-31 21:30:00    0.445
2023-01-31 21:30:00    2023-02-01 00:00:00    0.391
2023-02-01 00:00:00    2023-02-01 05:30:00    0.445
2023-02-01 05:30:00    2023-02-01 21:30:00    0.445
2023-02-01 21:30:00    2023-02-02 00:00:00    0.391
2023-02-02 00:00:00    2023-02-02 05:30:00    0.445
2023-02-02 05:30:00    2023-02-02 21:30:00    0.445
2023-02-02 21:30:00    2023-02-03 00:00:00    0.391
2023-02-03 00:00:00    2023-02-03 05:30:00    0.445
2023-02-03 05:30:00    2023-02-03 21:30:00    0.445
2023-02-03 21:30:00    2023-02-04 00:00:00    0.391
2023-02-04 00:00:00    2023-02-04 05:30:00    0.445
2023-02-04 05:30:00    2023-02-04 21:30:00    0.445
2023-02-04 21:30:00    2023-02-05 00:00:00    0.391
2023-02-05 00:00:00    2023-02-05 05:30:00    0.445
2023-02-05 05:30:00    2023-02-05 21:30:00    0.445
2023-02-05 21:30:00    2023-02-06 00:00:00    0.391
2023-02-06 00:00:00    2023-02-06 05:30:00    0.445
2023-02-06 05:30:00    2023-02-06 21:30:00    0.445
2023-02-06 21:30:00    2023-02-07 00:00:00    0.391

feedin:
From                   To                     Price/Cost
2023-01-31 00:00:00    2023-02-01 00:00:00    0.170
2023-02-01 00:00:00    2023-02-02 00:00:00    0.170
2023-02-02 00:00:00    2023-02-03 00:00:00    0.170
2023-02-03 00:00:00    2023-02-04 00:00:00    0.170
2023-02-04 00:00:00    2023-02-05 00:00:00    0.170
2023-02-05 00:00:00    2023-02-06 00:00:00    0.170
2023-02-06 00:00:00    2023-02-07 00:00:00    0.170

nuiler, 2023-01-31:
> Stimmt denn die Ausgabe von
> 
> ```
> evcc tariff
> ```
> 
> Die Variante mit 0:01 ist auf jeden Fall falsch. Das wäre dann 1min Standardtarif.

Wurde mir hier vorgeschlagen
https://github.com/evcc-io/evcc/discussions/5639#discussioncomment-4821762

andig, 2023-01-31:
Die Preisstaffel sieht doch richtig aus? Dann müssten wir "nur" noch rausfinden was bei der Visualisierung schief läuft.

nuiler, 2023-01-31:
> Die Preisstaffel sieht doch richtig aus? Dann müssten wir "nur" noch rausfinden was bei der Visualisierung schief läuft.

Nein ist nicht richtig. Das wird angezeigt.
2023-01-31 21:30:00 2023-02-01 00:00:00 0.391
2023-02-01 00:00:00 2023-02-01 05:30:00 0.445
2023-02-01 05:30:00 2023-02-01 21:30:00 0.445

sollte aber so aussehen 21:30-5:30 0.3912€
2023-01-31 21:30:00 2023-02-01 00:00:00 0.391
2023-02-01 00:00:00 2023-02-01 05:30:00 **0.391**
2023-02-01 05:30:00 2023-02-01 21:30:00 0.445

VolkerK62, 2023-01-31:
>Die Variante mit 0:01 ist auf jeden Fall falsch. Das wäre dann 1min Standardtarif.

Das war nur ein Test.
Es sieht so aus, als würde eine `0` als Startzeit nicht funktionieren.
Mit `0:01` als Startzeit hat zumindest die Visualisierung richtig ausgesehen, mit Ausnahme der Stunde 0-1.

andig, 2023-01-31:
> Nein ist nicht richtig. Das wird angezeigt.

Sehr gut, jetzt weiss ich wo ich suchen muss. Das hätte ich definitiv auch anders erwartet. Danke für Deine Geduld mich in die richtige Richtung zu lenken ;)

andig, 2023-01-31:
Ist im Nightly ab morgen früh behoben. Danke für den Bug report!

closed 2023-01-31 by andig

nuiler, 2023-02-01:
Vielen Dank für die schnelle Umsetzung. 
Falls ihr noch ein Testobjekt für das automatische Zielladen pro Fahrzeug benötigt, bin gerne dabei.

https://github.com/evcc-io/evcc/discussions/5639#discussioncomment-4675124