evcc-io/evcc #5987: Messaging - ${loadpoint} Parameter nicht verfügbar

princemaxwell, 2023-02-01:
### Discussed in https://github.com/evcc-io/evcc/discussions/5977

<div type='discussions-op-text'>

<sup>Originally posted by **princemaxwell** February  1, 2023</sup>
Laut Dokumentation [https://docs.evcc.io/docs/reference/configuration/messaging#msg](url) müsste der Parameter ${loadpoint} für das messaging zur Verfügung stehen.

```
    connect: # vehicle connect event
      title: VERBUNDEN (${loadpoint}) - {{.vehicleTitle}} mit {{.title}}
```
Funktioniert aber leider nicht:

[image]
</div>

andig, 2023-02-02:
@princemaxwell was erwartest Du denn von dem Parameter? Reicht Dir nicht `title`, also ist das eher ein Dokuthema?

/cc @VolkerK62

princemaxwell, 2023-02-02:
@andig ich wollte die loadpoint_id als Zahl an ein Script übergeben, damit dort beim Verbinden des Fahrzeugs via CURL die targettime fürs Zielladen automatisch gesetzt wird und ich das nicht manuell machen muss. Da ich zwei Ladepunkte habe, hätte ich gerne die ID dort übergeben. Nun musste ich über IF den title auswerten und zweimal CURL mit fester Zahl bauen, damit entsprechend dem Ladepunkt die Aktion durchgeführt wird.

```
if [[ "$1" == *"Mennekes"* ]] # loadpoint 1
	then
		curl -X POST http://$ip/api/loadpoints/1/target/time/$targettime > /dev/null
		echo "Setting targettime (loadpoint 1 - Mennekes): " $targettime >> $log
	else
		if [[ "$1" == *"Vestel"* ]] # loadpoint 2
		then
			curl -X POST http://$ip/api/loadpoints/2/target/time/$targettime > /dev/null
			echo "Setting targettime (loadpoint 2 - Vestel): " $targettime >> $log
		else
...
...
```

Das funktioniert zwar auch, ist aber unnötiger Code.

**Entscheidet ihr, ob ihr den ${loadpoint} noch mit aufnehmt oder einfach nur die Doku anpasst ;-)**

andig, 2023-02-02:
Use case verstanden, ich schau mal wo der hin verschwunden ist.

VolkerK62, 2023-02-02:
@andig  also kein Doku-Thema??

andig, 2023-02-03:
Nope, die hatte ich wegrationalisiert. Closed in https://github.com/evcc-io/evcc/commit/80acb35d76c36b34b93de703d013ca1f976f6d6e.

closed 2023-02-03 by andig