evcc-io/evcc #5992: Start Soc beim Offline-Vehicle

VolkerK62, 2023-02-02:
Idee ist, beim Offline-Vehicle im UI die Eingabe eines Start-Soc zu ermöglichen.
Aus diesem Start-Soc und dem Ziel-Soc ergibt sich dann die erforderliche Energiemenge.
Die kann man auch jetzt schon selber ausrechnen und einstellen, aber so wäre es etwas komfortabler.

closed 2023-02-02 by VolkerK62

andig, 2023-02-02:
Das würde ich wirklich nur sehr ungern machen- m.E. viel zu umständlich. Dann lieber direkt "10kWh" laden.

VolkerK62, 2023-02-02:
Alles gut. Ich habe erst das Issue aufgemacht und es dann mal ausprobiert. Deshalb auch wieder geschlossen. 
Wenn man beim Offline-Vehicle die Capacity eingetragen hat, kann man ganz einfach bei Ladeziel die zu ladende Prozentzahl auswählen. Ich hatte vorher nicht geschaut, wie das im UI aussieht. Ich dachte da sei nur eine Menge auswählbar.