evcc-io/evcc #5995: Messaging Verzögern bis alle Werte berechnet sind

princemaxwell, 2023-02-02:
### Discussed in https://github.com/evcc-io/evcc/discussions/5976

<div type='discussions-op-text'>

<sup>Originally posted by **princemaxwell** February  1, 2023</sup>
Ich habe das Messaging mit NTFY eingerichtet.
Manchmal kommt es vor, dass manche Variablen mit `<no value>` in der Message erscheinen.

Das passiert regelmäßig bei `chargeRemainingEnergy` und seltener bei `vehicleRange`.
Ich vermute, dass zum Zeitpunkt des Messaging die Werte noch nicht berechnet sind, oder von der Car-API abgeholt wurden.

Wäre es möglich, bzw. würde es was bringen, das Messaging **intern** etwas zu verzögern, bis alle Werte vorliegen?</div>

andig, 2023-02-02:
Von welchen Nachrichten sprechen wir denn mit welchen konkreten Problemen?

princemaxwell, 2023-02-02:
Das passiert nur, wenn der Wagen verbunden wird und aufgrund der Einstellungen zu Zielladen ein **sofortiges** Laden startet.
Das betrifft dann hin und wieder Daten aus der Fahrzeug-API (weil die wahrscheinlich zu lange zum Übermitteln benötigt) und in allen Fällen die `chargeRemainingDuration`, die anscheinend von EVCC noch nicht berechnet wurde, siehe Screenshot. Zu fehlenden API-Daten habe ich leider gerade keine Nachricht mehr.

Wenn der Wagen bereits längere Zeit verbunden ist und Zielladen zu einem späteren Zeitpunkt startet, ist alles vollständig.

```
  events:
    start: # charge start event
      title: START Ladevorgang - {{.vehicleTitle}}
      msg: |
        
        Wallbox {{.title}} hat das Laden von {{.vehicleTitle}} im {{ toString .mode | upper }} Modus gestartet.
        --------------------------
        evcc Status {{printf `(%d-%02d-%02d %02d:%02d:%02d)` now.Year now.Month now.Day now.Hour now.Minute now.Second}}
        
        Netz-Leistung: {{round (divf .gridPower 1000) 3 }} kW
        PV-Leistung: {{round (divf .pvPower 1000) 3 }} kW
        Hausverbrauch: {{round (divf .homePower 1000) 3 }} kW
        
        Ziel-Füllstand: {{.targetSoc}} % (={{round (divf .chargeRemainingEnergy 1000) 2 }} kWh / {{.chargeRemainingDuration}})
```
[image]

andig, 2023-02-02:
> Das betrifft hin und wieder Daten aus der Fahrzeug-API (weil die wahrscheinlich zu lange zum Übermitteln benötigt) 

Das ist so. Die sind einfach asynchron.

> und in vielen Fällen die chargeRemainingDuration, die anscheinend von EVCC aus irgendeinem Grund noch nicht berechnet wurde, siehe Screenshot. 

In welchen Fällen? Für welche Nachricht? Start?

princemaxwell, 2023-02-02:
@andig Ich nutze es aktuell nur in `start,` kann es aber gerne auch mit den anderen versuchen, obwohl bei `disconnect` und `stop` macht es wenig Sinn...

andig, 2023-02-03:
Ich hab Dein Beispiel nochmal angeschaut. Alles was vom Auto kommt kann beliebig spät oder gar nicht kommen. Mot golang templates (`{{ if .ChargeRemainingEnergy }}...{{ end }}`) kannst Du die Abwesenheit eines Wertes aber abfangen. Sonst lässt sich nicht viel tun...

closed 2023-02-03 by andig