evcc-io/evcc #6004: free trial Token

VolkerK62, 2023-02-03:
Wäre es möglich einen free trial Token anzubieten?

Für einige, teils beliebte Geräte ist ein Token notwendig.
Das hält Neu-Einsteiger (vor allem mit geringen IT-Kenntnissen) evtl davon ab, evcc zu nutzen, zumal ein geringes monatliches Sponsoring demnächst nur noch mit Kreditkarte möglich ist. (Wer kauft schon gerne die Katze im Sack ;-) )

Durch diesen free trial Token hätte man die Möglichkeit, evcc zu testen, für gut zu befinden und dann auf das normale Sponsoring zu wechseln.

andig, 2023-02-03:
Diskutieren wir. Wer Probleme hat kann sich auch ohne jederzeit bei info@evcc.io melden. Ich mach hier trotzdem zu da es auf diesem Weg nicht gelöst werden wird.

/cc @naltatis @premultiply

closed 2023-02-03 by andig

VolkerK62, 2023-02-03:
ok. Danke.