evcc-io/evcc #6005: Feintuning prioritySoc

VolkerK62, 2023-02-03:
Aktuell hat man mittels `prioritySoc` nur die Option „ganz oder gar nicht“.

Meine Idee ist, der Hausbatterie während der Fahrzeugladung und trotz Erreichens des `prioritySoc` eine gewisse Lademöglichkeit zu bieten.
Das Ein- und Ausschaltverhalten, bleibt unverändert. 
Hier geht es nur um die Regelung der Fahrzeugladeleistung.

Beispiel:
```
prioritySoc: 0
minBatteryPower: 500
maxBatteryPower: 1000
```

Der Überschuss steigt an, die Hausbatterie wird geladen.
Erreicht die Ladung der Hausbatterie die Mindestladeleistung des Fahrzeugs, startet die Ladung.
Bis hierhin unverändertes Verhalten.

Wenn jetzt der Überschuss steigt, wird `chargecurrent` erst dann erhöht, wenn `maxBatteryPower` erreicht ist.
Wenn der Überschuss sinkt, wird `chargecurrent` reduziert, sobald `minBatteryPower` erreicht ist.
Ist `mincurrent` erreicht, gilt wieder das unveränderte Verhalten bzgl. Abschaltung der Fahrzeugladung.

Im Endeffekt wird trotz `prioritySoc`, die Hausbatterie mit 500 bis 1000 W geladen (entsprechender Überschuss vorausgesetzt).
Das ist besonders in der Übergangszeit und im Winter nützlich, um Einspeisung zu vermeiden und den Eigenverbrauch zu optimieren.

andig, 2023-02-03:
Werden wir nicht umsetzen. Wir wollen weniger Parameter, nicht mehr. Die Alternative wird hoffentlich ein schlauerer PV Modus ("keine Einspeisung").

closed 2023-02-03 by andig

VolkerK62, 2023-02-03:
Schade. Aber ok, wenn am Ende etwas Ähnliches rauskommt ;-)