evcc-io/evcc #6009: Just change from minpv to now -  now i get an error for creating the

vmarcs, 2023-02-03:
### Describe the bug

I changed my working config from minpv to now on loadpoints
now i got following errors:

cannot create charger 'shelly'
cannot create charger 'template'
cannot create charger 'shelly'
(ECFABC6A619A) missing user/password

I can not find errors in the yaml also not know what kind of passoword this should be ? 

chargers part: 
  
chargers:  
- type: template  
  template: easee  
  user: *****  
  password: *****  
  charger: EH7W2ZKY  
  name: easee  
- type: template  
  template: shelly   
  host: 192.168.75.251    
  channel: 0    
  standbypower: 15  
  name: shelly  

Version 111.1

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070  
network:  
  schema: http  
  host: evcc.local # .local suffix announces the hostname on MDNS  
  port: 7070  
  
log: info  
levels:  
  cache: error  
  
interval: 10s # control cycle interval  
  
sponsortoken: *****  
  
meters:  
- type: template  
  template: fronius-gen24   
  usage: grid    
  host: <ip hidden>
  port: 502    
  name: grid1  
- type: template  
  template: fronius-gen24   
  usage: pv    
  host: <ip hidden>    
  port: 502    
  name: pv2  
- type: template  
  template: fronius-gen24   
  usage: battery    
  host: <ip hidden>
  port: 502    
  name: battery3  
- type: template  
  template: solarlog   
  usage: pv    
  host: <ip hidden>  
  port: 502    
  name: pv4  
- type: template  
  template: fronius-gen24   
  usage: battery    
  host: <ip hidden>   
  port: 502    
  name: battery5  
  
chargers:  
- type: template  
  template: easee  
  user: *****  
  password: *****  
  charger: EH7W2ZKY  
  name: easee  
- type: template  
  template: shelly   
  host: <ip hidden>
  channel: 0    
  standbypower: 15  
  name: shelly  
  
vehicles:  
- type: template  
  template: peugeot   
  title: e-Mona    
  user: *****  
  password: *****  
  vin: *****  
  capacity: 50    
  name: ev6  
- type: template  
  template: bmw  
  title: e-bimmer  
  user: *****  
  password: *****  
  vin: *****  
  capacity: 12  
  name: ev7  
  identifiers:   
  - D6E7174D  
  
loadpoints:  
- title: Wallbox  
  charger: easee  
  vehicles:  
  - ev6  
  - ev7  
  mode: now  
  phases: 3  
  mincurrent: 6  
  maxcurrent: 32  
  resetOnDisconnect: true  
- title: Shelly  
  charger: shelly  
  vehicles:  
  - ev7  
  mode: now  
  phases: 1  
  mincurrent: 6  
  maxcurrent: 12  
  resetOnDisconnect: true  
  
site:  
  title: Bisingen  
  meters:  
    grid: grid1  
    pvs:  
    - pv2  
    - pv4  
    batteries:  
    - battery3  
    - battery5  
  
mqtt:  
 broker: <ip hidden>
 topic: evcc  
 user: *****  
 password: *****
```

### Log details

```text
[main  ] INFO 2023/02/03 10:21:07 evcc 0.111.1
[main  ] INFO 2023/02/03 10:21:07 using config file: /etc/evcc.yaml
[main  ] INFO 2023/02/03 10:21:07 starting ui and api at :7070
[db    ] INFO 2023/02/03 10:21:08 using sqlite database: /home/sam/.evcc/evcc.db
[mqtt  ] INFO 2023/02/03 10:21:08 connecting evcc-546753443 at tcp://<hiddenip>:1882
[mqtt  ] DEBUG 2023/02/03 10:21:08 tcp://<hiddenip>:1882 connected
[main  ] FATAL 2023/02/03 10:21:13 cannot create charger 'shelly': cannot create charger 'template': cannot create charger 'shelly':  (ECFABC6A619A) missing user/password
[main  ] FATAL 2023/02/03 10:21:13 will attempt restart in: 5m0s
[main  ] FATAL 2023/02/03 10:21:13 listen tcp :7070: bind: address already in use
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.111.1

andig, 2023-02-03:
Dein evcc läuft doppelt

closed 2023-02-03 by andig

vmarcs, 2023-02-03:
Nein wenn ich systemctl stop evcc mache geht nix mehr

andig, 2023-02-03:
Mag sein. „Address already in use“ ist dennoch eindeutig.

vmarcs, 2023-02-03:
nehme ich nen anderen port: [main  ] FATAL 2023/02/03 12:13:50 listen tcp :9999: bind: address already in use
selbes spiel ... kann irgendwie nicht sein. 
Natürlich laufen auf dem server noch andere webserver, aber auf anderen ports.
muss eher an was anderem liegen.

marcsauter75, 2023-02-03:
sonst noch jemand eine Idee woran es liegen könnte  ?

premultiply, 2023-02-03:
evcc halt nicht mehrfach starten...

marcsauter75, 2023-02-03:
@premultiply du bis ja ein fux ... schon mal gelesen ... 
Es ist nicht mehrfach am laufen, kein prozess mehr da mit evcc nur einer ...
und auch wenn ich in der config andere ports wähle immer das gleiche ..

premultiply, 2023-02-03:
Die Meldung ist eindeutig.

marcsauter75, 2023-02-03:
röschtösch und entsteht wenn man nicht stopt bevor man es mit debug laufen lässt 
wenn ich es stoppe und dann dann mit debug ausführe ist die Meldung WEG ... aber die anderen Fehler noch da... 

ich hab die ganze config jetzt weg geworfen und über configure neu gemacht - weil wenn man das yaml von hand anpasst bits jedes mal stress. 
Hoffe es kommt bald mal ne webconfig - was sich ja viele wünschen.

andig, 2023-02-03:
@marcsauter75 es hilft immer vom Allgemeinen ins Konkrete zu gehen wenn Du Fehler beheben willst.

> wenn ich es stoppe und dann dann mit debug ausführe ist die Meldung WEG ... aber die anderen Fehler noch da...

Ich vermute, Du meinst:

> cannot create charger 'shelly': cannot create charger 'template': cannot create charger 'shelly':  (ECFABC6A619A) missing user/password

Die Meldung ist in meinen Augen 100%ig klar. Was ist das Problem damit?