evcc-io/evcc #6018: Km Stand in Session Sicht zeigt zu viele Nachkommastellen

at4hawo1, 2023-02-03:
### Describe the bug

Siehe Schreenshot. Version 0.112.0
[image]

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
-
```

### Log details

```text
-
```

### What type of operating system are you running?

Linux

andig, 2023-02-03:
Steht der gleiche Wert beim Export im csv?

at4hawo1, 2023-02-03:
Nein. Das csv schaut gut aus.
[image]

andig, 2023-02-03:
Closed in https://github.com/evcc-io/evcc/commit/c150b2e0cccab3e09bd89f780e77dcea2294f8d2

at4hawo1, 2023-02-26:
Problem besteht in 0.113.x leider immer noch.

andig, 2023-02-26:
Der älteste Go Fehler überhaupt. Fixed in https://github.com/evcc-io/evcc/commit/48931c4a650d4bcff214db0ad4f1b25201727482

closed 2023-02-26 by andig