evcc-io/evcc #6019: Support AXIbox wallbox from AXItec

jan-r91, 2023-02-03:
Hallo,
ist es möglich, die AXIbox ([Webseite](https://www.axitecsolar.com/ladestation-von-axitec), [Bedienungsanleitung](https://www.axitecsolar.com/sites/default/files/documents/221129_Manual%20AXIbox%2011K_DE.pdf)) zu integrieren?
Anscheinend handelt es sich dabei um eine gebrandete Variante von [EN+](https://www.en-plustech.com/smart-home-series-ce.html).
Aktuell kann ich die Ladevorgänge über die [ChargeIn App](https://play.google.com/store/apps/details?id=cn.enplus.chargein&hl=de&gl=US) steuern und dort einige Einstellungen, wie Ladestrom und Zeitpläne, vornehmen.
OCPP 1.6 wird auch unterstützt, aber soweit ich sehe, gibt es kein Webinterface oder Ähnliches wo ich das konfigurieren könnte.
Vielen Dank!
Jan

premultiply, 2023-02-03:
AXItec, Azzurro, Saturn, EVSEDO, Univers by Fte, EN+, Entratek, Alpha ESS
such dir einen Brand-Namen für den mutmaßlich chinesischen Hersteller aus ;)

Laut AlphaESS hat die Box einen RS485-Anschluss. Eine passende Protokollbeschreibung konnte ich aber bislang noch nirgend entdecken. Wenn du eine findest oder anfragst und dies z. B. Modbus RTU als Protokoll verwendet ist das eine schnelle und einfache Sache die Box anzusteuern.

premultiply, 2023-02-03:
Duplicate of https://github.com/evcc-io/evcc/discussions/2699

Wenn es eine Protokollbeschreibung gibt gerne nochmal melden.
Bis dahin mache ich hier mal zu.

closed 2023-02-03 by premultiply

Allcrafter1, 2024-08-22:
OCPP kann wie gesagt nicht nativ Konfiguriert werden. Bei einem Anruf bei Axitec wurde mir gesagt dass mir die Anleitung dafür per E-Mail zukommen wird zusammen mit einem in der Fabrik gesetzen Passwort. Eine OCPP integration gibt es für Homeassistant über HACS. Ich berichte dann ob es funktioniert.

mamu0015, 2024-09-27:
Hallo Allcrafter1,

hast Du von Axitec etwas bekommen?
hat das funktioniert?

hab auch grad meine Axibox da liegen und bin etwas ratlos wie ich mit Ihr kommunizieren soll

Grüße

Allcrafter1, 2024-09-27:
Nein leider noch nicht. Es hieß meine Wallbox sei "zu alt" dafür (hab die neu gekauft) und sie schauen aktuell nach einer Lösung. Bisher habe ich diese nicht. Es gibt allerdings ein Projekt hier auf GitHub mit dem man selbst Firmware Flashen kann die das kann. Wenn benötigt schick ich gern den Link rein.

mamu0015, 2024-09-27:
Das wäre super,
Brauche die original Firmware ja nicht wirklich 🙂

mamu0015, 2024-09-27:
Kann es sein, dass die Box einen ESP8266 bzw. ESP32 nutzt?

Allcrafter1, 2024-09-27:
Ja, sie nutzt einen ESP32

mamu0015, 2024-09-27:
Kannst du mir den link bitte noch schicken, finde leider kein Projekt. Vermutlich suche ich mit den falschen Suchbegriffen

Allcrafter1, 2024-09-27:
Na klar. War vorher zu faul ihn rauszusuchen. https://github.com/warp-more-hardware/esp32-firmware

Ist so wie ich das nach dem Schaltplan gesehen habe die selbe Wallbox. Mir ist nur nicht klar ob die Axitec die 8MB oder die 4MB Firmware braucht. Ich habe es mir auch noch nicht genau angeschaut und für mich sieht es kompliziert vom Flashen her aus. Das anschließen ist kein Problem. Und ob es funktioniert keine Ahnung. Hast du schonmal einen ESP32 auf diese Art geflashed? Weil ich hatte bisher immer nicht so gute Erfahrungen und würde ungern die Wallbox ganz Nutzlos machen.

mamu0015, 2024-09-28:
Danke Dir.
Das werde ich mir am Feiertag mal zur Gemüte führen. Esp32 hab ich bisher nur über USB geflasht... Entweder auf den Boards, da ist ja immer ein USB mit drauf oder mit Rs232 umsetzen und angelöteten pins. Hab jetzt nur kurz geschaut, da scheint es ja ein Tool zum over the Air programmieren zu geben.

mamu0015, 2024-10-04:
Nochmal vielen Dank,
das Projekt ist super und sehr zu empfehlen.
Habe die Box mit dem FTDI Programmer mit der vorbereiteten *.bin Firmware für den ESP Controller bespielt. 
Kurzer Check ergab, dass ich schon die beste Firmware für den GD Prozessor drauf habe.
würde aber mittlerweile ohne Programmer einfach über den Browser programmiert werden.
Steuere jetzt die Wallbox aus dem IoBroker mit MQTT.

Allcrafter1, 2024-12-21:
Hey,
Wie hast du das genau gemacht mit dem direkten Flaschen? Welchen FTDI Programmer hast du und wie bzw. mit welcher Software hast du die bin Firmware auf den ESP geflashed? Ich habe das halt noch nie gemacht und möchte nix Falsch machen weil ich ja sonst ne Defekte Wallbox habe.
Wäre super wenn du mir da helfen könntest!

mamu0015, 2024-12-22:
Hi,
hab mehrere FTDI Programmer, weis garnicht mehr welchen ich genutzt hab. aber die sind normalerweise alle gleich gut und für paar EUR zu kriegen.
Übertragen hab ich das ganze mit der esptool software wie hier beschrieben:
https://github.com/warp-more-hardware/esp32-firmware/wiki#flashen-der-alternativen-firmware
der ESP32 mit der neuen Firmware kann dann den GD Chip selbst flashen.
ging alles sehr problemlos.

crazy-ahmet, 2025-11-19:
> Nochmal vielen Dank, das Projekt ist super und sehr zu empfehlen. Habe die Box mit dem FTDI Programmer mit der vorbereiteten *.bin Firmware für den ESP Controller bespielt. Kurzer Check ergab, dass [...]

Hast du die Firmware wirklich auf die Axitec Axibox 11kw bekommen? 
Habe es mit dem Transwarp OTA Tool probiert. Beim Aufspielversuch hat er abgebrochen
Mit einem seriellen USB Flasher konnte ich dann zumindest die 8MB Originalversion wieder zurück spielen.
Mit der kleinen knapp 2MB großen merged Firmware habe ich es aber nicht zum laufen bekommen.
Die Box macht zwar kurz einen Accesspoint auf aber es gibt immer Verbindungsfehler und man bekommt keinen Wifi Connect, ich denke das Programm läuft auf diesem ESP also nicht richtig.