evcc-io/evcc #6024: The webserver does not start with 112.1

RaveGun, 2023-02-03:
### Describe the bug

After downloading the 112.1 the web server does not start.
With the 112.0 it is working fine.

### Steps to reproduce

1. download the lateset AMD64 112.1
2. unpack and run
3. the log stops here:
```
[main  ] INFO 2023/02/03 23:22:22 evcc 0.112.1
[main  ] INFO 2023/02/03 23:22:22 using config file: /root/evcc/evcc.yaml
[main  ] INFO 2023/02/03 23:22:22 starting ui and api at :7070
[db    ] INFO 2023/02/03 23:22:22 using sqlite database: /root/.evcc/evcc.db
[site  ] INFO 2023/02/03 23:22:24 site config:
[site  ] INFO 2023/02/03 23:22:24   meters:      grid ✓ pv ✓ battery ✗
[site  ] INFO 2023/02/03 23:22:24     grid:      power ✓ energy ✗ currents ✗
[site  ] INFO 2023/02/03 23:22:24     pv 1:      power ✓ energy ✗ currents ✗
[site  ] INFO 2023/02/03 23:22:24   vehicles:
[site  ] INFO 2023/02/03 23:22:24     vehicle 1: range ✓ finish ✓ status ✓ climate ✗ wakeup ✓
[lp-1  ] INFO 2023/02/03 23:22:24 loadpoint 1:
[lp-1  ] INFO 2023/02/03 23:22:24   mode:        pv
[lp-1  ] INFO 2023/02/03 23:22:24   charger:     power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
[lp-1  ] INFO 2023/02/03 23:22:24   meters:      charge ✓
[lp-1  ] INFO 2023/02/03 23:22:24     charge:    power ✓ energy ✓ currents ✓
[lp-1  ] INFO 2023/02/03 23:22:24 vehicle updated: unknown -> IQ5
```

With the 112.0 there is an additional line:
`[lp-1  ] INFO 2023/02/03 23:23:13 car disconnected`

### Configuration details

```yaml
normal nothing special, I guess
```

### Log details

```text
[main  ] INFO 2023/02/03 23:30:49 evcc 0.112.1
[main  ] INFO 2023/02/03 23:30:49 using config file: /root/evcc/evcc.yaml
[main  ] INFO 2023/02/03 23:30:49 starting ui and api at :7070
[db    ] INFO 2023/02/03 23:30:50 using sqlite database: /root/.evcc/evcc.db
[site  ] INFO 2023/02/03 23:30:52 site config:
[site  ] INFO 2023/02/03 23:30:52   meters:      grid ✓ pv ✓ battery ✗
[site  ] INFO 2023/02/03 23:30:52     grid:      power ✓ energy ✗ currents ✗
[site  ] INFO 2023/02/03 23:30:52     pv 1:      power ✓ energy ✗ currents ✗
[site  ] INFO 2023/02/03 23:30:52   vehicles:
[site  ] INFO 2023/02/03 23:30:52     vehicle 1: range ✓ finish ✓ status ✓ climate ✗ wakeup ✓
[lp-1  ] INFO 2023/02/03 23:30:52 loadpoint 1:
[lp-1  ] INFO 2023/02/03 23:30:52   mode:        pv
[lp-1  ] INFO 2023/02/03 23:30:52   charger:     power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
[lp-1  ] INFO 2023/02/03 23:30:52   meters:      charge ✓
[lp-1  ] INFO 2023/02/03 23:30:52     charge:    power ✓ energy ✓ currents ✓
[lp-1  ] DEBUG 2023/02/03 23:30:52 phase timer inactive
[lp-1  ] DEBUG 2023/02/03 23:30:52 pv timer inactive
[lp-1  ] INFO 2023/02/03 23:30:52 vehicle updated: unknown -> IQ5
```

### What type of operating system are you running?

Linux

### Version

112.1

VolkerK62, 2023-02-04:
ist bei mir auch. hängt bestimmt mit #6025 zusammen

schuepc, 2023-02-04:
Ich musste die Influx DB abhängen, dann ging der Webserver so halbwegs wieder. Das Fahrzeug wird aber auch nicht Initialisiert.

andig, 2023-02-04:
Fixed in https://github.com/evcc-io/evcc/commit/1a5e34d23d7e74646534b851e969856b92ad37a8

closed 2023-02-04 by andig