evcc-io/evcc #6026: (112.1 stable) JSON error

princemaxwell, 2023-02-04:
Failed to parse web socket data: JSON.parse: unexpected character at line 1 column 2112 of the JSON data [{"loadpoints.0.connectedDuration":0,"loadpoints.0.charging":false,"loadpoints.1.chargeCurrent":0,"loadpoints.1.planActive":false,"bufferSoc":0,"loadpoints.0.vehiclePresent":true,"loadpoints.0.vehicleTitle":"Nissan Leaf 21E","loadpoints.1.phasesActive":1,"loadpoints.1.chargePower":3,"gridPowers":[61.800000000000004,258.90000000000003,305.90000000000003],"gridEnergy":16198,"prioritySoc":0,"loadpoints.0.targetSoc":80,"loadpoints.1.vehicleTargetSoc":0,"loadpoints.0.chargePower":0,"loadpoints.1.vehicleSoc":59,"savingsEffectivePrice":0.24938,"loadpoints.0.phasesEnabled":3,"loadpoints.0.phaseAction":"inactive","loadpoints.1.minCurrent":6,"loadpoints.1.vehicleCapacity":17.7,"loadpoints.1.enabled":false,"loadpoints.0.phasesConfigured":null,"loadpoints.1.phaseRemaining":0,"loadpoints.1.targetEnergy":0,"savingsTotalCharged":153.72,"residualPower":0,"loadpoints.1.pvAction":"inactive","homePower":623.7,"savingsAmount":0.082453,"loadpoints.0.maxCurrent":16,"loadpoints.0.vehicleSoc":62,"loadpoints.0.vehicleFeatureOffline":false,"tariffFeedIn":0.1004,"savingsSelfConsumptionPercent":0.30006,"loadpoints.1.chargedEnergy":0,"loadpoints.0.pvRemaining":0,"loadpoints.1.phaseAction":"inactive","auth":{"vehicles":{}},"tariffGrid":0.2353,"loadpoints.0.vehicleRange":145,"loadpoints.0.chargeDuration":0,"loadpoints.1.targetTime":"","loadpoints.1.connected":false,"loadpoints.0.vehicleIcon":"car","loadpoints.1.pvRemaining":0,"loadpoints.1.vehicleRange":54,"loadpoints.0.chargedEnergy":16932,"loadpoints.1.vehicleFeatureOffline":false,"savingsGridCharged":153.26,"loadpoints.1.chargeDuration":0,"loadpoints.1.charging":false,"batteryConfigured":false,"loadpoints.0.pvAction":"inactive","loadpoints.1.phasesConfigured":null,"loadpoints.1.vehicleIcon":"car","loadpoints.1.connectedDuration":9223372036,"currency":"EUR","gridCurrents":[0.982,2.474,2.117],"loadpoints.1.chargeCurrents":[0.049,0.018,0.018],"loadpoints.1.chargeTotalImport":527.4,"loadpoints.0.mode":"pv","loadpoints.1.mode":"pv","loadpoints.0.enabled":false,"loadpoints.0.planActive":false,"siteTitle":"This Home","gridConfigured":true,"loadpoints.0.title":,"loadpoints.0.phasesActive":1,"loadpoints.1.chargeVoltages":[233,234,234],"pvPower":0,"gridPower":626.7,"savingsSelfConsumptionCharged":0.46127,"loadpoints.0.vehicleCapacity":40,"loadpoints.0.targetEnergy":0,"loadpoints.1.phasesEnabled":3,"version":"0.112.1","pv":[{"power":0}],"loadpoints.1.chargeRemainingEnergy":4130,"savingsSince":"2023-01-26T08:55:39+01:00","vehicles":["Nissan Leaf 16E","Nissan Leaf 21E","Smart EQ fortwo"],"loadpoints.0.minSoc":0,"loadpoints.1.vehicleDetectionActive":false,"sponsor":"princemaxwell","loadpoints.0.chargeCurrent":0,"loadpoints.0.chargeRemainingEnergy":8000,"loadpoints.0.phaseRemaining":0,"loadpoints.1.minSoc":0,"loadpoints.1.targetSoc":80,"loadpoints.1.vehiclePresent":true,"loadpoints.1.title":,"loadpoints.1.maxCurrent":16,"loadpoints.1.vehicleTitle":"Smart EQ fortwo","loadpoints.0.connected":true,"pvConfigured":true,"loadpoints.0.minCurrent":6,"loadpoints.0.vehicleTargetSoc":0,"loadpoints.0.vehicleDetectionActive":false,"loadpoints.1.vehicleOdometer":2978}]

Failed to parse web socket data: JSON.parse: unexpected character at line 1 column 1380 of the JSON data [{"loadpoints.0.chargeDuration":0,"savingsSelfConsumptionCharged":0.46127,"loadpoints.1.chargeTotalImport":527.4,"loadpoints.1.enabled":false,"loadpoints.0.phaseRemaining":0,"loadpoints.0.vehicleDetectionActive":false,"loadpoints.1.phasesEnabled":3,"pv":[{"power":0}],"loadpoints.0.maxCurrent":16,"loadpoints.1.minCurrent":6,"loadpoints.0.chargePower":0,"loadpoints.1.vehiclePresent":true,"savingsAmount":0.082453,"loadpoints.1.targetTime":"","pvPower":0,"loadpoints.0.connected":true,"loadpoints.1.chargeCurrent":0,"residualPower":0,"loadpoints.0.vehicleIcon":"car","loadpoints.1.maxCurrent":16,"loadpoints.1.vehicleOdometer":2978,"loadpoints.0.planActive":false,"homePower":616.4,"tariffFeedIn":0.1004,"loadpoints.0.vehicleCapacity":40,"loadpoints.1.phasesActive":1,"loadpoints.1.phaseAction":"inactive","gridEnergy":16198,"loadpoints.0.connectedDuration":0,"batteryConfigured":false,"savingsSince":"2023-01-26T08:55:39+01:00","loadpoints.0.phaseAction":"inactive","loadpoints.1.vehicleIcon":"car","loadpoints.0.charging":false,"vehicles":["Nissan Leaf 16E","Nissan Leaf 21E","Smart EQ fortwo"],"loadpoints.0.vehicleTitle":"Nissan Leaf 21E","loadpoints.0.vehicleRange":145,"loadpoints.1.vehicleRange":54,"savingsSelfConsumptionPercent":0.30006,"loadpoints.1.connectedDuration":9223372036,"loadpoints.1.chargeVoltages":[232,234,234],"loadpoints.0.mode":"pv","loadpoints.1.title":,"loadpoints.1.phaseRemaining":0,"loadpoints.0.vehiclePresent":true,"loadpoints.1.targetSoc":80,"loadpoints.1.vehicleTitle":"Smart EQ fortwo","sponsor":"princemaxwell","gridCurrents":[0.999,2.487,2.036],"loadpoints.0.minCurrent":6,"loadpoints.0.phasesActive":1,"loadpoints.0.minSoc":0,"loadpoints.0.chargeCurrent":0,"loadpoints.0.chargedEnergy":16932,"loadpoints.1.vehicleFeatureOffline":false,"loadpoints.1.chargeDuration":0,"loadpoints.1.chargeRemainingEnergy":4130,"gridConfigured":true,"loadpoints.0.vehicleSoc":62,"loadpoints.1.phasesConfigured":null,"loadpoints.1.vehicleSoc":59,"loadpoints.1.vehicleTargetSoc":0,"version":"0.112.1","loadpoints.1.charging":false,"loadpoints.1.chargeCurrents":[0.045,0.018,0.018],"loadpoints.0.pvAction":"inactive","loadpoints.0.targetSoc":80,"loadpoints.1.pvAction":"inactive","loadpoints.0.vehicleTargetSoc":0,"gridPowers":[72.3,263.40000000000003,283.7],"loadpoints.1.connected":false,"loadpoints.0.phasesConfigured":null,"loadpoints.0.phasesEnabled":3,"loadpoints.0.targetEnergy":0,"loadpoints.1.vehicleCapacity":17.7,"loadpoints.1.mode":"pv","gridPower":619.4,"loadpoints.0.enabled":false,"tariffGrid":0.2353,"loadpoints.0.title":,"loadpoints.0.pvRemaining":0,"loadpoints.1.minSoc":0,"loadpoints.0.chargeRemainingEnergy":8000,"loadpoints.1.chargePower":3,"loadpoints.0.vehicleFeatureOffline":false,"loadpoints.1.vehicleDetectionActive":false,"savingsTotalCharged":153.73,"siteTitle":"This Home","currency":"EUR","auth":{"vehicles":{}},"loadpoints.1.pvRemaining":0,"loadpoints.1.targetEnergy":0,"savingsGridCharged":153.26,"savingsEffectivePrice":0.24938,"loadpoints.1.chargedEnergy":0,"pvConfigured":true,"bufferSoc":0,"prioritySoc":0,"loadpoints.1.planActive":false}]

Failed to parse web socket data: JSON.parse: unexpected character at line 1 column 1160 of the JSON data [{"loadpoints.0.maxCurrent":16,"loadpoints.0.vehicleSoc":62,"loadpoints.0.vehicleFeatureOffline":false,"tariffFeedIn":0.1004,"savingsAmount":0.082453,"loadpoints.0.pvRemaining":0,"loadpoints.1.phaseAction":"inactive","auth":{"vehicles":{}},"tariffGrid":0.2353,"savingsSelfConsumptionPercent":0.30006,"loadpoints.1.chargedEnergy":0,"loadpoints.0.vehicleRange":145,"loadpoints.0.chargeDuration":0,"loadpoints.1.targetTime":"","loadpoints.1.connected":false,"loadpoints.0.vehicleIcon":"car","loadpoints.1.pvRemaining":0,"loadpoints.1.vehicleRange":54,"loadpoints.0.chargedEnergy":16932,"loadpoints.1.chargeDuration":0,"loadpoints.1.charging":false,"batteryConfigured":false,"loadpoints.0.pvAction":"inactive","loadpoints.1.phasesConfigured":null,"loadpoints.1.vehicleIcon":"car","loadpoints.1.vehicleFeatureOffline":false,"savingsGridCharged":153.26,"currency":"EUR","gridCurrents":[0.981,2.446,2.109],"loadpoints.1.chargeCurrents":[0.045,0.018,0.018],"loadpoints.1.chargeTotalImport":527.4,"loadpoints.1.connectedDuration":9223372036,"loadpoints.0.enabled":false,"loadpoints.0.planActive":false,"siteTitle":"This Home","gridConfigured":true,"loadpoints.0.title":,"loadpoints.0.phasesActive":1,"loadpoints.0.mode":"pv","loadpoints.1.mode":"pv","loadpoints.1.chargeVoltages":[233,234,235],"savingsSelfConsumptionCharged":0.46127,"loadpoints.0.vehicleCapacity":40,"loadpoints.0.targetEnergy":0,"loadpoints.1.phasesEnabled":3,"version":"0.112.1","pvPower":0,"gridPower":613.3,"savingsSince":"2023-01-26T08:55:39+01:00","vehicles":["Nissan Leaf 16E","Nissan Leaf 21E","Smart EQ fortwo"],"loadpoints.0.minSoc":0,"loadpoints.1.vehicleDetectionActive":false,"pv":[{"power":0}],"loadpoints.1.chargeRemainingEnergy":4130,"loadpoints.0.chargeRemainingEnergy":8000,"loadpoints.0.phaseRemaining":0,"loadpoints.1.minSoc":0,"loadpoints.1.targetSoc":80,"loadpoints.1.vehiclePresent":true,"sponsor":"princemaxwell","loadpoints.0.chargeCurrent":0,"loadpoints.1.vehicleTitle":"Smart EQ fortw [... cut ...] on":"inactive","auth":{"vehicles":{}},"loadpoints.0.vehicleRange":0,"loadpoints.0.vehicleIcon":"car","loadpoints.1.pvRemaining":0,"loadpoints.1.vehicleRange":0,"loadpoints.1.vehicleFeatureOffline":false,"batteryConfigured":false,"loadpoints.0.pvAction":"inactive","loadpoints.1.phasesConfigured":null,"loadpoints.1.vehicleIcon":"car","currency":"EUR","loadpoints.0.mode":"pv","loadpoints.1.mode":"pv","siteTitle":"This Home","gridConfigured":true,"loadpoints.0.title":,"loadpoints.0.phasesActive":1}]

panzerdev, 2023-02-04:
#6025 sollte das Problem beheben.

andig, 2023-02-04:
Fixed in https://github.com/evcc-io/evcc/commit/1a5e34d23d7e74646534b851e969856b92ad37a8

closed 2023-02-04 by andig