evcc-io/evcc #6028: Unexpected token nach Updatet auf v0.112.1

wwshaui, 2023-02-04:
### Describe the bug

Guten Morgen, 
haben eben upgdatet, jetzt der Fehler:
Failed to parse web socket data: Unexpected token ',', ...".0.title":,"gridPowe"... is not valid JSON [{"loadpoints.0.pvRemaining":0,"loadpoints.0.vehicleDetectionActive":false,"sponsor":"wwshaui","loadpoints.0.enabled":false,"loadpoints.0.climater":"off","loadpoints.0.pvAction":"inactive","savingsSince":"2022-11-

Kann mir jemand helfen, was mach ich Falsch?
liegts am Sponsor Token, der wird mir so angezeigt.
Danke und Gruß

### Steps to reproduce

1. Update evcc
2. Restart evcc
3. Meldung siehe Log
...

### Configuration details

```yaml
*
```

### Log details

```text
Failed to parse web socket data: Unexpected token ',', ...".0.title":,"gridPowe"... is not valid JSON [{"loadpoints.0.pvRemaining":0,"loadpoints.0.vehicleDetectionActive":false,"sponsor":"wwshaui","loadpoints.0.enabled":false,"loadpoints.0.climater":"off","loadpoints.0.pvAction":"inactive","savingsSince":"2022-11-.............................................
```

### What type of operating system are you running?

Linux

### Version

v0.112.1

VolkerK62, 2023-02-04:
hängt bestimmt mit #6025 zusammen

wwshaui, 2023-02-04:
Danke aber bin kein Linux Spezialist,
Habe am System nichts geändert, nur updatet durchgeführt.

VolkerK62, 2023-02-04:
ja. und in dem update ist ein bug. abwarten, tee/kaffee trinken. so wie ich andig kenne, wird das in kurzer Zeit behoben sein.

wwshaui, 2023-02-04:
OK, das kann ich, danke 👍

andig, 2023-02-04:
Fixed in https://github.com/evcc-io/evcc/commit/1a5e34d23d7e74646534b851e969856b92ad37a8

closed 2023-02-04 by andig