evcc-io/evcc #6029: Roadmap for UI-based config

andig, 2023-02-04:
UI-based configuration is probably the single biggest game changer that we need to make evcc more approachable for non-technical users. This will be a large change and time-consuming. To keep seeing results we should develop an incremental roadmap that delivers usable results and can be shipped step by step. This issue should structure the work.

### Single, new devices for testing only
Scope: template devices only, no linked config. This can be done standalone without touching bigger parts of the application and will already allow users with demo installation to play with things.
- [ ] configure a new device in the ui
- [ ] test device
- [ ] export configuration as yaml

### More stuff to break into steps
- [ ] site
- [ ] loadpoints
- [ ] tariffs
- [ ] vehicles (these should be pretty easy and not not need be linked to other parts of the config)
- [ ] push messages (this one would be really good if we can add a `Test` button)
- [ ] mqtt
- [ ] influxdb
- [ ] modbus proxy (this seems another good candidate for testing things)
 
Challenges:
- integrate "online" with "offline" config to still allow using complex configs
- make things "pluggable" and implement restart behaviour for all classes of devices

StevieC121176, 2023-02-09:
wäre es nicht besser, den erstelle neues Fahrzeug Dialog aus dem Ladelog (ändern Fahrzeug) auszublenden? Ich denke es wechselt niemand in diese Ansicht um ein Fahrzeug zu erstellen

[image]

andig, 2023-02-09:
Ja 🤪

naltatis, 2023-02-11:
Ja, das ist alles noch "exerimental" 😄
Geht hier erstmal darum die Funktionalität an die Oberfläche zu bekommen.

Hofyyy, 2023-04-29:
Auf die Gefahr hin das ihr das schon kennt. Ich bin von genau so einem Dialog bei der Weboberfläche "homebridge" sehr angetan. Speichern führt eine Validierung durch und speichert nur, wenn das Schema passt. Vielleicht kann man sich dran anlehnen, oder es hilft beim abgucken.

[image]

andig, 2024-06-09:
> move site meters to global config, identified by type (grid, pv, battery, aux)

@naltatis das ist nicht so einfach, da nicht alle Zähler einen Typ haben. Usage dient nur dazu, bei integrierten Geräten die richtigen Werten abzugreifen. Sonst spezifiziert erst die Auswahl in Site, wofür der Zähler verwendet werden soll.

naltatis, 2024-06-09:
> @naltatis das ist nicht so einfach, da nicht alle Zähler einen Typ haben. Usage dient nur dazu, bei integrierten Geräten die richtigen Werten abzugreifen. Sonst spezifiziert erst die Auswahl in Site, wofür der Zähler verwendet werden soll.

Das wir dafür eine neue Eigenschaft am Zähler brauchen, hatte ich auf dem Zettel. Vielleicht ein guter Anlass das mit den Meter-Titles https://github.com/evcc-io/evcc/issues/6781 auch anzugehen 😄.

Tombra1889, 2024-06-09:
Wäre echt cool, wenn die AUX-Zähler, bzw. noch besser externe Zähler, die den Hausverbrauch detaillieren in die Ansicht eingebaut werden könnten!

andig, 2025-01-06:
@naltatis how complete are we?

TestMyLox, 2025-05-02:
Just a note: The designation in the web interface for the myPV AC Elwa-E is not consistent.
It's an electric heating element, not a meter. 
Furthermore, "power" probably refers to the current "consumption," and "charge level" refers not to a percentage but to the current water temperature in °C.

[image]

naltatis, 2025-05-02:
@andig how can we detect this on meter level? On loadpoint/chargers-level we already handle this.

[image]

andig, 2025-05-02:
Ouch, we can‘t without goving meters additional meta data :/