evcc-io/evcc #6034: Configured tariffs.planner is blocking Web UI start

goebelmeier, 2023-02-04:
### Describe the bug

After upgrading from 0.111.1 to 0.112.1 the boot process stopped somewhere before initializing web interface and checking which cars are connected toward the charger.
In 0.112.2 the problem persisted.
Removing 
```
planner:
    # planner tariff can be used for target charging if not grid tariff is specified
    # GrünStromIndex (Germany only) or ElectricityMaps provide CO2 intensity forecast

    type: grünstromindex
    zip: 28779
```
from my config.yaml solved the issue and evcc is operating back normal. 

### Steps to reproduce

1. Install evcc newer than 0.111.1
2. Configure tariffs.planner using gruenstromindex
3. Start evcc
...

### Configuration details

```yaml
network:
  # schema is the HTTP schema
  # setting to `https` does not enable https, it only changes the way URLs are generated
  schema: http
  # host is the hostname or IP address
  # if the host name contains a `.local` suffix, the name will be announced on MDNS
  # docker: MDNS announcements don't work. host must be set to the docker host's name.
  host: evcc.local
  # port is the listening port for UI and api
  # evcc will listen on all available interfaces
  port: 7070

interval: 10s # control cycle interval

# sponsor token: *****
sponsortoken: *****

# telemetry enables aggregated statistics
#
# Telemetry allows collecting usage data (grid and green energy, charge power).
# Data is aggregated, no individual charging sessions are tracked. The collected,
# anonymous data can be retrieved using https://api.evcc.io.
#
# See https://github.com/evcc-io/evcc/pull/4343 or details.
#
# For time being, this is only available to sponsors, hence data is associated with
# the sponsor token's identity.
#
telemetry: true

plant: *****

# log settings
log: debug
levels:
  site: debug
  core: debug
  lp-1: debug
  lp-2: debug
  cache: error
  db: error
  vw: debug
  id: debug
  tesla: debug

[... 235 lines cut ...]
[lp-1  ] WARN 2023/02/04 09:56:32 poll interval '30m0s' is lower than 1h0m0s and may deplete your battery or lead to API misuse. USE AT YOUR OWN RISK.
[site  ] INFO 2023/02/04 09:56:32 site config:
[site  ] INFO 2023/02/04 09:56:32   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2023/02/04 09:56:32     grid:      power ✓ energy ✗ currents ✗
[site  ] INFO 2023/02/04 09:56:32     pv 1:      power ✓ energy ✗ currents ✗
[site  ] INFO 2023/02/04 09:56:32     battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✗
[site  ] INFO 2023/02/04 09:56:32   vehicles:
[site  ] INFO 2023/02/04 09:56:32     vehicle 1: range ✓ finish ✓ status ✓ climate ✓ wakeup ✗
[site  ] INFO 2023/02/04 09:56:32     vehicle 2: range ✓ finish ✓ status ✓ climate ✗ wakeup ✗
[lp-1  ] INFO 2023/02/04 09:56:32 loadpoint 1:
[lp-1  ] INFO 2023/02/04 09:56:32   mode:        off
[lp-1  ] INFO 2023/02/04 09:56:32   charger:     power ✓ energy ✓ currents ✓ phases ✗ wakeup ✗
[lp-1  ] INFO 2023/02/04 09:56:32   meters:      charge ✓
[lp-1  ] INFO 2023/02/04 09:56:32     charge:    power ✓ energy ✓ currents ✓
[lp-1  ] DEBUG 2023/02/04 09:56:32 phase timer inactive
[lp-1  ] DEBUG 2023/02/04 09:56:32 pv timer inactive
```

### What type of operating system are you running?

Linux

### Version

0.112.2

goebelmeier, 2023-02-04:
Might not be the only issue, see https://github.com/evcc-io/evcc/discussions/6030#discussioncomment-4868545

mseele, 2023-02-04:
Das selbe bei mir mit tibber:

```YAML
tariffs:
  grid:
    type: tibber
    token: XXX
```

StevieC121176, 2023-02-04:
> Might not be the only issue, see [#6030 (reply in thread)](https://github.com/evcc-io/evcc/discussions/6030#discussioncomment-4868545)

Bei mir liegt es an influx, ohne läuft es

```
#influx:
#  url: http://192.168.178.80:8086
#  database: ****
#  user:****
#  password: ****
```

rediculum, 2023-02-04:
Jap, ebenfalls. Habe kein Planner konfiguriert, aber Influx. Sobald ich es auch kommentiere, funktionert es.

F-L-0, 2023-02-04:
Bei mir ebenfalls nach dem minor upgrade von 0.112.0 auf 0.112.1.
evcc startet, bleibt aber "hängen", der Prozess ist da, es fehlt aber der listen port auf 7070:

`root@rpi4-2:~# netstat -tulpen|grep evcc
udp        0      0 0.0.0.0:5353            0.0.0.0:*                           0          14803789   25780/evcc
udp        0      0 0.0.0.0:9522            0.0.0.0:*                           0          14802340   25780/evcc
udp6       0      0 :::5353                 :::*                                0          14803790   25780/evcc`

Der Prozess steht aber nicht, sowohl evcc --log trace als auch ein strace auf die PID zeigen Aktivität - allerdings nur Kommunikation mit den SMA Wechselrichtern.
Auskommentieren von influx hat bei mir ebenfalls dazu geführt das die 0.112.1 wieder funktioniert.
Hab ein Downgrade auf die 0.112.0 gemacht, die funktioniert - der Feher ist also von 0.112.0 auf 0.112.1 rein gekommen.

andig, 2023-02-05:
Also Ursache dass Influx blockt? Tut mir leid, schau nach dem Frühstück rein.

naltatis, 2023-02-05:
Ja, Influx scheint hier die Ursache zu sein. Seit diesem Commit https://github.com/evcc-io/evcc/commit/1f068d3083b3cd411c10479a87ade30c908d6e4d tritt das Problem auf wenn man influx konfiguriert hat.

closed 2023-02-05 by andig

StevieC121176, 2023-02-05:
auch bei der Version  `0.112.2 (04308d9d)` startet evcc erst wenn ich die influxdb auskommentiere

naltatis, 2023-02-05:
Mit dem Nightly was gerade rausgekommen ist läuft alles wieder 🙌

thierolm, 2023-02-05:
Auf meinem PI hatte sich die Influx DB auch aufgehängt und dann evcc blockiert. Ihr solltet die dann auch durchstarten.

emign, 2023-02-05:
Die Home Assistant Version 0.112.3 startet weiterhin bei mir nicht mit Influx. Wenn ich die Settings auskommentiere, dann schon

andig, 2023-02-05:
@emign hast Du eine Möglichkeit, Dein Setup mal mit VScode im Debugger laufen zu lassen? Dann könnte ich sehen wo der Hänger produziert wird.

emign, 2023-02-05:
> @emign hast Du eine Möglichkeit, Dein Setup mal mit VScode im Debugger laufen zu lassen? Dann könnte ich sehen wo der Hänger produziert wird.

Wenn du mich da etwas an die Hand nehmen kannst, gerne. Denn es gibt weiterhin Probleme. Wenn man den evcc Container dann in Home Assistant neu startet (zum Beispiel für ne Config-Änderung), hängt er wieder (502 Bad Request).

Was dann hilft, ist nur den Zyklus 1-4 aus dem obigen Posting zu durchlaufen.

Also grundsätzlich ist keine Funktionalität nach einem Restart des HA Addons mehr gegeben.

andig, 2023-02-05:
Also:

- Go und Node installieren
- npm ci
- make ui
- Launch config erstellen:

```
    {
      "name": "test",
      "type": "go",
      "request": "launch",
      "mode": "auto",
      "trace": "error",
      "program": "${workspaceFolder}/main.go",
      "args": [
        "--profile",
        "-c",
        "pfad zu deiner.yaml"
      ]
    },
```

- debugger starten und wenn es hängt Pause drücken
- im Callstack nach hängenden "goroutines" schauen und gucken wo die hängt- meist die oberste- und wie sie da hin gekommen ist

Alternativ: remotesession auf Deinem Rechner, gerne mal bei info@evcc.io melden

emign, 2023-02-05:
[image]

emign, 2023-02-05:
[image]

andig, 2023-02-05:
Sehr gut. Ich schau morgen rein.

andig, 2023-02-05:
Könntest du nochmal die beiden letzten Zeilennummern einblenden (Spalte breiter)? Letzter Stand aus git?

emign, 2023-02-05:
[image]

Ich habe einfach den Source Code aus dem Release hier geladen: https://github.com/evcc-io/evcc/releases/tag/0.112.3

andig, 2023-02-05:
Könntest du nochmal master probieren? Entweder git clone oder auch am Stück runter laden?

emign, 2023-02-05:
> Könntest du nochmal master probieren? Entweder git clone oder auch am Stück runter laden?
Funktioniert einwandfrei

andig, 2023-02-05:
Und probier bitte mal in https://github.com/evcc-io/evcc/blob/master/util/tee.go#L15 statt 1 z.b. eine 16 einzusetzen. Denke das reicht.

andig, 2023-02-05:
Master geht? Ok, dann mach ich fix noch ein Release.

emign, 2023-02-05:
ja der geht

andig, 2023-02-05:
Top, danke für die Analyse. Tut mir leid, dachte der Commit hätte es noch mit ins Release geschafft.

emign, 2023-02-05:
Ich danke für die schnelle Behebung. Bis morgen geht es ja wieder.Da scheint hier die Sonne =)

emign, 2023-02-05:
0.112.4 fixed das Problem :)

msome, 2023-02-06:
@andig , ich muss leider melden, dass es auch mit der 0.112.4 immer noch hängt. In Docker & im Debugger.
Es ist das gleiche Phänomen wie oben beschrieben. Wenn influx auskommentiert ist, läuft der Webserver, mit influx nicht.

Download [zip 0.112.4](https://github.com/evcc-io/evcc/archive/refs/tags/0.112.4.zip), ich habe jetzt das oben genannte Debugging gleich mit gemacht.
Note: Zur Sicherheit im Screenshot - das tee.go hatte im Release zip bereits wie gewollt den Wert 1 in Zeile 15.

Könntest du bitte nochmal einen Blick reinwerfen warum es immer noch hängt?

Der erste Abschnitt des Callstacks aufgeklappt:

[image]

Hier noch der komplette angezeigte Callstack:

[image]

Reicht der Screenshot, oder brauchts noch mehr Informationen?

goebelmeier, 2023-02-06:
> @andig , ich muss leider melden, dass es auch mit der 0.112.4 immer noch hängt.

Can confirm, ist mir heute Nachmittag auch aufgefallen.

thierolm, 2023-02-06:
Jepp, ich bekomme als letzte Version nur die 0.112.0 in docker zum laufen. Ab 0.112.1 bis 0.112.3 bleiben alle hängen. :-(

andig, 2023-02-06:
Sollte per Nightly jetzt behoben sein.

msome, 2023-02-06:
Die aktuelle nightly vom Docker [ laut log: evcc 0.112.4 (7faa8bb8) ] behebt das Problem nicht.
Auch der Master-Branch von heute 20:38 Uhr zeigt noch immer kein Web-UI.

Den Master hab ich im Debugger, dort ist der Tee Attach Parameter [wieder auf 16](https://github.com/evcc-io/evcc/commit/7faa8bb8e419da00d5008ae0e3be437258d5e7f8).

emign, 2023-02-06:
Bei mir läuft es auf HA und lokal.

msome, 2023-02-06:
Ich habe eben bemerkt, bei mir laufen die neuen Versionen mit Influx, wenn ich **nur einen Loadpoint** konfiguriere. 
Sobald ich den zweiten Loadpoint wieder in die Konfig aufnehme, startet es nicht mehr.

andig, 2023-02-06:
Probier bitte die .5

thierolm, 2023-02-06:
Das Docker Nightly funktioniert bei mir ohne Hänger und Error mit 3 Loadpoints.
Ich probier gleich die .5 ...

msome, 2023-02-06:
Die .5 läuft mit beiden LPs und Influx. 
Der Docker Container geht damit auch wieder auf "healthy". Perfekt.

Vielen Dank.

thierolm, 2023-02-06:
Jo, kann ich auch bestätigen. Nice! :-)