evcc-io/evcc #6038: Failed to parse web socket data: Unexpected token

VitaliKaiser, 2023-02-04:
### Describe the bug

I just updated to evcc 0.112.1 (I think I had 1.11 before) and now I get the following error message at the top left of the UI interface:

Failed to parse web socket data: Unexpected token ',', ...".0.title":,"loadpoin"... is not valid JSON [{"siteTitle":"Zuhause","loadpoints.0.chargePower":4140,"loadpoints.0.chargedEnergy":1706.7,"loadpoints.0.chargeDuration":1464,"loadpoints.0.connectedDuration":0,"tariffFeedIn":0.08,"loadpoints.0.pvAction":"inactive","loadpoints.0.targetSoc":100,"loadpoints.0.vehicleCapacity":64,"savingsEffectivePrice":0.2369,"loadpoints.0.maxCurrent":16,"loadpoints.0.chargeCurrent":6,"loadpoints.0.vehicleRange":23,"savingsSelfConsumptionPercent":28.681,"bufferSoc":0,"loadpoints.0.minCurrent":6,"loadpoints.0.phasesEnabled":3,"loadpoints.0.phaseRemaining":0,"auth":{"vehicles":{}},"releaseNotes":"\u003ch1\u003e0.112.2\u003c/h1\u003e\n\u003ch2 dir=\"auto\"\u003eChangelog\u003c/h2\u003e\n\u003cul dir=\"auto\"\u003e\n\u003cli\u003e\u003ca class=\"commit-link\" data-hovercard-type=\"commit\" data-hovercard-url=\"https://github.com/evcc-io/evcc/commit/c150b2e0cccab3e09bd89f780e77dcea2294f8d2/hovercard\" href=\"https://github.com/evcc-io/evcc/commit/c150b2e0cccab3e09bd89f780e77dcea2294f8d2\"\u003e\u003ctt\u003ec150b2e\u003c/tt\u003e\u003c/a\u003e Sessions: round odometer\u003c/li\u003e\n\u003c/ul\u003e","savingsAmount":19.992,"residualPower":0,"loadpoints.0.pvRemaining":0,"loadpoints.0.vehicleSoc":7,"prioritySoc":0,"loadpoints.0.enabled":true,"loadpoints.0.targetEnergy":0,"gridConfigured":true,"version":"0.112.1","loadpoints.0.vehicleIcon":"car","loadpoints.0.vehicleFeatureOffline":false,"savingsSelfConsumptionCharged":90.873,"pvConfigured":true,"pvPower":3415,"loadpoints.0.charging":true,"loadpoints.0.title":,"loadpoints.0.vehiclePresent":true,"loadpoints.0.planActive":false,"loadpoints.0.chargeRemainingDuration":56918,"vehicles":["Kona"],"loadpoints.0.phasesActive":3,"loadpoints.0.vehicleDetectionActive":false,"availableVersion":"0.112.2","savingsTotalCharged":316.84,"loadpoints.0.vehicleOdometer":8503,"loadpoints.0.connected":true,"loadpoints.0.vehicleTitle":"Kona","loadpoints.0.vehicleTargetSoc":100,"loadpoints.0.chargeRemainingEnergy":65455,"tariffGrid":0.3,"savingsSince":"2022-11-14T05:54:05Z","loadpoints.0.phasesConfigured":null,"gridPower":780,"batteryConfigured":false,"loadpoints.0.phaseAction":"inactive","loadpoints.0.minSoc":0,"savingsGridCharged":225.97,"currency":"EUR","loadpoints.0.mode":"minpv","pv":[{"power":3415}],"homePower":55}]

And also my car, a Hyundai Kona (Bluelink) is not recognized anymore :(

### Steps to reproduce

1. Go to the web interface and click on the left

### Configuration details

```yaml
Will provide if needed. If not needed for this issue, I would be happy not to share my config in the internet :D
```

### Log details

```text
Currently charging my car, so I dont want to restart evcc. If this is needed, give me shot.
```

### What type of operating system are you running?

Linux

### Version

evcc -v

tbe2013, 2023-02-04:
I can confirm the issue, same problem.

Michel83, 2023-02-04:
I can confirm the same issue.

closed 2023-02-04 by andig