evcc-io/evcc #6044: Zielladen pausiert mitten im Ladevorgang

princemaxwell, 2023-02-04:
### Describe the bug

Mir ist aufgefallen, dass mitten im Ziellade-Vorgang die Ladung gestoppt wird, obwohl dieser bereits gestartet war.

Ich habe Messaging per NTFY aktiviert und bekomme über jeden Vorgang eine Nachricht, habe dann das Web UI geöffnet und gesehen, dass bei dem Fahrzeug eine neue Anfangszeit für das Zielladen gesetzt wurde.

Vermutlich wird die Zielladen Anfangszeit mitten im Vorgang neu berechnet und neu gesetzt, was dann zu einer Pause führt.

Folgende Messaging Variablen werden genutzt:
Füllstand: `{{.vehicleSoc}}`
Ziel: `{{.targetSoc}}` / `{{round (divf .chargeRemainingEnergy 1000) 2 }} kWh`
geladen: `{{round (divf .chargedEnergy 1000) 2 }} kWh` in `{{.chargeDuration}}`

**Smart EQ fortwo**
2023-02-02 23:38:01 START Ladevorgang (Füllstand: 40%, Ziel: 80% / =7,87 kWh)
2023-02-03 00:09:01 ENDE Ladevorgang (Füllstand: 51%, geladen: 1,74 kWh in 31m17s)
2023-02-03 00:37:41 START Ladevorgang (Füllstand: 51%, Ziel: 80% / =4,57 kWh)
2023-02-03 02:06:41 ENDE Ladevorgang (Füllstand: 80%, geladen: 6,79 kWh in 2h28m57s)

**Nissan Leaf**
2023-02-02 23:42:11 START Ladevorgang (Füllstand: 32%, Ziel: 80% / =20,89 kWh)
2023-02-03 00:14:11 ENDE Ladevorgang (Füllstand: 39%, geladen: 1,85 kWh in 32m0s)
2023-02-03 00:30:31 START Ladevorgang (Füllstand: 39%, Ziel: 80% / =18,22 kWh)
2023-02-03 04:27:51 ENDE Ladevorgang (Füllstand: 80%, geladen: 15,79 kWh in 4h29m20s)

2023-02-03 21:01:10 START Ladevorgang (Füllstand: 12%, Ziel: 80% / =29,78 kWh)
2023-02-03 00:00:30 ENDE Ladevorgang (Füllstand: 42%, geladen: 10,58 kWh in 2h59m20s)
2023-02-03 01:01:10 START Ladevorgang (Füllstand: 42%, Ziel: 80% / =16,51 kWh)

Das Problem tritt seit 02.02.2023 auf, vorher lief es immer ohne Unterbrechungen.
Kann [https://github.com/evcc-io/evcc/pull/5973](url) was damit zu tun haben??

### Steps to reproduce

Auto relativ leer, Zielladen aktivieren, beim Laden UI beobachten.

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  site: debug
#  lp-1: debug
#  lp-2: debug
#  sma: debug
#  nissan: debug
#  smart: debug
#  cache: error

# unique installation id
plant: *****

interval: 10s # control cycle interval

sponsortoken: *****

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

meters:
- type: template
  template: sma-home-manager
  usage: grid
[... 447 lines cut ...]
[lp-1  ] DEBUG 2023/02/04 13:22:38 next soc poll remaining time: 58m0s
[lp-1  ] DEBUG 2023/02/04 13:22:38 targetSoc reached: 81.0% > 80%
[site  ] DEBUG 2023/02/04 13:22:48 ----
[lp-1  ] DEBUG 2023/02/04 13:22:48 charge power: 0W
[lp-2  ] DEBUG 2023/02/04 13:22:48 charge power: 3W
[site  ] DEBUG 2023/02/04 13:22:48 pv power: 3099W
[site  ] DEBUG 2023/02/04 13:22:48 grid power: -2143W
[site  ] DEBUG 2023/02/04 13:22:48 grid powers: [-968 -478 -698]W
[site  ] DEBUG 2023/02/04 13:22:48 grid currents: [-4.3 -3.14 -3.44]A
[site  ] DEBUG 2023/02/04 13:22:48 site power: -2143W
[lp-2  ] DEBUG 2023/02/04 13:22:48 charge voltages: [230 232 233]V
[lp-2  ] DEBUG 2023/02/04 13:22:48 detected connected phases: 3p
[lp-2  ] DEBUG 2023/02/04 13:22:48 charge currents: [0.049 0.018 0.018]A
[lp-2  ] DEBUG 2023/02/04 13:22:48 charger status: A
[lp-2  ] DEBUG 2023/02/04 13:22:48 next soc poll remaining time: 57m41s
[lp-2  ] WARN 2023/02/04 13:22:48 charger out of sync: expected disabled, got enabled
```

### What type of operating system are you running?

Linux

### Version

0.112.2 (1a5e34d2)

andig, 2023-02-05:
Du hast einen Tarif mit mehreren Zonen. Es kann also gut sein, dass da gestückelt werden muss. Ob das so ist siehst Du im Ladeplan wenn Du Zielladen aktivierst. Hier im Logfile ist das nicht zu sehen.

Da scheint allerdings einiges krumm zu sein:

> charger out of sync: expected disabled, got enabled

Das ist aber ein anderes Thema...

andig, 2023-02-05:
> Kann https://github.com/evcc-io/evcc/pull/5973 was damit zu tun haben??

Möglich ist alles, ohne Logfile aber nicht zu erkennen.

princemaxwell, 2023-02-05:
@andig 

Erstmal vorab, `charger out of sync: expected disabled, got enabled` betrifft nur eine Wallbox, dazu gibt es ne Diskussion. 

**Kannst Du mir bitte sagen, wie ich ein Log von den Ladevorgängen erstelle oder wo ich das finde?**

Das beschriebene Verhalten mit den Ladepausen tritt an beiden Boxen auf.
Der Tarif kann eigentlich nichts damit zu tun haben, der ist nur Mo-Fr von 6-20 anders.
Letzte Nacht ist es schon wieder passiert.

**Nissan Leaf**
2023-02-04 22:22:59 START Ladevorgang (Füllstand: 73%, Ziel: 100% / =12 kWh)
2023-02-05 00:00:39 ENDE Ladevorgang (Füllstand: 87%, geladen: 5,79 kWh in 1h37m40s)
2023-02-05 00:23:39 START Ladevorgang (Füllstand: 87%, Ziel: 100% / =5,43 kWh)
2023-02-05 01:47:39 ENDE Ladevorgang (Füllstand: 100%, geladen: 10,42 kWh in 3h1m40s)
2023-02-05 10:48:59 START Ladevorgang (Füllstand: 100%, Ziel: 100% / =0 kWh)
2023-02-05 10:53:19 ENDE Ladevorgang (Füllstand: 98%, geladen: 10,52 kWh in 3h6m0s)

**Hab um 00:01 Uhr heute Nacht mal in die UI reingeguckt, da wurde Zielladen neu geplant mit Restart um 00:23 Uhr.**

[image]

andig, 2023-02-05:
Logging ist in der Doku beschrieben

princemaxwell, 2023-02-05:
@andig 
Vielen Dank, den Part der Doku hatte ich bisher nicht. Aber ich habe die Suchfunktion in der Doku gefunden ;-)

Hab jetzt mal die letzte Nacht rausgezogen:

> **Nissan Leaf**
> 2023-02-04 22:22:59 START Ladevorgang (Füllstand: 73%, Ziel: 100% / =12 kWh)
> 2023-02-05 00:00:39 ENDE Ladevorgang (Füllstand: 87%, geladen: 5,79 kWh in 1h37m40s)
> 2023-02-05 00:23:39 [...]

[zielladen.txt](https://github.com/evcc-io/evcc/files/10613975/zielladen.txt)

Sieht mir so aus, als ob das hier das Laden stoppt und dann Zielladen auf eine neue Zeit gesetzt wird:
`Feb 05 00:00:18 evcc evcc[1008]: [lp-1  ] DEBUG 2023/02/05 00:00:18 site power 4356W >= 1000W disable threshold`

Würde mich freuen, wenn Du Dir das Log mal anschaust...DANKE!!

princemaxwell, 2023-02-05:
@andig 
Hab heute vorm Laden noch ein Update auf 0.112.4 gemacht.
Leider wieder Abbruch heute, siehe Logging Nr.2.

Heute ist direkt um kurz nach 0:00 Uhr ein Eintrag auffällig:
`Feb 06 00:00:18 evcc evcc[463]: [lp-1  ] DEBUG 2023/02/06 00:00:18 plan will re-start shortly, continuing for remaining 8m0s`

Die dort angezeigt Remaining Zeit wird in den folgenden Logs mal mehr mal weniger und wieder umgekehrt.

Um 0:28 dann noch einmal:
`Feb 06 00:28:19 evcc evcc[463]: [lp-1  ] DEBUG 2023/02/06 00:28:19 plan will re-start shortly, continuing for remaining 9m2s`

und kurz drauf:
```
Feb 06 00:28:38 evcc evcc[463]: [lp-1  ] DEBUG 2023/02/06 00:28:38 pv charge current: 0A = 16A + -18.6A (4268W @ 1p)
Feb 06 00:28:38 evcc evcc[463]: [lp-1  ] DEBUG 2023/02/06 00:28:38 site power 4268W >= 1000W disable threshold
Feb 06 00:28:38 evcc evcc[463]: [lp-1  ] DEBUG 2023/02/06 00:28:38 pv disable in 0s
Feb 06 00:28:38 evcc evcc[463]: [lp-1  ] DEBUG 2023/02/06 00:28:38 pv disable timer elapsed
Feb 06 00:28:38 evcc evcc[463]: [lp-1  ] DEBUG 2023/02/06 00:28:38 charger disable
Feb 06 00:28:38 evcc evcc[463]: [lp-1  ] DEBUG 2023/02/06 00:28:38 wake-up timer: stop
Feb 06 00:28:48 evcc evcc[463]: [site  ] DEBUG 2023/02/06 00:28:48 ----
```

[zielladen2.txt](https://github.com/evcc-io/evcc/files/10614117/zielladen2.txt)

andig, 2023-02-06:
Ich verstehe lnicht was hier die Frage ist. Du hast einen Plan von 3h Dauer. Der ist nachts. Was hat der Plan jetzt mit Ladepausen am Tag zu tun? Könntest du bitte mal ein zusammenhängendes Problem beschreiben? Mir fehlt leider die Zeit, mir das Problem selbst aus den Logfiles zu suchen. Was funktioniert nicht so, wie du es erwarten würdest? Woher kommt diese Erwartung in Anbetracht des Ladeplanes?

princemaxwell, 2023-02-06:
@andig 
Mit dem Bild des Ladeplans in meinem vorigen Post wollte ich nur veranschaulichen, dass ich keine tarifabhängigen Unterbrechungen habe, wie von Dir vermutet:
> Du hast einen Tarif mit mehreren Zonen. Es kann also gut sein, dass da gestückelt werden muss.

Die Ladepausen sind nicht am Tag sondern mitten im Ladeplan nachts. 
Nehmen wir das Beispiel von letzter Nacht:
Der Wagen wurde irgendwann abends angeschlossen (mit 59% Ladung und 80% Ziel) und das Zielladen auf 2:30 Uhr gestellt.
Start des Ladevorgangs war gegen 23:40, alles soweit normal.
```
Feb 05 23:59:58 evcc evcc[463]: [lp-1  ] DEBUG 2023/02/05 23:59:58 charger status: C
Feb 05 23:59:58 evcc evcc[463]: [lp-1  ] DEBUG 2023/02/05 23:59:58 next soc poll remaining time: 59m40s
Feb 05 23:59:58 evcc evcc[463]: [lp-1  ] DEBUG 2023/02/05 23:59:58 soc estimated: 62.36% (vehicle: 62.00%)
Feb 05 23:59:58 evcc evcc[463]: [lp-1  ] DEBUG 2023/02/05 23:59:58 vehicle soc: 62%
Feb 05 23:59:58 evcc evcc[463]: [lp-1  ] DEBUG 2023/02/05 23:59:58 vehicle range: 145km
Feb 05 23:59:58 evcc evcc[463]: [lp-1  ] DEBUG 2023/02/05 23:59:58 planned 2h22m3s until 2023-02-06 02:30:00 +0100 CET at 3680W: total plan duration: 2h22m3s, avg cost: 0.235
Feb 05 23:59:58 evcc evcc[463]: [lp-1  ] DEBUG 2023/02/05 23:59:58 continuing until end of slot
```

Kurz nach 0:00 steht im Log plötzlich:
```
Feb 06 00:00:18 evcc evcc[463]: [lp-1  ] DEBUG 2023/02/06 00:00:18 planned 2h21m42s until 2023-02-06 02:30:00 +0100 CET at 3680W: total plan duration: 2h21m42s, avg cost: 0.235
Feb 06 00:00:18 evcc evcc[463]: [lp-1  ] DEBUG 2023/02/06 00:00:18 plan will re-start shortly, continuing for remaining 8m0s
```

Das Laden läuft noch weiter bis 0:28 Uhr und dann stoppt der Ladevorgang, obwohl das Ziel 80% nicht erreicht ist:
```
Feb 06 00:28:38 evcc evcc[463]: [lp-1  ] DEBUG 2023/02/06 00:28:38 charge power: 3713W
Feb 06 00:28:38 evcc evcc[463]: [site  ] DEBUG 2023/02/06 00:28:38 pv power: 0W
Feb 06 00:28:38 evcc evcc[463]: [site  ] DEBUG 2023/02/06 00:28:38 grid power: 4268W
Feb 06 00:28:38 evcc evcc[463]: [site  ] DEBUG 2023/02/06 00:28:38 grid powers: [69 249 3950]W
Feb 06 00:28:38 evcc evcc[463]: [site  ] DEBUG 2023/02/06 00:28:38 grid currents: [0.997 2.44 17.3]A
Feb 06 00:28:38 evcc evcc[463]: [site  ] DEBUG 2023/02/06 00:28:38 site power: 4268W
Feb 06 00:28:38 evcc evcc[463]: [lp-1  ] DEBUG 2023/02/06 00:28:38 charger status: C
Feb 06 00:28:38 evcc evcc[463]: [lp-1  ] DEBUG 2023/02/06 00:28:38 vehicle status: B
Feb 06 00:28:38 evcc evcc[463]: [lp-1  ] DEBUG 2023/02/06 00:28:38 vehicle soc: 67%
Feb 06 00:28:38 evcc evcc[463]: [lp-1  ] DEBUG 2023/02/06 00:28:38 vehicle range: 156km
Feb 06 00:28:38 evcc evcc[463]: [lp-1  ] DEBUG 2023/02/06 00:28:38 planned 1h44m40s until 2023-02-06 02:30:00 +0100 CET at 3680W: total plan duration: 1h44m40s, avg cost: 0.235
Feb 06 00:28:38 evcc evcc[463]: [lp-1  ] DEBUG 2023/02/06 00:28:38 pv charge current: 0A = 16A + -18.6A (4268W @ 1p)
Feb 06 00:28:38 evcc evcc[463]: [lp-1  ] DEBUG 2023/02/06 00:28:38 site power 4268W >= 1000W disable threshold
Feb 06 00:28:38 evcc evcc[463]: [lp-1  ] DEBUG 2023/02/06 00:28:38 pv disable in 0s
Feb 06 00:28:38 evcc evcc[463]: [lp-1  ] DEBUG 2023/02/06 00:28:38 pv disable timer elapsed
Feb 06 00:28:38 evcc evcc[463]: [lp-1  ] DEBUG 2023/02/06 00:28:38 charger disable
Feb 06 00:28:38 evcc evcc[463]: [lp-1  ] DEBUG 2023/02/06 00:28:38 wake-up timer: stop
Feb 06 00:28:48 evcc evcc[463]: [site  ] DEBUG 2023/02/06 00:28:48 ----
Feb 06 00:28:48 evcc evcc[463]: [lp-1  ] DEBUG 2023/02/06 00:28:48 charge power: 0W
Feb 06 00:28:48 evcc evcc[463]: [site  ] DEBUG 2023/02/06 00:28:48 pv power: 0W
Feb 06 00:28:48 evcc evcc[463]: [site  ] DEBUG 2023/02/06 00:28:48 grid power: 677W
Feb 06 00:28:48 evcc evcc[463]: [site  ] DEBUG 2023/02/06 00:28:48 grid powers: [64 247 366]W
Feb 06 00:28:48 evcc evcc[463]: [site  ] DEBUG 2023/02/06 00:28:48 grid currents: [0.98 2.44 2.27]A
Feb 06 00:28:48 evcc evcc[463]: [site  ] DEBUG 2023/02/06 00:28:48 site power: 677W
Feb 06 00:28:58 evcc evcc[463]: [site  ] DEBUG 2023/02/06 00:28:58 ----
Feb 06 00:28:58 evcc evcc[463]: [lp-1  ] DEBUG 2023/02/06 00:28:58 charge power: 0W
Feb 06 00:28:58 evcc evcc[463]: [site  ] DEBUG 2023/02/06 00:28:58 pv power: 0W
Feb 06 00:28:58 evcc evcc[463]: [site  ] DEBUG 2023/02/06 00:28:58 grid power: 666W
Feb 06 00:28:58 evcc evcc[463]: [site  ] DEBUG 2023/02/06 00:28:58 grid powers: [62 249 356]W
Feb 06 00:28:58 evcc evcc[463]: [site  ] DEBUG 2023/02/06 00:28:58 grid currents: [0.977 2.44 2.25]A
Feb 06 00:28:58 evcc evcc[463]: [site  ] DEBUG 2023/02/06 00:28:58 site power: 666W
Feb 06 00:28:58 evcc evcc[463]: [lp-1  ] DEBUG 2023/02/06 00:28:58 charger status: B
Feb 06 00:28:58 evcc evcc[463]: [lp-1  ] INFO 2023/02/06 00:28:58 stop charging <-
```

Der Ladevorgang ist gestoppt und in der WebUI steht bei Zielladen eine neue Startzeit 0:45 Uhr.
Um 0:45 Uhr geht der Ladevorgang dann weiter.

Meine Frage ist jetzt, warum stoppt der den Ladevorgang um 0:28 und macht dann um 0:45 weiter und lädt nicht durchgehend?

Hofyyy, 2023-02-06:
Das kann ich sehr wahrscheinlich erklären. Der Estimator schätzt initial wahrscheinlich eine zu hohe dauer.
- Rechne mal den initalen ladeplan per hand nach, ob der für dich stimmt.

Wärend des ladens wird dieser korrigiert und er merkt, dass er viel schneller fertig wird. Dann hat spätes laden priorität, da es ja noch sonne geben könnte und er stoppt. Dann startet er nach bestem wissen und gewissen das laden zum spät möglichsten zeitpunkt neu. 

Ist aber nur geraden.

Aktuelle Themen sind ein hochwertiger Estimator.

andig, 2023-02-06:
> Während des ladens wird dieser korrigiert und er merkt, dass er viel schneller fertig wird. 

Und genau das sieht man im Logfile. Hast Du Dir die Updates des Ladeplanes schon angeschaut?

princemaxwell, 2023-02-06:
@Hofyyy Also es ist tatsächlich so, der Wagen lädt "zu schnell" und die "Pause" ist meist so 15-30 Minuten nur, bis der neu geplante Zielladen-Vorgang dann startet.

@andig  Also direkt im Anschluss ans Stoppen der Ladung wird der neue Plan gesetzt.
```
Feb 06 00:29:02 evcc evcc[463]: [lp-1  ] DEBUG 2023/02/06 00:29:02 planned 1h44m39s until 2023-02-06 02:30:00 +0100 CET at 3680W: total plan duration: 1h44m39s, avg cost: 0.235
Feb 06 00:29:02 evcc evcc[463]: [lp-1  ] DEBUG 2023/02/06 00:29:02 pv charge current: 0A = 0A + -2.9A (666W @ 1p)
Feb 06 00:29:02 evcc evcc[463]: [lp-1  ] DEBUG 2023/02/06 00:29:02 pv enable timer reset
Feb 06 00:29:02 evcc evcc[463]: [lp-1  ] DEBUG 2023/02/06 00:29:02 pv timer inactive
```

Hofyyy, 2023-02-06:
Also ich finde das verhalten auch mehr als unschön und würde gerne die hälfte umbauen. Aber es hilft ja nichts. Können und Zeit :-)

Es gibt aktuell folgende Überlegungen
- Einen besseren estimator zu nehmen
- Und die aufschläge bei 80% und 90% target soc zu entfernen. 

Ich lade aktuell per evcc nur bis 90% weil es da für meinen anwendungsfall gut funktioniert.

princemaxwell, 2023-02-06:
@Hofyyy Also ist das Verhalten nicht nur bei mir so, sondern technisch so gegeben?

Nach welcher Logik entscheidet sich denn, wann ein Vorgang stoppt und neu berechnet wird?
Warum läuft das nicht einfach durch?

Hofyyy, 2023-02-06:
Naja. Die grundidee ist folgender maßen und auch nachvollziehbar. 
- Es wird in slots geplant nach kosten. Diese müssen erstmal nicht zusammenhängen. 
- und es wird danach so spät wie möglich geladen, damit sonne vorher kommen könnte. 

Wenn also alles gleichteuer ist. Normales zielladen ohne spezielle tarife. dann fängt er halt so spät wie möglich an. 

Wenn er sich nur sehr verrechnet. was er nach meiner auffassung nach, altuell bei > 90% target soc immer tut, dann hört er halt auf, wenn er merkt er ist viel zu früh. 

Mögliche quickfixes wären:
- offsets > 90% massiv zu verkleinern
- zwischen sonnen untergang und aufgang nicht mehr zu stoppen (innerhalb eines slots)
- die zusätzlichen offsets zu löschen und ladeeffizienz und einen zeitoffset einstellbar zu machen.

princemaxwell, 2023-02-06:
@Hofyyy 
Ich habe target soc = 80%, also nicht so weit davon entfernt, viel weiter drunter macht für mich wenig Sinn, ich brauche 60% am Tag. Ein wenig Puffer wirkt beruhigend.

Zu den Quickfixes:
Bei PV Strom = 0W sollte nicht gestoppt werden. Da brauchts dann auch keine Auswertung von Sonnenauf- und untergangszeit.

Oder?

andig, 2023-02-06:
> Bei PV Strom = 0W sollte nicht gestoppt werden. Da brauchts dann auch keine Auswertung von Sonnenauf- und untergangszeit.

Wir auch nicht wenn der Planner aktiv ist.

andig, 2023-02-06:
> planned 1h44m39s until 2023-02-06 02:30:00

@princemaxwell wieso geht der Plan bis 2:30? War das Deine Zielzeit?

Was hier passiert ist ja stopp und nach 15min wieder go. Aktuell unterbinden wir das wenn das Delta < 10min ist. Damit könnten wir spielen und z.B. auch <30min draus machen.

/cc @schenlap @Hofyyy

Hofyyy, 2023-02-06:
@andig 
Mit dem Wert hoch zu gehen finde ich gut. Bekommen wir raus wie lang die Planungsslots sind? bei Tibber scheint das ja "immer" 1 Stunde zu sein. Ich würde sogar soweit gehen das Intervall der Planungsslots zu nehmen, wenn nicht > als 1 Stunde.

Wenn der Estimator es richtig berechnet tut es nicht weh, und wenn der Estimator sich verrechnet springt er nur in ganzen planungsintervallen und max. 1 Stunden.

Hofyyy, 2023-02-06:
> @Hofyyy Ich habe target soc = 80%, also nicht so weit davon entfernt, viel weiter drunter macht für mich wenig Sinn, ich brauche 60% am Tag. Ein wenig Puffer wirkt beruhigend.
>
> Zu den Quickfixes: [...]

Ok, wenn du kleiner 90% einstellst sollte nichts ganz verrücktes passieren. Allerdings rechnen wir bei EVCC aktuell an zwei stellen die Ladeverluste mit drauf. Es wird also implizit ein Puffer von 2x10% angenommen. Wenn dein Auto nur 6% Ladeverluste hat liegt man halt 14% daneben. Und genau das bekommt man erst über die Zeit raus.

Daher die Idee den Wert einstellbar zu machen. Dann läd man einmal durch rechnet sich seinen Ladeverlust aus und stellt ihn ein mit meinetwegen +3% um sicher zu sein.

princemaxwell, 2023-02-06:
@andig 
> wieso geht der Plan bis 2:30? War das Deine Zielzeit?
Hatte ich um 23:30 **manuell** so eingestellt, damit es um 23:45 startet.
Aber die Ladepause kommt auch bei anderen Uhrzeiten.

Ich weiß nur nicht, ob das Re-Planing immer um 0:00 Uhr startet, weil erst dort der Eintrag im Log erschien.

> Aktuell unterbinden wir das wenn das Delta < 10min ist. Damit könnten wir spielen und z.B. auch <30min draus machen.
Ihr könntet es ja mal ins nightly packen und ich teste es gerne. Ich muss halt jeden Werktag ca. 60% laden, da haben wir schnell ein Ergebnis.

@Hofyyy Die Ladeverluste zu messen ist nicht so schwer, meine Wallbox hat nen eigenen Zähler. Brauche ich ja nur vorher und hinterher ablesen und mit den Logs vergleichen.

andig, 2023-02-06:
Hast du morgen: https://github.com/evcc-io/evcc/commit/1b71c4ff82963084b15b8db436d19e965b01e6b1

> Allerdings rechnen wir bei EVCC aktuell an zwei stellen die Ladeverluste mit drauf. 

Das stimmt. Und zusätzlich Ladeabfall zum Ladeende/ bei hohen Leistungen/ kleinen Akkugrößen. Genau genommen ist das eine Kurvenschar die man sicher linear approximieren könnte wenn man sie kennen würde ;) Es könnte helfen mal ein Modell davon zu entwickeln.

schenlap, 2023-02-06:
Was ich noch nicht verstehe ist warum er überhaupt stoppt. Wenn ein Slot begonnen wurde beendet er ihn nicht mehr.
Könntest du mal lp-1 auf TRACE stellen, dann sieht man den gesamten Plan schön?

princemaxwell, 2023-02-07:
@Hofyyy @schenlap 
Ich habe heute nochmal mit Version 0.112.4 ein Trace-Log erstellt, die Erkenntnisse sind interessant.
Füllstand: 13%, Ziel: 6:00 Uhr.

```
Feb 06 21:33:00 evcc evcc[2181]: [lp-1  ] DEBUG 2023/02/06 21:33:00 planned 8h19m12s until 2023-02-07 06:00:00 +0100 CET at 3680W: total plan duration: 8h19m12s, avg cost: 0.235
Feb 06 21:33:00 evcc evcc[2181]: [lp-1  ] TRACE 2023/02/06 21:33:00   slot from: 2023-02-06 21:40:48 +0100 CET to 2023-02-07 00:00:00 +0100 CET cost 0.235
Feb 06 21:33:00 evcc evcc[2181]: [lp-1  ] TRACE 2023/02/06 21:33:00   slot from: 2023-02-07 00:00:00 +0100 CET to 2023-02-07 06:00:00 +0100 CET cost 0.235
```

Man sieht, es sind wohl 2 Slots vorgesehen, der Zweite ab 0:00 Uhr.

Um 21:41 Uhr startet das Laden ganz normal, den Log erspar ich Euch. Zeitgleich wird auch nochmal die Car-API angefragt und liefert aktuelle Werte.

`Feb 06 21:41:24 evcc evcc[2181]: [nissan] TRACE 2023/02/06 21:41:24 {"id":"***","chargeStatus":0,"timeRequiredToFullFast":420,"rangeHvacOff":34,"plugStatusDetail":1,"batteryBarLevel":32,"plugStatus":1,"chargePower":1,"lastUpdateTime":"2023-02-06T20:29:51Z","rangeHvacOn":28,"batteryLevel":13,"timeRequiredToFullSlow":1470,"timeRequiredToFullNormal":630,"batteryCapacity":40000}`

Um 21:56 Uhr haben wir eine Abweichung von ca. 30 Sekunden `TRACE 2023/02/06 21:56:40   slot from: 2023-02-06 21:57:13`.

**Und jetzt wirds spannend**, denn um 21:57 Uhr wird die Car-API erneut abgefragt und liefert neue Werte:

`Feb 06 21:57:21 evcc evcc[2181]: [nissan] TRACE 2023/02/06 21:57:21 {"id":"***","chargeStatus":1,"plugStatus":1,"batteryBarLevel":38,"rangeHvacOff":40,"batteryCapacity":40000,"timeRequiredToFullSlow":0,"rangeHvacOn":33,"chargePower":2,"timeRequiredToFullNormal":630,"batteryLevel":16,"timeRequiredToFullFast":0,"plugStatusDetail":1,"lastUpdateTime":"2023-02-06T20:57:25Z"}`

Die neue Berechnung vom Estimator zeigt eine Abweichung von 7 Minuten.

`Feb 06 21:57:21 evcc evcc[2181]: [lp-1  ] TRACE 2023/02/06 21:57:21   slot from: 2023-02-06 22:04:58 +0100 CET to 2023-02-07 00:00:00 +0100 CET cost 0.235`

Um 22:13 Uhr erneute Anfrage an die Car-API, Batterielevel jetzt 19%, Abweichung nun schon 16 Minuten.

`Feb 06 22:13:01 evcc evcc[2181]: [lp-1  ] TRACE 2023/02/06 22:13:01   slot from: 2023-02-06 22:29:07 +0100 CET to 2023-02-07 00:00:00 +0100 CET cost 0.235`

Und die Abfragen bei der Car-API gehen im 15-Minuten-Takt so weiter.
22:28 Uhr, Abweichung 17 Minuten
22:44 Uhr, Abweichung 17 Minuten
23:00 Uhr, Abweichung 25 Minuten
23:15 Uhr, Abweichung 26 Minuten
23:31 Uhr, Abweichung 34 Minuten

`Feb 06 23:31:01 evcc evcc[2181]: [lp-1  ] TRACE 2023/02/06 23:31:01   slot from: 2023-02-06 23:57:22 +0100 CET to 2023-02-07 00:00:00 +0100 CET cost 0.235`

Kurz drauf fliegt Slot 1 auch aus dem Logging raus (Ladung läuft aber noch weiter).

```
Feb 06 23:31:21 evcc evcc[2181]: [lp-1  ] DEBUG 2023/02/06 23:31:21 planned 5h54m16s until 2023-02-07 06:00:00 +0100 CET at 3680W: total plan duration: 5h54m16s, avg cost: 0.235
Feb 06 23:31:21 evcc evcc[2181]: [lp-1  ] TRACE 2023/02/06 23:31:21   slot from: 2023-02-07 00:05:44 +0100 CET to 2023-02-07 06:00:00 +0100 CET cost 0.235
```

23:47 Uhr, Abweichung 43 Minuten

Um exakt 0:00 Uhr stoppt der Ladevorgang und pausiert bis zum Beginn des 2. Slots um 0:43 Uhr

```
Feb 07 00:00:00 evcc evcc[2181]: [lp-1  ] DEBUG 2023/02/07 00:00:00 planned 5h16m9s until 2023-02-07 06:00:00 +0100 CET at 3680W: total plan duration: 5h16m9s, avg cost: 0.235
Feb 07 00:00:00 evcc evcc[2181]: [lp-1  ] TRACE 2023/02/07 00:00:00   slot from: 2023-02-07 00:43:51 +0100 CET to 2023-02-07 06:00:00 +0100 CET cost 0.235
Feb 07 00:00:00 evcc evcc[2181]: [lp-1  ] DEBUG 2023/02/07 00:00:00 pv charge current: 0A = 16A + -18.9A (4349W @ 1p)
Feb 07 00:00:00 evcc evcc[2181]: [lp-1  ] DEBUG 2023/02/07 00:00:00 site power 4349W >= 1000W disable threshold
Feb 07 00:00:00 evcc evcc[2181]: [lp-1  ] DEBUG 2023/02/07 00:00:00 pv disable in 0s
Feb 07 00:00:00 evcc evcc[2181]: [lp-1  ] DEBUG 2023/02/07 00:00:00 pv disable timer elapsed
Feb 07 00:00:00 evcc evcc[2181]: [lp-1  ] DEBUG 2023/02/07 00:00:00 charger disable
Feb 07 00:00:00 evcc evcc[2181]: [lp-1  ] DEBUG 2023/02/07 00:00:00 wake-up timer: stop
Feb 07 00:00:10 evcc evcc[2181]: [site  ] DEBUG 2023/02/07 00:00:10 ----
Feb 07 00:00:10 evcc evcc[2181]: [lp-1  ] DEBUG 2023/02/07 00:00:10 charge power: 0W
Feb 07 00:00:10 evcc evcc[2181]: [site  ] DEBUG 2023/02/07 00:00:10 pv power: 0W
Feb 07 00:00:10 evcc evcc[2181]: [site  ] DEBUG 2023/02/07 00:00:10 grid power: 757W
Feb 07 00:00:10 evcc evcc[2181]: [site  ] DEBUG 2023/02/07 00:00:10 grid powers: [66 408 283]W
Feb 07 00:00:10 evcc evcc[2181]: [site  ] DEBUG 2023/02/07 00:00:10 grid currents: [1 2.9 1.97]A
Feb 07 00:00:10 evcc evcc[2181]: [site  ] DEBUG 2023/02/07 00:00:10 site power: 757W
Feb 07 00:00:20 evcc evcc[2181]: [site  ] DEBUG 2023/02/07 00:00:20 ----
Feb 07 00:00:20 evcc evcc[2181]: [lp-1  ] DEBUG 2023/02/07 00:00:20 charge power: 0W
Feb 07 00:00:20 evcc evcc[2181]: [site  ] DEBUG 2023/02/07 00:00:20 pv power: 0W
Feb 07 00:00:20 evcc evcc[2181]: [site  ] DEBUG 2023/02/07 00:00:20 grid power: 744W
Feb 07 00:00:20 evcc evcc[2181]: [site  ] DEBUG 2023/02/07 00:00:20 grid powers: [67 397 281]W
Feb 07 00:00:20 evcc evcc[2181]: [site  ] DEBUG 2023/02/07 00:00:20 grid currents: [1.01 2.86 1.96]A
Feb 07 00:00:20 evcc evcc[2181]: [site  ] DEBUG 2023/02/07 00:00:20 site power: 744W
Feb 07 00:00:20 evcc evcc[2181]: [lp-1  ] DEBUG 2023/02/07 00:00:20 charger status: B
Feb 07 00:00:20 evcc evcc[2181]: [lp-1  ] INFO 2023/02/07 00:00:20 stop charging <-
```

@andig Da hätte mir leider auch die Anpassung auf 30 Minuten in Version 0.112.5 nichts gebracht :-/

Was mich wundert ist die 15-minütige API Abfrage, ich habe das auf 60m stehen:
```
  soc:
    poll:
      mode: always
      interval: 60m
    estimate: true
```

Ich hoffe, Ihr könnt aus den ganzen Logs was ableiten, wenn gewünscht kann ich auch eine zusammenhängende Log schicken, die ist aber recht lang.

andig, 2023-02-07:
Deine Polleinstellung betrifft nur die Abfrage wenn nicht geladen wird. Während des Ladens passiert das über `cache` am Fahrzeug.

> Ich hoffe, Ihr könnt aus den ganzen Logs was ableiten

Eigentlich nur, dass alles funktioniert wie erwartet. Anders gesagt: welches Problem willst Du hier denn lösen?

princemaxwell, 2023-02-07:
@andig  eigentlich hätte ich nicht erwartet, dass eine gestartete Ladung unterbrochen wird. Also die Aufteilung in 2 Slots scheint ja dafür verantwortlich.

Denn vor 0:00 Uhr gab es ja bereits eine Abweichung die nicht zur Unterbrechung geführt hat, so wie @schenlap  gesagt hat, ein begonnener Slot sollte nicht unterbrochen werden.

Kann man die tagesübergreifende Ladung in 1 Slot packen? Dann dürfte es nicht zu einer Unterbrechung führen, oder?

schenlap, 2023-02-08:
Die Slots selbst werden über den Tarif vorgegeben, der Planner wählt dann nur aus. Ich wusste nicht dass mit Mitternacht ein neuer Slot generiert wird wenn man tariff über zones vorgibt. Aber so ist das Verhalten für mich nachvollziehbar und der planner macht was er soll.

princemaxwell, 2023-02-08:
@schenlap @Hofyyy @andig 
Ist es denn richtig, das ab Mitternacht ein neuer zweiter Slot entsteht? Kann man das ändern?

andig, 2023-02-08:
Nein, kannst du nicht ändern. Ich fasse mal zusammen: die Ladung funktioniert, hat aber eine Pause. Pausen sind möglich und per Design. Das Auto wird geladen. Aus meiner Sicht kein Fehler, der zu beheben wäre. Closing.

closed 2023-02-08 by andig

princemaxwell, 2023-04-11:
@andig @schenlap @Hofyyy 

Wir hatten ja mal das Thema mit den Slots beim Zielladen und den Pausen zwischendurch. Damals war es eine Pause beim Tageswechsel um 0 Uhr.

Mittlerweile wird der Ladevorgang in deutlich mehr Slots (hier: 8) gepackt, sodass durch etwas schnelleres Laden als der Estimator annimmt, immer wieder pausiert wird.

Hat das was mit #6970 oder #6960 zu tun?

Ich frage mich, ob dieser vielen Stopps + Starts nicht langfristig übers Material bei Wallbox und Auto gehen.

```
Apr 11 20:14:11 evcc evcc[873]: [nissan] TRACE 2023/04/11 20:14:11 {"id":"***","chargePower":1,"plugStatus":0,"chargeStatus":0,"timeRequiredToFullFast":330,"timeRequiredToFullNormal":540,"batteryCapacity":40000,"timeRequiredToFullSlow":1470,"batteryBarLevel":65,"batteryLevel":27,"rangeHvacOff":70,"rangeHvacOn":67,"lastUpdateTime":"2023-04-11T18:10:40Z"}
Apr 11 20:14:11 evcc evcc[873]: [amtron] TRACE 2023/04/11 20:14:11 modbus: send bc 96 00 00 00 06 ff 04 03 02 00 01
Apr 11 20:14:11 evcc evcc[873]: [amtron] TRACE 2023/04/11 20:14:11 modbus: recv bc 96 00 00 00 05 ff 04 02 00 03
Apr 11 20:14:11 evcc evcc[873]: [lp-1  ] DEBUG 2023/04/11 20:14:11 vehicle status: A
Apr 11 20:14:11 evcc evcc[873]: [lp-1  ] DEBUG 2023/04/11 20:14:11 vehicle soc: 27%
Apr 11 20:14:11 evcc evcc[873]: [lp-1  ] DEBUG 2023/04/11 20:14:11 vehicle range: 70km
Apr 11 20:14:11 evcc evcc[873]: [amtron] TRACE 2023/04/11 20:14:11 modbus: send bc 97 00 00 00 06 ff 03 04 00 00 01
Apr 11 20:14:11 evcc evcc[873]: [amtron] TRACE 2023/04/11 20:14:11 modbus: recv bc 97 00 00 00 05 ff 03 02 00 00
Apr 11 20:14:11 evcc evcc[873]: [lp-1  ] DEBUG 2023/04/11 20:14:11 planned 7h6m43s until 2023-04-12 06:00:00 +0200 CEST at 3680W: total plan duration: 7h6m43s, avg cost: 0.235
Apr 11 20:14:11 evcc evcc[873]: [lp-1  ] TRACE 2023/04/11 20:14:11   slot from: 2023-04-11 22:53:17 +0200 CEST to 2023-04-11 23:00:00 +0200 CEST cost 0.235
Apr 11 20:14:11 evcc evcc[873]: [lp-1  ] TRACE 2023/04/11 20:14:11   slot from: 2023-04-11 23:00:00 +0200 CEST to 2023-04-12 00:00:00 +0200 CEST cost 0.235
Apr 11 20:14:11 evcc evcc[873]: [lp-1  ] TRACE 2023/04/11 20:14:11   slot from: 2023-04-12 00:00:00 +0200 CEST to 2023-04-12 01:00:00 +0200 CEST cost 0.235
Apr 11 20:14:11 evcc evcc[873]: [lp-1  ] TRACE 2023/04/11 20:14:11   slot from: 2023-04-12 01:00:00 +0200 CEST to 2023-04-12 02:00:00 +0200 CEST cost 0.235
Apr 11 20:14:11 evcc evcc[873]: [lp-1  ] TRACE 2023/04/11 20:14:11   slot from: 2023-04-12 02:00:00 +0200 CEST to 2023-04-12 03:00:00 +0200 CEST cost 0.235
Apr 11 20:14:11 evcc evcc[873]: [lp-1  ] TRACE 2023/04/11 20:14:11   slot from: 2023-04-12 03:00:00 +0200 CEST to 2023-04-12 04:00:00 +0200 CEST cost 0.235
Apr 11 20:14:11 evcc evcc[873]: [lp-1  ] TRACE 2023/04/11 20:14:11   slot from: 2023-04-12 04:00:00 +0200 CEST to 2023-04-12 05:00:00 +0200 CEST cost 0.235
Apr 11 20:14:11 evcc evcc[873]: [lp-1  ] TRACE 2023/04/11 20:14:11   slot from: 2023-04-12 05:00:00 +0200 CEST to 2023-04-12 06:00:00 +0200 CEST cost 0.235
Apr 11 20:14:11 evcc evcc[873]: [lp-1  ] DEBUG 2023/04/11 20:14:11 pv charge current: 0A = 0A + -1.98A (455W @ 1p)
Apr 11 20:14:11 evcc evcc[873]: [lp-1  ] DEBUG 2023/04/11 20:14:11 pv enable timer reset
Apr 11 20:14:11 evcc evcc[873]: [lp-1  ] DEBUG 2023/04/11 20:14:11 pv timer inactive
```

Hofyyy, 2023-04-12:
Es wäre interessant zu dem Ladeplan den dann realen ladeplan mit den pausen zu kennen. und ob du auf dem nigtly oder release bist. 

Pauschal wäre interessant ob du das problem bei Target SoC 90% auch hast.

Aber ja über SOC 90 ist der estimator im moment schwierig. Da könnten wir ggf den faktor als quickfix anpasseb. 

Aber es sollten trotzdem keine oder kaum pausen gemacht werden.

princemaxwell, 2023-04-12:
@Hofyyy 
Bin auf der aktuellen nightly.
Eingestellt ist, Ziel 80% bis 6:00 Uhr.

Was meinst du mit realem Ladeplan?

Hier ist mein ntfy Verlauf wo man Starts und Ende und jeweiligen Füllstand sieht. Da wird auch bei geringem Stand pausiert.

[image]

[image]

[image]

[image]

[image]

andig, 2023-04-12:
> Mittlerweile wird der Ladevorgang in deutlich mehr Slots (hier: 8) gepackt, sodass durch etwas schnelleres Laden als der Estimator annimmt, immer wieder pausiert wird.

Ist das denn wirklich so oder eine Annahme? Passieren kann das egtl. nur wenn dadurch der Ladestart immer wieder nach hinten springt (=fixer Tarif). Geändert hatten wir, dass fixe Tarife jetzt in Stundenslots aufgeteilt werden.

Hofyyy, 2023-04-12:
@princemaxwell 
Du hast ja die Ladeplanung schon auf Trace. Interessant wäre, wenn du das Log von einer Ladung mal exportierst und ein wenig zusammenstreichst, wo man die Pausen sieht und den jeweils (live) geupdaten Ladeplan. Dann sehen wir was passiert.

Eine Frage noch. wie schnell ist deine maximale Ladegeschwindigkeit? 11kW oder mehr?

@andig 
Naja wenn wir noch Bugs haben sollten, können Sie jetzt wegen den Stunden halt zuschlagen. allerdings denke ich die änderung mit den Stunden war schon gut. Wir müssen jetzt nur gucken wo es noch nicht klappt.

Ich kann das hier aber erstmal begleiten, bis wir wissen was sache ist.

princemaxwell, 2023-04-12:
@Hofyyy 
Ich hab ne 11kW Wallbox, aber nur ein 1-phasig ladendes Fahrzeug.
Somit liegt die Geschwindigkeit bei 3,7kW (16A) und Zielladen habe ich auf 80% / 6:00 Uhr.

Da die Starts+Stopps ebenfalls per Messaging protokolliert werden, hier die verkürzte Übersicht anstelle der großen Bilder:

```
Nachricht erhalten: VERBUNDEN - Nissan Leaf 21E mit Mennekes AMTRON® Xtra 11 C2 - Dienstag, 11.04.2023 um 20:14:10
Setze targettime (Ladepunkt 1 - Mennekes):  2023-04-12T04:00:00Z

Nachricht erhalten: START Ladevorgang - Nissan Leaf 21E - Dienstag, 11.04.2023 um 22:53:50
Nachricht erhalten: ENDE Ladevorgang - Nissan Leaf 21E - Dienstag, 11.04.2023 um 23:57:10

Nachricht erhalten: START Ladevorgang - Nissan Leaf 21E - Mittwoch, 12.04.2023 um 00:30:30
Nachricht erhalten: ENDE Ladevorgang - Nissan Leaf 21E - Mittwoch, 12.04.2023 um 01:18:10

Nachricht erhalten: START Ladevorgang - Nissan Leaf 21E - Mittwoch, 12.04.2023 um 01:50:50
Nachricht erhalten: ENDE Ladevorgang - Nissan Leaf 21E - Mittwoch, 12.04.2023 um 03:56:50

Nachricht erhalten: START Ladevorgang - Nissan Leaf 21E - Mittwoch, 12.04.2023 um 04:31:50
Nachricht erhalten: ENDE Ladevorgang - Nissan Leaf 21E - Mittwoch, 12.04.2023 um 05:42:50

Nachricht erhalten: GETRENNT - Nissan Leaf 21E - Mittwoch, 12.04.2023 um 07:46:30
```

Als ich den Fall aufgemacht habe vor einiger Zeit, gab es regelmäßig eine Pause um exakt 0:00 Uhr (siehe https://github.com/evcc-io/evcc/issues/6044#issuecomment-1419981869), weil dort von vornherein ein neuer Slot geplant wurde, der estimator aber anscheinend neu rechnet und sagt: "hey, wir können später beginnen". Das liegt eben an der Idee des "Zielladens".

[trace.log](https://github.com/evcc-io/evcc/files/11214725/trace.log)

Im Trace Log habe ich jetzt alle [lp-1] und [nissan} Zeilen gefiltert, weil alles andere aus meiner Sicht keinen Sinn ergibt.
Was ich erkennen kann. Bis zum Start des 1. Slots bleibt die Planung unverändert.

Nach dem Start "verrutscht" die neue Slotzeit alle 20 Sekunden um 1-2 Sekunden, z.B.:

Apr 11 23:01:10 evcc evcc[873]: [lp-1  ] TRACE 2023/04/11 **23:01:10**   slot from: 2023-04-11 **23:01:20** +0200 CEST to 2023-04-12 00:00:00 +0200 CEST cost 0.235
Apr 11 23:01:30 evcc evcc[873]: [lp-1  ] TRACE 2023/04/11 **23:01:30**   slot from: 2023-04-11 **23:01:41** +0200 CEST to 2023-04-12 00:00:00 +0200 CEST cost 0.235

Richtig krass wirds jedes Mal, wenn von der API neue Wert gezogen werden:
Apr 11 23:09:31 evcc evcc[873]: [lp-1  ] TRACE 2023/04/11 **23:09:31**   slot from: 2023-04-11 **23:09:59** +0200 CEST to 2023-04-12 00:00:00 +0200 CEST cost 0.235

Apr 11 23:09:50 evcc evcc[873]: [nissan] TRACE 2023/04/11 23:09:50 GET https://alliance-platform-caradapter-prod.apps.eu2.kamereon.io/car-adapter/v1/cars/***/battery-status
Apr 11 23:09:51 evcc evcc[873]: [nissan] TRACE 2023/04/11 23:09:51 {"id":"***","batteryLevel":31,"timeRequiredToFullSlow":0,"timeRequiredToFullNormal":540,"chargeStatus":1,"rangeHvacOn":75,"rangeHvacOff":78,"chargePower":2,"lastUpdateTime":"2023-04-11T21:09:46Z","plugStatus":1,"batteryBarLevel":74,"batteryCapacity":40000,"timeRequiredToFullFast":0,"plugStatusDetail":1}

Apr 11 23:09:51 evcc evcc[873]: [lp-1  ] TRACE 2023/04/11 **23:09:51**   slot from: 2023-04-11 **23:25:29** +0200 CEST to 2023-04-12 00:00:00 +0200 CEST cost 0.235

Ich hoffe, damit kannst du was anfangen, im Trace LOG findest Du alle Einträge der letzten Nacht.

Hofyyy, 2023-04-12:
@andig
Ich hab den Fehler verstanden.
Es ist so das leicht schneller geladen wird als gedacht und damit der erste Slot immer weiter nach hinten rückt.
Dazu musst du nur mal durchs logfile scrollen, und dir nur den ersten Slot angucken.

Wenn dann der Abstand zwischen ist und Start > 30min ist, bricht er ab. Das passiert im Log 3x.
Man läd also ausserhalb vor dem ersten Slot vor sich hin.
Eine gute Antwort habe ich dafür noch nicht.

Siehe: 23:56:51

@princemaxwell :
Kannst du mir sagen warum du alle 10sec eine "charge power" meldung hast, aber nur alle 20 sec wirklich was passiert?
Die Antwort warum es dir wegläuft habe ich denke ich gefunden.
Der Ladeplan ist mit 3680W = 16A *230V berechnet.
Du lädst aber mit bis zu 3755W = 16A * 235V und du hast wahrscheinlich weniger als 10% verlust, welche der Estimator annimmt.

Kurzfristige Lösung für dich:
Patch die "capacity: 40" von deinem Vehicle, da diese zur Zeit berechnung genommen wird. Macht du sie kleiner rechnet er weniger Zeit. Ich schlage vor mal mit 5% anzufangen. Also auf z.b. 38KWH zu gehen. Die Lösung hört sich zwar erstmal komisch an, sollte aber sehr gut funktionieren.

princemaxwell, 2023-04-13:
@Hofyyy 
> Kannst du mir sagen warum du alle 10sec eine "charge power" meldung hast, aber nur alle 20 sec wirklich was passiert?

Keine Ahnung, in meiner YAML hab ich `interval: 10s # control cycle interval` drin. Warum nur alle 20s der Plan angezeigt/aktualisiert wird, weiß ich nicht.

> Patch die "capacity: 40" von deinem Vehicle, da diese zur Zeit berechnung genommen wird.

Habe ich jetzt angepasst und werde berichten, denke morgen abend kann ich es testen, weil heute schon aus PV vollgeladen wurde. Trace-Log läuft ja mit.

andig, 2023-04-13:
> Es ist so, das leicht schneller geladen wird als gedacht und damit der erste Slot immer weiter nach hinten rückt.

Dann sehen wir das Problem jetzt in der Breite (auch bei fixen Tarifen), eigentlich ist es aber bei dynamischen Tarifen auch schon immer da gewesen. Wenn wir das mit Deinen Maßnahmen nicht in den Griff bekommen, wäre die Alternative immer noch das Vorgehen komplett umzuwerfen und einfach nur einen einzigen durchgängigen Slot zu planen.

Hofyyy, 2023-04-13:
Also ich würde heute einen vorschlag comitten der die aufschläge bei > 80 und 90 % betreffen. Der punkt hier tritt leider auch schon < 80% auf. 

Meine ehrliche meinung ist:
- Ich denke wir brauchen zeitnah einen ein slot modus. da wir für den rest den perf. estimator brauchen, den wir nicht haben.
- und es wäre hilfreich die fahrzeug lade effizient (10%) optional einstellbar zu machen, wie die kapazität. Dann können die user schnell agieren.

princemaxwell, 2023-04-13:
@andig @Hofyyy 
Ich verstehe Euren Ansatz, die Ladung möglichst weit herauszuzögern, weil es ja Fälle geben kann, da kann man möglichst spät noch PV abgreifen.

In **meinem** Fall ist das so, dass ich mein Auto nach der Arbeit anschließe, das Script setzt automatisch den nächsten Morgen 6:00 als Ziel und fertig. Ich dachte erst, `smartCostLimit (ehemals cheap)` könnte mir helfen, weil ich das so verstehe, dass der bei Erreichen meines günstigen Tarifs um 20:00 Uhr direkt anfängt zu laden, oder?

Da es bei mir aber Ausnahmen gibt, z.B. Mittwochs (Ladung muss erst Freitag 6:00 fertig sein) oder Freitags (Ladung muss erst Montag 6:00 fertig sein) funktioniert das auch nicht, weil dann in der Nacht geladen wird, bevor am nächsten Tag evtl. PV genutzt werden könnte.

Wäre folgender Kompromiss möglich?
Könnt Ihr in den Einstellungen nicht eine Art "FORCE CHARGING" Button einbauen, der dafür sorgt, alles in einen Slot zu packen, bei Leuten, die das benötigen? Obwohl es natürlich auch bei allen anderen unschön ist, dass es Pausen gibt, die evtl. übers Material gehen?

Hofyyy, 2023-04-13:
https://github.com/evcc-io/evcc/pull/7419
Ich lerne :-)

schenlap, 2023-04-13:
Ich habe mir das Log https://github.com/evcc-io/evcc/issues/6044#issuecomment-1505756523 angesehen. Da wird um 23:57:10 beendet, das dürfte er nicht machen da ein gestarteter Slot nicht beendet wird. Die Ladung hätte bis 0:0 laufen müssen.
Ich bin mittlerweile dafür dass Delay im Slot rauszuschmeißen, macht mehr Probleme als es bring. Damit würde es vermutlich stark vereinfacht.

Hofyyy, 2023-04-13:
Das komische ist das er ausserhalb des Slots läd. Weil ihm der Planner vorne wegläuft. Daher greift glaube ich nur der 30min (jetzt 60min) fix in die Zukunft. Es sieht so aus als hat er garkeine aktive korrekte range

Hofyyy, 2023-04-13:
Hier sieht man das er sauber in einem Slot läd:
- Apr 11 22:59:50

Und dann läd er nur weiter wegen dem IF
- plan will re-start shortly, continuing for remaining 4s
- Und startet den Slot glaube ich nie richtig.
- Siehe: 23:00:10

Da ist irgendwas faul

princemaxwell, 2023-04-16:
@Hofyyy 
Hab am Freitag abend nochmal mit der nightly geladen, die die 60 Minuten drin hatte.
Ergebnis: 1 dicke Pause anstatt 4 beim Letzten Mal.

[trace2.log](https://github.com/evcc-io/evcc/files/11243326/trace2.log)

> Da wird um 23:57:10 beendet, das dürfte er nicht machen da ein gestarteter Slot nicht beendet wird. Die Ladung hätte bis 0:0 laufen müssen.

Eure Theorie, dass ein gestarteter Slot nicht beendet wird, ist falsch. Um 00:30 Uhr sind die 60 Minuten Differenz überschritten und er beendet sofort.

```
Apr 15 00:29:59 evcc evcc[7943]: [lp-1  ] DEBUG 2023/04/15 00:29:59 charger status: C
Apr 15 00:29:59 evcc evcc[7943]: [nissan] TRACE 2023/04/15 00:29:59 GET https://alliance-platform-caradapter-prod.apps.eu2.kamereon.io/car-adapter/v1/cars/***/battery-status
Apr 15 00:30:00 evcc evcc[7943]: [nissan] TRACE 2023/04/15 00:30:00 {"id":"***","plugStatus":1,"timeRequiredToFullSlow":0,"timeRequiredToFullNormal":450,"chargePower":2,"batteryLevel":48,"batteryBarLevel":115,"timeRequiredToFullFast":0,"plugStatusDetail":1,"rangeHvacOff":122,"chargeStatus":1,"batteryCapacity":40000,"rangeHvacOn":119,"lastUpdateTime":"2023-04-14T22:30:02Z"}
Apr 15 00:30:00 evcc evcc[7943]: [lp-1  ] DEBUG 2023/04/15 00:30:00 vehicle status: B
Apr 15 00:30:00 evcc evcc[7943]: [lp-1  ] DEBUG 2023/04/15 00:30:00 vehicle soc: 48%
Apr 15 00:30:00 evcc evcc[7943]: [lp-1  ] DEBUG 2023/04/15 00:30:00 vehicle range: 122km
Apr 15 00:30:00 evcc evcc[7943]: [lp-1  ] DEBUG 2023/04/15 00:30:00 planned 4h17m39s until 2023-04-15 06:00:00 +0200 CEST at 3680W: total plan duration: 4h17m39s, avg cost: 0.235
Apr 15 00:30:00 evcc evcc[7943]: [lp-1  ] TRACE 2023/04/15 00:30:00   slot from: 2023-04-15 01:42:21 +0200 CEST to 2023-04-15 02:00:00 +0200 CEST cost 0.235
Apr 15 00:30:00 evcc evcc[7943]: [lp-1  ] TRACE 2023/04/15 00:30:00   slot from: 2023-04-15 02:00:00 +0200 CEST to 2023-04-15 03:00:00 +0200 CEST cost 0.235
Apr 15 00:30:00 evcc evcc[7943]: [lp-1  ] TRACE 2023/04/15 00:30:00   slot from: 2023-04-15 03:00:00 +0200 CEST to 2023-04-15 04:00:00 +0200 CEST cost 0.235
Apr 15 00:30:00 evcc evcc[7943]: [lp-1  ] TRACE 2023/04/15 00:30:00   slot from: 2023-04-15 04:00:00 +0200 CEST to 2023-04-15 05:00:00 +0200 CEST cost 0.235
Apr 15 00:30:00 evcc evcc[7943]: [lp-1  ] TRACE 2023/04/15 00:30:00   slot from: 2023-04-15 05:00:00 +0200 CEST to 2023-04-15 06:00:00 +0200 CEST cost 0.235
Apr 15 00:30:00 evcc evcc[7943]: [lp-1  ] DEBUG 2023/04/15 00:30:00 pv charge current: 0A = 16A + -19.6A (4515W @ 1p)
Apr 15 00:30:00 evcc evcc[7943]: [lp-1  ] DEBUG 2023/04/15 00:30:00 site power 4515W >= 1000W disable threshold
Apr 15 00:30:00 evcc evcc[7943]: [lp-1  ] DEBUG 2023/04/15 00:30:00 pv disable in 0s
Apr 15 00:30:00 evcc evcc[7943]: [lp-1  ] DEBUG 2023/04/15 00:30:00 pv disable timer elapsed
Apr 15 00:30:00 evcc evcc[7943]: [lp-1  ] DEBUG 2023/04/15 00:30:00 charger disable
Apr 15 00:30:00 evcc evcc[7943]: [lp-1  ] DEBUG 2023/04/15 00:30:00 wake-up timer: stop
Apr 15 00:30:09 evcc evcc[7943]: [lp-1  ] DEBUG 2023/04/15 00:30:09 charge power: 0W
Apr 15 00:30:19 evcc evcc[7943]: [lp-1  ] DEBUG 2023/04/15 00:30:19 charge power: 0W
Apr 15 00:30:19 evcc evcc[7943]: [lp-1  ] DEBUG 2023/04/15 00:30:19 charger status: B
Apr 15 00:30:19 evcc evcc[7943]: [lp-1  ] INFO 2023/04/15 00:30:19 stop charging <-
```

Hofyyy, 2023-04-17:
Also bei dir stimmt die Ladegeschwindigkeitsvorhersage nicht und daher hört er ggf. bei einer Differenz von 60 min auf.
Siehe:
- Apr 15 00:29:40 - slot from: 2023-04-15 01:29:07 +0200 CEST to 2023-04-15 02:00:00 +0200 CEST cost 0.235
Differenz von Uhrzeit jetzt zu eigentlichem Slot Start ungefähr 60min
Dann wird neuer SoC gezogen und du bekommst
- Apr 15 00:30:00 -  slot from: 2023-04-15 01:42:21 +0200 CEST to 2023-04-15 02:00:00 +0200 CEST cost 0.235
Hier ist 00:30 Uhr und nächster Start 01:42:21 daher stopt er.

Stell mal deine Kapazität um 5%-10% runter. Kannst auch die geplante Ladedauer mit der realen (ohne Pause) vergleichen und es darauf anpassen. Ist erstmal nur ein Pflaster. Es macht nur keinen Sinn hier weiter zu testen bis der neue Estimator drin ist.

Passt das so für dich oder hast du noch eine andere Beobachtung gemacht?

Hofyyy, 2023-04-17:
@andig 
Hier haben wir genau ein Fall wo wir gerade an der Kapazität drehen, um nicht in Pausen zu laufen. Das mag sich mit dem neuen Estimator lösen. Wenn ja gut, wenn nicht wäre hier der einstellbare Ladeverlust hilfreich. Hatten wir zwar schon besprochen, wollte ich nur noch einmal an ein real Beispiel pinnen.