evcc-io/evcc #6045: ID.4 wird nicht mehr erkannt

mseele, 2023-02-04:
### Describe the bug

Mein konfigurierter ID.4 wird nicht mehr erkannt. Ich hab ihn versucht nochmal zu konfigurieren aber da kommt der selbe Fehler. Scheint sich also auf VW Seite was geändert zu haben.

### Steps to reproduce

1. Vehicle "Volkswagen We Connect ID" konfigurieren

### Configuration details

```yaml
Aktuelle Konfiguration des ID.4 (hat vor ein paar Tagen definitiv noch funktioniert):

vehicles:
  - title: ID.4
    type: template
    template: id
    user: XXX
    password: XXX
    vin: XXX
    capacity: 82
    phases: 3
    icon: car
    mode: pv
    minSoc: 10
    targetSoc: 80
    minCurrent: 6
    maxCurrent: 16
    name: ev_big
```

### Log details

```text
Testing the configuration of Volkswagen We Connect ID ...
[id    ] TRACE 2023/02/04 13:58:28 GET https://login.apps.emea.vwapps.io/authorize?client_id=a24fba63-34b3-4d43-b181-942111e6bda8%40apps_vw-dilab_com&code_challenge=kyY4E1fY1tE94bWj0irPi4Tc1M_8834cX1bYyBYay1g&code_challenge_method=S256&redirect_uri=weconnect%3A%2F%2Fauthenticated&response_type=code+id_token+token&scope=openid+profile+badge+cars+vin&nonce=cPetWEvbQGtclksyJCbhXAfZlJbrtCFDCfMjtZJupYV&state=214ea0a0-d4c9-454a-9d49-507310e681cb
[id    ] TRACE 2023/02/04 13:58:28 GET https://identity.vwgroup.io/oidc/v1/authorize?client_id=a24fba63-34b3-4d43-b181-942111e6bda8%40apps_vw-dilab_com&scope=openid+profile+badge+cars+dealers+vin&response_type=code+id_token+token&redirect_uri=weconnect%3A%2F%2Fauthenticated&state=97d17c37b3b9bfbce5ca52c39a9f693887cb6e66faeb76b4d4b4e7cc77b3bf25&nonce=cPetWEvbQGtclksyJCbhXAfZlJbrtCFDCfMjtZJupYV
[id    ] TRACE 2023/02/04 13:58:29 GET https://identity.vwgroup.io/signin-service/v1/signin/a24fba63-34b3-4d43-b181-942111e6bda8@apps_vw-dilab_com?relayState=e3b214666edf0ac73460bebabe5e4d023ea8ccfa
[id    ] TRACE 2023/02/04 13:58:29 <!DOCTYPE html>
<html lang="en" xmlns="http://www.w3.org/1999/html">
<head>
    <meta charset="utf-8"/>
    <meta name="viewport"
          content="width=device-width, height=device-height, initial-scale=1, maximum-scale=1, user-scalable=no"/>
    <meta name="identitykit" content="loginIdentifier"/>

<script type="text/javascript">(window.NREUM||(NREUM={})).init={privacy:{cookies_enabled:true},ajax:{deny_list:["bam.eu01.nr-data.net"]},distributed_tracing:{enabled:true}};(window.NREUM||(NREUM={})).loader_config={agentID:"535875755",accountID:"2547829",trustKey:"2217462",xpid:"VgMDVl5RARAEUlFQBQQOXlY=",licenseKey:"NRJS-ecd5929e8859410bbd9",applicationID:"460225982"};;(()=>{var e,t,r={9071:(e,t,r)=>{"use strict";r.d(t,{I:()=>n});var n=0,i=navigator.userAgent.match(/Firefox[\/\s](\d+\.\d+)/);i&&(n=+i[1])},6900:(e,t,r)=>{"use strict";let n;if(r.d(t,{H:()=>i}),r(2374).il){const e=document.createElement("div");e.innerHTML="\x3c!--[if lte IE 6]><div></div><![endif]--\x3e\x3c!--[if lte IE 7]><div></div><![endif]--\x3e\x3c!--[if lte IE 8]><div></div><![endif]--\x3e\x3c!--[if lte IE 9]><div></div><![endif]--\x3e",n=e.getElementsByTagName("div").length}var i;i=4===n?6:3===n?7:2===n?8:1===n?9:0},6562:(e,t,r)=>{"use strict";r.d(t,{P_:()=>p,Mt:()=>v,C5:()=>d,DL:()=>y,OP:()=>L,lF:()=>H,Yu:()=>E,Dg:()=>g,CX:()=>f,GE:()=>w,sU:()=>M});var n={};r.r(n),r.d(n,{agent:()=>O,match:()=>N,version:()=>A});var i=r(6797),o=r(909),a=r(8610);class s{constructor(e,t){try{if(!e||"object"!=typeof e)return(0,a.Z)("New setting a Configurable requires an object as input");if(!t||"object"!=typeof t)return(0,a.Z)("Setting a Configurable requires a model to set its initial properties");Object.assign(this,t),Object.entries(e).forEach((e=>{let[t,r]=e;const n=(0,o.q)(t);n.length&&r&&"object"==typeof r&&n.forEach((e=>{e in r&&((0,a.Z)(`"${e}" is a protected attribute and can not be changed in feature ${t}.  It will have no effect.`),delete r[e])})),this[t]=r}))}catch(e){(0,a.Z)("An error occured while setting a Configurable",e)}}}const c={beacon:i.ce.beacon,errorBeacon:i.ce.errorBeacon,licenseKey:void 0,applicationID:void 0,sa:void 0,queueTime:void 0,applicationTime:void 0,ttGuid:void 0,user:void 0,account:void 0,product:void 0,extra:void 0,jsAttributes:{},userAttributes:void 0,atts:void 0,transactionName:void 0,tNamePlain:void 0},u={};function d(e){if(!e)throw new Error("All info objects require an agent identifier!");if(!u[e])throw new Error(`Info for ${e} was never set`);return u[e]}function f(e,t){if(!e)throw new Error("All info objects require an agent identifier!");u[e]=new s(t,c),(0,i.Qy)(e,u[e],"info")}const l={allow_bfcache:!1,privacy:{cookies_enabled:!0},ajax:{deny_list:void 0,enabled:!0,harvestTimeSeconds:10},distributed_tracing:{enabled:void 0,exclude_newrelic_header:void 0,cors_use_newrelic_header:void 0,cors_use_tracecontext_headers:void 0,allowed_origins:void 0},ssl:void 0,obfuscate:void 0,jserrors:{enabled:!0,harvestTimeSeconds:10},metrics:{enabled:!0,harvestTimeSeconds:10},page_action:{enabled:!0,harvestTimeSeconds:30},page_view_event:{enabled:!0},page_view_timing:{enabled:!0,harvestTimeSeconds:30},session_trace:{enabled:!0,harvestTimeSeconds:10},spa:{enabled:!0,harvestTimeSeconds:10}},h={};function p(e){if(!e)throw new Error("All configuration objects require an agent identifier!");if(!h[e])throw new Error(`Configuration for ${e} was never set`);return h[e]}function g(e,t){if(!e)throw new Error("All configuration objects require an agent identifier!");h[e]=new s(t,l),(0,i.Qy)(e,h[e],"config")}function v(e,t){if(!e)throw new Error("All configuration objects require an agent identifier!");var r=p(e);if(r){for(var n=t.split("."),i=0;i<n.length-1;i++)if("object"!=typeof(r=r[n[i]]))return;r=r[n[n.length-1]]}return r}const m={accountID:void 0,trustKey:void 0,agentID:void 0,licenseKey:void 0,applicationID:void 0,xpid:void 0},b={};function y(e){if(!e)throw new Error("All loader-config objects require an agent identifier!");if(!b[e])throw new Error(`LoaderConfig for ${e} was never set`);return b[e]}function w(e,t){if(!e)throw new Error("All loader-config objects require an agent identifier!");b[e]=new s(t,m),(0,i.Qy)(e,b[e],"loader_config")}const E=(0,i.mF)().o;var T=r(6900),x=r(2053),O=null,A=null;if(navigator.userAgent){var _=navigator.userAgent,S=_.match(/Version\/(\S+)\s+Safari/);S&&-1===_.indexOf("Chrome")&&-1===_.indexOf("Chromium")&&(O="Safari",A=S[1])}function N(e,t){if(!O)return!1;if(e!==O)return!1;if(!t)return!0;if(!A)return!1;for(var r=A.split("."),n=t.split("."),i=0;i<n.length;i++)if(n[i]!==r[i])return!1;return!0}var D=r(5526),P=r(2374);const I="NRBA_SESSION_ID";function R(){if(!P.il)return null;try{let e;return null===(e=window.sessionStorage.getItem(I))&&(e=(0,D.ky)(16),window.sessionStorage.setItem(I,e)),e}catch(e){return null}}var C=P._A?.XMLHttpRequest,j=C&&C.prototype;const k={};function L(e){if(!e)throw new Error("All runtime objects require an agent identifier!");if(!k[e])throw new Error(`Runtime for ${e} was never set`);return k[e]}function M(e,t){if(!e)throw new Error("All runtime objects require an agent identifier!");var r;k[e]=new s(t,(r=e,{customTransaction:void 0,disabled:!1,features:{},isolatedBacklog:!1,loaderType:void 0,maxBytes:6===T.H?2e3:3e4,offset:(0,x.yf)(),onerror:void 0,origin:""+P._A?.location,ptid:void 0,releaseIds:{},sessionId:1==v(r,"privacy.cookies_enabled")?R():null,xhrWrappable:C&&j&&j.addEventListener&&!/CriOS/.test(navigator.userAgent),userAgent:n})),(0,i.Qy)(e,k[e],"runtime")}function H(e){return function(e){try{const t=d(e);return!!t.licenseKey&&!!t.errorBeacon&&!!t.applicationID}catch(e){return!1}}(e)}},8226:(e,t,r)=>{"use strict";r.d(t,{q:()=>n});const n="1223.PROD"},9557:(e,t,r)=>{"use strict";r.d(t,{w:()=>o});var n=r(8610);const i={agentIdentifier:""};class o{constructor(e){try{if("object"!=typeof e)return(0,n.Z)("shared context requires an object as input");this.sharedContext={},Object.assign(this.sharedContext,i),Object.entries(e).forEach((e=>{let[t,r]=e;Object.keys(i).includes(t)&&(this.sharedContext[t]=r)}))}catch(e){(0,n.Z)("An error occured while setting SharedContext",e)}}}},4329:(e,t,r)=>{"use strict";r.d(t,{L:()=>d,R:()=>c});var n=r(3752),i=r(7022),o=r(4045),a=r(2325);const s={};function c(e,t){const r={staged:!1,priority:a.p[t]||0};u(e),s[e].get(t)||s[e].set(t,r)}function u(e){e&&(s[e]||(s[e]=new Map))}function d(){let e=arguments.length>0&&void 0!==arguments[0]?arguments[0]:"",t=arguments.length>1&&void 0!==arguments[1]?arguments[1]:"feature";if(u(e),!e||!s[e].get(t))return a(t);s[e].get(t).staged=!0;const r=Array.from(s[e]);function a(t){const r=e?n.ee.get(e):n.ee,a=o.X.handlers;if(r.backlog&&a){var s=r.backlog[t],c=a[t];if(c){ [... cut ...] =0;f<d;f++)s[f].apply(a,r);var l=O()[c[e]];return l&&l.push([g,e,r,a]),a}}function b(e,t){n[e]=w(e).concat(t)}function y(e,t){var r=n[e];if(r)for(var i=0;i<r.length;i++)r[i]===t&&r.splice(i,1)}function w(e){return n[e]||[]}function E(t){return f[t]=f[t]||e(g,t)}function T(e,t){var r=O();g.aborted||(0,o.D)(e,(fu
   Error: cannot create vehicle 'id': invalid character 'b' looking for beginning of object key string
```

### What type of operating system are you running?

Linux

### Version

0.111.1

PaddyX, 2023-02-04:
Gleiches Problem mit einem Skoda Enyaq:

> [main  ] ERROR 2023/02/04 14:27:31 creating vehicle UnserEnyaq failed: cannot create vehicle 'enyaq': invalid character 'b' looking for beginning of object key string
> .Soc:     vehicle not available: cannot create vehicle 'enyaq': invalid character 'b' looking for beginning of object key string

msome, 2023-02-04:
Gleiches Problem mit dem VW eUp, lief seit Monaten problemlos. 

lp-1  ] ERROR 2023/02/04 14:40:35 vehicle soc: vehicle not available: cannot create vehicle 'vw': invalid character 'b' looking for beginning of object key string

seppbusch, 2023-02-04:
Hi kann ich auch bestätigen gleiches Problem beim E-Up seit dieser Woche:

`[main  ] ERROR 2023/02/04 14:42:27 creating vehicle eup failed: cannot create vehicle 'template': cannot create vehicle 'vw': invalid character 'b' looking for beginning of object key string | stdout`

VolkerK62, 2023-02-04:
schaut doch mal in die Release Notes. Da gab es bei VW mal wieder eine Änderung die mit 0.112.0 behoben wurde.

PaddyX, 2023-02-04:
> schaut doch mal in die Release Notes. Da gab es bei VW mal wieder eine Änderung die mit 0.112.0 behoben wurde.

Danke für den Hinweis. Ich hatte mir nur die Release Notes von 0.112.2 angeschaut und nicht gesehen, dass es seit gestern noch zwei neue Releases gab.

msome, 2023-02-04:
Ich hatte heute erst ein Update auf den Docker nightly gemacht, aber scheinbar gibt es diese 0.112er noch nicht als nightly bei Docker, nur die 0.111er. Das neueste nightly ist schon 2 Wochen alt, das erklärt es natürlich. Dann warte ich mal ab.  Danke.

seppbusch, 2023-02-04:
Bei mir läuft es auch wieder, ich nutze den EVCC Docker Container am Synology NAS.
Aus irgend einem Grund aber den "andig/evcc" Container da war die latest Version eine 0.111.1 
Der evcc/evcc Container hattet dann als latest Version die 0.112.2 mit dem ging es !

Danke Sepp

msome, 2023-02-04:
> Der evcc/evcc Container hattet dann als latest Version die 0.112.2 mit dem ging es !

Danke für den Hinweis, ich war auch noch auf dem alten andig/Ecco unterwegs und habe jetzt umgestellt.

mseele, 2023-02-04:
Fixed in 0.112.2

closed 2023-02-04 by mseele