evcc-io/evcc #6046: EVCC hangs after restart

pauxus, 2023-02-04:
### Describe the bug

Everything worked with 0.111.1. (except for the vw API change).

After updating to 0.112.2, ui was not reachable, last log entry was `pv timer inactive`.

After a restart, UI was visible, log showed detected vehicles, but they were not shown in the ui.

After next restart back to no ui at all.

Downgrading to 0.111.1 made the ui work again (but no VW...)

### Steps to reproduce

1. Upgrade to 0.112.2
2.
3.
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: https
  host: *** # .local suffix announces the hostname on MDNS
  port: 7070

log: trace
levels:
  cache: error
  # vehicle: debug
  lp-1: debug
  lp-2: debug
  renault: trace
  vw: debug
  opel: debug

interval: 10s # control cycle interval

meters:
- type: template
  template: e3dc
  usage: grid
  host: 10.0.70.233
  port: 502
  name: grid1
- type: template
  template: e3dc
  usage: pv
  host: 10.0.70.233
  port: 502
  name: pv2
- type: template
  template: e3dc
  usage: battery
  host: 10.0.70.233
  port: 502
  name: battery3

chargers:
- type: template
  template: wallbe-meter
  host: 10.0.70.235
  port: 502
  name: wallbox5
- type: template
  template: wallbe-meter
  host: 10.0.70.234
  port: 502
  name: wallbox6

vehicles:
- name: opel
  type: template
  template: opel
  title: e-Corsa
  user: *****
  password: *****
  capacity: 46
  targetSoC: 80
  minSoC: 20
- name: multivan
  type: template
  template: vw
  phases: 1
  title: Multivan
  user: *****
  password: *****
  capacity: 10
  targetSoC: 80
  minSoC: 20
- name: renault
  type: template
  template: renault
  title: Megane
  user: *****
  password: *****
  capacity: 60
  targetSoC: 80
  minSoC: 20

loadpoints:
- title: Garage
  charger: wallbox5
  mode: pv
  phases: 3
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: true
  soc:
    estimate: true
- title: Haustür
  charger: wallbox6
  mode: pv
  phases: 3
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: true
  soc:
    estimate: true

site:
  title: BiBo
  meters:
    grid: grid1
    pvs:
    - pv2
    batteries:
    - battery3
  residualPower: 100
  bufferSoC: 100
  prioritySoC: 100

tariffs:
  currency: EUR # (default EUR)
  grid:
    # static grid price
    type: fixed
    price: 0.2898 # [currency]/kWh

  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.071155 # [currency]/kWh

influx:
  url: http://localhost:8086
  database: evcc
  token: *****
  org: "***"
```

### Log details

```text
Feb 04 15:17:57 homedroid evcc[3681916]: [wallbe] TRACE 2023/02/04 15:17:57 modbus: recv 00 0d 00 00 00 05 ff 03 02 3b 14
Feb 04 15:17:57 homedroid evcc[3681916]: [influx] TRACE 2023/02/04 15:17:57 write chargeCurrent=map[value:0] (map[loadpoint:Garage])
Feb 04 15:17:57 homedroid evcc[3681916]: [main  ] DEBUG 2023/02/04 15:17:57 vehicle status: B (e-Corsa)
Feb 04 15:17:57 homedroid evcc[3681916]: [renault] TRACE 2023/02/04 15:17:57 GET https://api-wired-prod-1-euw1.wrd-aws.com/commerce/v1/accounts/***/kamereon/kca/car-adapter/v2/cars/***/battery-status?country=DE
Feb 04 15:17:58 homedroid evcc[3681916]: [renault] TRACE 2023/02/04 15:17:58 {"data":{"type":"Car","id":"***","attributes":{"timestamp":"2023-02-04T11:36:34Z","batteryLevel":47,"batteryTemperature":20,"batteryAutonomy":136,"batteryCapacity":0,"batteryAvailableEnergy":28,"plugStatus":0,"chargingStatus":0.0,"chargingRemainingTime":460,"chargingInstantaneousPower":23.5}}}
Feb 04 15:17:58 homedroid evcc[3681916]: [main  ] DEBUG 2023/02/04 15:17:58 vehicle status: A (Megane)
Feb 04 15:17:58 homedroid evcc[3681916]: [lp-1  ] INFO 2023/02/04 15:17:58 vehicle updated: unknown -> e-Corsa
Feb 04 15:18:02 homedroid evcc[3681916]: [wallbe] TRACE 2023/02/04 15:18:02 modbus: closing connection due to idle timeout: 5.000562911s
Feb 04 15:18:02 homedroid evcc[3681916]: [modbus] TRACE 2023/02/04 15:18:02 modbus: closing connection due to idle timeout: 5.000447364s
Feb 04 15:18:02 homedroid evcc[3681916]: [wallbe] TRACE 2023/02/04 15:18:02 modbus: closing connection due to idle timeout: 5.000318525s
```
```

### What type of operating system are you running?

Linux

### Version

0.112.2

mseele, 2023-02-04:
Maybe related to https://github.com/evcc-io/evcc/issues/6034

pauxus, 2023-02-04:
The error happens with 0.112.1 as well, but not with 0.112.0.

pauxus, 2023-02-04:
> Maybe related to #6034

Sounds familiar, but I do not use a configured tariff planner.

StevieC121176, 2023-02-04:
nimm mal die influxdb aus deiner config

schuepc, 2023-02-04:
Gleicher Fehler wie #6040 die Verbindung zur InfluxDB funktioniert nicht mehr im 0.112.1.
`sudo apt-get install evcc=0.112.0` hilft. oder gleich zurück auf die 0.111.x

naltatis, 2023-02-05:
Duplicate #6034

closed 2023-02-05 by naltatis