evcc-io/evcc #6052: Fix Goodwe

premultiply, 2023-02-04:
Hallo, habe meinen Goodwe Umrichter per RS485 eingebunden. (Danke für die schnelle Integration)
Direkt aufgefallen sind mir 2 Sachen:
Die Erzeugte PV Leistung scheint um eine Kommastelle falsch zu sein.
Habe bei 9 kW PV Leistung eine Anzeige von 900 Watt in evcc.
Und der Hausverbrauch wird immer mit 0 W angeben.
Batterie SOC & Ladeleistung passt. Netzbezug auch.

_Originally posted by @CKlapper in https://github.com/evcc-io/evcc/issues/5869#issuecomment-1416748656_

closed 2023-02-05 by andig

dxwil, 2023-08-25:
Hallo, ich habe das gleiche Problem. Netzexport und Eigenverbrauch werden als 0W angezeigt

andig, 2023-08-25:
Das Problem hier wurde im Februar gelöst. Mach bitte ein neues Issue mit Logfile und Ausgabe von 

    evcc meter