evcc-io/evcc #6055: etron: cannot create vehicle (since a few days, worked before)

hlathome, 2023-02-04:
### Describe the bug

since a few days the log shows this. The config is however unchanged since weeks and used to work fine (except the missing climater status). I triple checked evcc.yaml, there is absolutely no character "b" where it should not be.

[main  ] INFO 2023/02/04 22:43:45 evcc 0.111.1
[main  ] INFO 2023/02/04 22:43:45 using config file: /etc/evcc.yaml
[main  ] ERROR 2023/02/04 22:43:45 creating vehicle ePanzer failed: cannot create vehicle 'template': cannot create vehicle 'etron': invalid character 'b' looking for beginning of object key string
[lp-1  ] ERROR 2023/02/04 22:43:46 vehicle soc: vehicle not available: cannot create vehicle 'template': cannot create vehicle 'etron': invalid character 'b' looking for beginning of object key string

### Steps to reproduce

always

### Configuration details

```yaml
...
vehicles:
- name: ePanzer
  type: template
  template: etron
  title: ePanzer
  capacity: 64 # kWh
  user: mail@xxx.yyy
  password: mySecretPW
  vin: WAUZZZmysecretvin
  cache: 5m
```

### Log details

```text
[main  ] ERROR 2023/02/04 22:50:34 creating vehicle ePanzer failed: cannot create vehicle 'template': cannot create vehicle 'etron': invalid character 'b' looking for beginning of object key string
[site  ] INFO 2023/02/04 22:50:34 site config:
[site  ] INFO 2023/02/04 22:50:34   meters:      grid ✓ pv ✓ battery ✗
[site  ] INFO 2023/02/04 22:50:34     grid:      power ✓ energy ✗ currents ✗
[site  ] INFO 2023/02/04 22:50:34     pv 1:      power ✓ energy ✓ currents ✗
[site  ] INFO 2023/02/04 22:50:34   vehicles:
[site  ] INFO 2023/02/04 22:50:34     vehicle 1: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
```

### What type of operating system are you running?

Linux

### Version

0.111.1

VolkerK62, 2023-02-05:
ReleaseNotes 0.112.0 . Da gab es einen Fix für VAG.

closed 2023-02-05 by andig

hlathome, 2023-02-05:
Danke - könntet Ihr es bitte auf Docker Hub pushen. Oder gibt es eine alternative Repository für die nightly builds?

StefanSchoof, 2023-02-06:
Der Image Name hat sich geändert. Ist jetzt https://hub.docker.com/r/evcc/evcc

pocki80, 2025-03-06:
Hi. This issue appeared again today using v0.200.5 and v0.200.6. It worked until yesterday totally fine, issue started today. 

```
[main ] ERROR 2025/03/06 17:49:46 creating vehicle ev2 failed: cannot create vehicle type 'template': cannot create vehicle type 'etron': invalid character 'c' looking for beginning of object key string
[site ] INFO 2025/03/06 17:49:46 vehicles:
[site ] INFO 2025/03/06 17:49:46 vehicle 1: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
```

Config

```
vehicles:
- type: template
  template: audi 
  user: xxxxxx
  password: yyyyyyy
  vin: WAUZZZxxxxxxx  
  title: A6 Avant TFSIe 
  icon: car  
  capacity: 11  
  phases: 2  
  minCurrent: 6  
  maxCurrent: 16  
  name: ev2
  mode: pv

```