evcc-io/evcc #6074: template: eastron - usage tag not working after update to 0.112.3

k3flo, 2023-02-05:
### Describe the bug

After updating I get these errors:

cannot create meter 'grid1'
cannot create meter 'template'
invalid key: usage

If I comment "usage: grid" and "usag: charge" out for my eastron templates then the UI starts again, but I do not see the charge power anymore as the associated meter cannot be identified.

### Steps to reproduce

use a meter with template "eastron" and "usage"-Tag

### Configuration details

```yaml
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 80

log: info
levels:
  cache: error

interval: 10s # control cycle interval

sponsortoken: xxxxx

tariffs:
  currency: EUR # (default EUR)
  grid:
    # static grid price
    type: fixed
    price: 0.28 # [currency]/kWh

  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.08 # [currency]/kWh

vehicles:
  - name: AiwaysU5
    type: template
    template: aiways
    title: AiwaysU5 # Wird in der Benutzeroberfl  che angezeigt # Optional
    user: xxxxxx # Benutzerkonto (bspw. E-Mail Adresse, User Id, etc.)
    password: xxxxxx # Passwort des Benutzerkontos (bei f  hrenden Nullen bitte in einfache Hochkommata setzen)
    vin: xxxxxx# Erforderlich, wenn mehrere Fahrzeuge des Herstellers vorhanden sind
    capacity: 63 # Akkukapazit  t in kWh # Optional
    targetSoc: 80

meters:
- type: custom
  name: pv2
  power:
    source: modbus
    uri: 192.168.178.70:502
    id: 1
    model: sdm
    value: Power
    scale: -1

- type: template
  template: eastron
  id: 2
  host: 192.168.178.70
  port: 502
  usage: grid
  modbus: rs485tcpip
  name: grid1

- type: template
  template: eastron
  id: 3
  host: 192.168.178.70
  port: 502
  usage: charge
  modbus: rs485tcpip
  name: wallboxMeter

chargers:
- name: wallbox1
  type: template
  template: easee
  user: xxxxx
  password: xxxxxx
  charger: xxxxxx

loadpoints:
- title: Parkplatz
  charger: wallbox1
  vehicle: AiwaysU5
  soc:
    poll:
      mode: charging
      interval: 60m
    estimate: true
  mode: pv
  phases: 1
  mincurrent: 6
  maxcurrent: 32
  resetOnDisconnect: false
```

### Log details

```text
[main  ] INFO 2023/02/05 13:44:29 evcc 0.112.3
[main  ] INFO 2023/02/05 13:44:29 using config file: /etc/evcc.yaml
[main  ] INFO 2023/02/05 13:44:29 starting ui and api at :80
[db    ] INFO 2023/02/05 13:44:29 using sqlite database: /root/.evcc/evcc.db
[main  ] FATAL 2023/02/05 13:44:29 cannot create meter 'grid1': cannot create meter 'template': invalid key: usage
```

### What type of operating system are you running?

Linux

### Version

0.112.3

k3flo, 2023-02-05:
Code was changed here:
https://github.com/evcc-io/evcc/commit/fb899d33ae682a04e12c04dc0fd017a2cde28a72#diff-f68f85d580b6e340e8f7f473279c4fcda0e4894277f0ecfe56de014444625a0b

But why? :(

premultiply, 2023-02-05:
Weil es überhaupt nicht weiter verarbeitet wurde.
Einfach usage: aus deiner Konfig entfernen.

closed 2023-02-05 by premultiply

k3flo, 2023-02-05:
Das kann ehrlichgesagt nicht sein. Ich habe einen Eastron vor meiner Wallbox, den ich als "usage: charge" konfiguriert hatte um den aktuellen Energiefluss zur Wallbox zu erhalten. Nach dem Auskommentieren startet evcc wieder, aber der Energiemesser wird nun logischerweise nicht als "charge" erkannt und die UI zeigt den Energiefluss ins Auto nicht mehr an.

andig, 2023-02-05:
Der Eastron liest immer das gleiche Register, deshalb macht usage hier keinen Sinn und wurde entfernt.

premultiply, 2023-02-05:
Die Zuweisung erfolgt über die Site bzw. am Loadpoint. `usage` hat dabei keine Funktion. Bei anderen Template werden damit unterschiedliche Register/Funktionen abgefragt. Das gibt es bei einzelnen Zählern idR nicht.