evcc-io/evcc #6077: InlfuxDB: evcc 0.111.1 - Berechnung housePower

JAK0721, 2023-02-05:
### Describe the bug

Nach meinem Verständnis wird mit dem Starten der Ladung der Hausverbrauch falsch berechnet.

pi@raspberrypi:~ $ influx
Connected to http://localhost:8086 version 1.8.10
InfluxDB shell version: 1.8.10
> use evcc
Using database evcc
> select * from homePower order by time desc Limit 20
name: homePower
time                value
----                -----

1675612895000000000 342.18171858787537
1675612885000000000 340.538699388504
1675612874000000000 345.80657386779785
1675612864000000000 0
1675612855000000000 701.3166198730469
1675612844000000000 700.5748162269592
1675612842000000000 700.6527247428894

> select * from chargePower order by time desc Limit 20
name: chargePower
time                loadpoint value vehicle
----                --------- ----- -------

1675612884000000000 Garage    7421  Mercedes E300e
1675612874000000000 Garage    7416  Mercedes E300e
1675612864000000000 Garage    7417  Mercedes E300e
1675612854000000000 Garage    0     Mercedes E300e
1675612844000000000 Garage    0     Mercedes E300e
1675612841000000000 Garage    0     Mercedes E300e

### Steps to reproduce

1. Hausverbrauch ohne Ladung bei ca. 700W
2. Ladung mit 7,4kW gestartet
3. Hausverbrauch geht schlagartig auf 350W zurück
4. Ladung gestoppt Hausverbrauch geht wieder auf 700W

### Configuration details

```yaml
network:
  schema: http
  host: 192.168.0.240 # .local suffix announces the hostname on MDNS
  port: 7070

log: info
levels:
  cache: error

interval: 10s # control cycle interval

sponsortoken: *******
meters:
- name: grid
  type: template
  template: volkszaehler-http
  usage: grid
  url: http://192.168.0.240/middleware/data # Beispiel: http://zaehler.network.local:8080/api/data # Optional
  uuid: ee24ade0-1ae3-11ed-a4ff-a56046a7b2fe
- type: template
  template: rct-power
  usage: pv
  host: 192.168.0.225
  name: pv
- type: template
  template: rct-power
  usage: battery
  host: 192.168.0.225
  name: battery

chargers:
- type: template
  template: heidelberg
  id: 2
  device: /dev/ttyUSB0
  baudrate: 19200
  comset: 8E1
  modbus: rs485serial
  name: wallbox

vehicles:
- name: opel
  type: template
  template: opel
  title: Corsa e
  user: *****
  password: *****
  vin: *****
  capacity: 50
  phases: 3
  mode: pv
  minSoC: 0
  targetSoC: 100
  minCurrent: 6
  maxCurrent: 16

- name: mercedes
  type: mercedes
  title: Mercedes E300e
  vin: ******
  clientId: ******
  clientSecret: V********
  capacity: 13.5
  phases: 2
  onIdentify: # set defaults when vehicle is identified
    mode: pv
    minSoC: 0
    targetSoC: 100
    minCurrent: 6
    maxCurrent: 16

loadpoints:
- title: Garage
  charger: wallbox
  mode: PV
  phases: 3
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: true
  soc:
    poll:
      mode: charging
      interval: 60m
    estimate: true

site:
  title: My home
  meters:
    grid: grid
    pvs: pv
    battery: battery
  residualPower: 200
  bufferSoC: 90 # Hausbatterie wird oberhalb SoC 90% als Puffer genutzt
  prioritySoC: 45 # Hausbatterie bekommt bis zum SoC 45% Priorität beim laden

tariffs:
  currency: EUR # (default EUR)
  grid:
    type: fixed
    price: 0.2358 # [currency]/kWh

  feedin:
    type: fixed
    price: 0.0944 # [currency]/kWh

influx:
  url: http://localhost:8086
  database: evcc
  user: ****
  password: *********
```

### Log details

```text
evcc --log debug
[main  ] INFO 2023/02/05 17:11:47 evcc 0.111.1
[main  ] INFO 2023/02/05 17:11:47 using config file: /etc/evcc.yaml
[main  ] INFO 2023/02/05 17:11:47 starting ui and api at :7070
[db    ] INFO 2023/02/05 17:11:56 using sqlite database: /home/pi/.evcc/evcc.db
[site  ] WARN 2023/02/05 17:11:57 vehicle 'Mercedes E300e' does not support automatic detection
[site  ] INFO 2023/02/05 17:11:57 site config:
[site  ] INFO 2023/02/05 17:11:57   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2023/02/05 17:11:57     grid:      power ✓ energy ✗ currents ✗
[site  ] INFO 2023/02/05 17:11:57     pv 1:      power ✓ energy ✗ currents ✗
[site  ] INFO 2023/02/05 17:11:57     battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✗
[site  ] INFO 2023/02/05 17:11:57   vehicles:
[site  ] INFO 2023/02/05 17:11:57     vehicle 1: range ✓ finish ✗ status ✗ climate ✗ wakeup ✗
[site  ] INFO 2023/02/05 17:11:57     vehicle 2: range ✓ finish ✓ status ✓ climate ✓ wakeup ✗
[lp-1  ] INFO 2023/02/05 17:11:57 loadpoint 1:
[lp-1  ] INFO 2023/02/05 17:11:57   mode:        pv
[lp-1  ] INFO 2023/02/05 17:11:57   charger:     power ✓ energy ✓ currents ✓ phases ✗ wakeup ✓
[lp-1  ] INFO 2023/02/05 17:11:57   meters:      charge ✓
[lp-1  ] INFO 2023/02/05 17:11:57     charge:    power ✓ energy ✓ currents ✓
[lp-1  ] DEBUG 2023/02/05 17:11:57 phase timer inactive
[lp-1  ] DEBUG 2023/02/05 17:11:57 pv timer inactive
[lp-1  ] DEBUG 2023/02/05 17:11:57 max charge current: 6A
[main  ] INFO 2023/02/05 17:11:57 ensure the oauth client redirect/callback is configured for Mercedes E300e: http://192.168.0.240:7070/oauth/vehicles/1/callback
[site  ] DEBUG 2023/02/05 17:11:57 ----
[server] FATAL 2023/02/05 17:11:57 unlinkat /tmp/evcc: operation not permitted
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.111.1

andig, 2023-02-05:
Hat deine Heidelberg einen Zähler?

JAK0721, 2023-02-06:
Hallo, die Heidelberg Wallbox erfasst eigenständig Daten, welche via Modbus ausgelesen werden können. Wie die EVCC Software das macht und die Verrechnung mit Meterdaten (Stromzähler) inkl. der Befüllung der InfluxDB, entzieht sich meiner Kenntnis.
Ich habe eine Grafana Visu am laufen, wo ich nur die Daten aus der InfluxDB auslesen.

andig, 2023-02-06:
Korrekt, die Heidelberg hat einen Zähler.

> Hausverbrauch geht schlagartig auf 350W zurück

Wie gerechnet wird steht im Logfile- da müsstest du einfach mal rein schauen. Aus Influx lässt sich zur Fehlersuche nichts raus interpretieren.

closed 2023-02-06 by andig