evcc-io/evcc #6086: go-e charger - 1p setting ignored since 0.112

DocBru, 2023-02-06:
### Describe the bug

I am having a "classic" go-e charger, which is physically connected on 3phases, but I am using a 1p charging cable so to being able to load solar power with low production in winter.

Therefore I have hardcoded the loadpoint to "1 phase" and it was working fine until the recent update.

I suppose that this PR breaks it: https://github.com/evcc-io/evcc/pull/5764 by overriding the loadpoint setting with something derived from connected voltages.

### Steps to reproduce

1. Start evcc in "pv" mode with loadpoint 1p
2. Check the logs

### Configuration details

```yaml
loadpoints:
  - title: go-e # display name for UI
    charger: goe # charger
    mode: pv
    phases: 1 # ev phases (default 3)
    enable: # pv mode enable behavior
      delay: 30s # threshold must be exceeded for this long
      threshold: 0 # minimum export power (W). If zero, export must exceed minimum charge power to enable
    disable: # pv mode disable behavior
      delay: 10m # threshold must be exceeded for this long
      threshold: 500 # maximum import power (W)
```

### Log details

```text
Startup:

Using config file: /config/evcc.yaml
starting evcc: 'evcc --config /config/evcc.yaml --sqlite /data/evcc.db'
[main  ] INFO 2023/02/06 09:29:30 evcc 0.112.4
[main  ] INFO 2023/02/06 09:29:30 using config file: /config/evcc.yaml
[lp-1  ] INFO 2023/02/06 09:29:30 loadpoint 1:
[lp-1  ] INFO 2023/02/06 09:29:30   mode:        pv
[lp-1  ] INFO 2023/02/06 09:29:30   charger:     power ✓ energy ✓ currents ✓ phases ✗ wakeup ✗
[lp-1  ] INFO 2023/02/06 09:29:30   meters:      charge ✓
[lp-1  ] INFO 2023/02/06 09:29:30     charge:    power ✓ energy ✓ currents ✓
[lp-1  ] DEBUG 2023/02/06 09:29:30 phase timer inactive
[lp-1  ] DEBUG 2023/02/06 09:29:30 pv timer inactive
[lp-1  ] DEBUG 2023/02/06 09:29:30 charge power: 0W
[lp-1  ] DEBUG 2023/02/06 09:29:31 charge voltages: [223 220 223]V
[lp-1  ] DEBUG 2023/02/06 09:29:31 detected connected phases: 3p
[lp-1  ] DEBUG 2023/02/06 09:29:31 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2023/02/06 09:29:31 charger status: B
[lp-1  ] INFO 2023/02/06 09:29:31 car connected
[lp-1  ] DEBUG 2023/02/06 09:29:31 pv timer elapse
[lp-1  ] DEBUG 2023/02/06 09:29:31 pv timer inactive
[lp-1  ] DEBUG 2023/02/06 09:29:31 pv charge current: 2.3A = 0A + 2.3A (-1590W @ 3p)
[lp-1  ] DEBUG 2023/02/06 09:29:31 pv enable timer reset
[lp-1  ] DEBUG 2023/02/06 09:29:31 pv timer inactive

After manual switch to "minpv" mode, the problem fixes itself, as it detects that only 1p is working:

[lp-1  ] DEBUG 2023/02/06 09:29:51 set charge mode: minpv
[lp-1  ] DEBUG 2023/02/06 09:29:51 charge power: 0W
[lp-1  ] DEBUG 2023/02/06 09:29:51 charge voltages: [224 219 222]V
[lp-1  ] DEBUG 2023/02/06 09:29:51 detected connected phases: 3p
[lp-1  ] DEBUG 2023/02/06 09:29:51 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2023/02/06 09:29:52 charger status: B
[lp-1  ] DEBUG 2023/02/06 09:29:52 pv charge current: 2.21A = 0A + 2.21A (-1527W @ 3p)
[lp-1  ] DEBUG 2023/02/06 09:29:52 max charge current: 6A
[lp-1  ] DEBUG 2023/02/06 09:29:52 charger enable
[lp-1  ] DEBUG 2023/02/06 09:29:52 wake-up timer: start
[lp-1  ] DEBUG 2023/02/06 09:30:00 charge power: 1350W
[lp-1  ] DEBUG 2023/02/06 09:30:01 charge voltages: [223 220 221]V
[lp-1  ] DEBUG 2023/02/06 09:30:01 detected connected phases: 3p
[lp-1  ] DEBUG 2023/02/06 09:30:01 charge currents: [6 0 0]A
[lp-1  ] DEBUG 2023/02/06 09:30:01 charger status: C
[lp-1  ] INFO 2023/02/06 09:30:01 start charging ->
[lp-1  ] DEBUG 2023/02/06 09:30:01 pv charge current: 6.2A = 6A + 0.204A (-141W @ 3p)
[lp-1  ] DEBUG 2023/02/06 09:30:01 max charge current: 6A
[lp-1  ] DEBUG 2023/02/06 09:30:03 set charge mode: pv
[lp-1  ] DEBUG 2023/02/06 09:30:03 charge power: 1330W
[lp-1  ] DEBUG 2023/02/06 09:30:03 charge voltages: [222 221 222]V
[lp-1  ] DEBUG 2023/02/06 09:30:03 detected connected phases: 3p
[lp-1  ] DEBUG 2023/02/06 09:30:03 charge currents: [5.9 0 0]A
[lp-1  ] DEBUG 2023/02/06 09:30:03 detected active phases: 1p
[lp-1  ] DEBUG 2023/02/06 09:30:03 charger status: C
[lp-1  ] DEBUG 2023/02/06 09:30:03 pv charge current: 6.87A = 6A + 0.867A (-200W @ 1p)
[lp-1  ] DEBUG 2023/02/06 09:30:03 max charge current: 6A
```

### What type of operating system are you running?

Docker container

### Version

0.112.4

DocBru, 2023-02-06:
Ja genau, ein einphasige Labekabel .
Eine Phasenumschaltung habe ich nicht.

In der Doku ist genau mein Szenario beschrieben:
https://docs.evcc.io/docs/reference/configuration/loadpoints/#phases

`Dieser Parameter ändert nichts am physikalischen Anschluss der Wallbox, sondern dient lediglich dazu, im PV-Modus (in Verbindung mit minCurrent) die minimale Startleistung für die Ladung zu ermitteln.`

DocBru, 2023-02-06:
Ich glaube der Fehler ist hier, wo die loadpoint phases einfach überschrieben werden, unabhängig davon was in der config stand und / oder ob in der config ein niedrigerer Wert definiert wurde als ermittelt.
https://github.com/evcc-io/evcc/blob/1a5e34d23d7e74646534b851e969856b92ad37a8/core/loadpoint.go#L1571-L1574

VolkerK62, 2023-02-06:
Das ist eigentlich als feature reingekommen. Automatische Erkennung anhand der anliegenden Spannungen.

DocBru, 2023-02-06:
> Das ist eigentlich als feature reingekommen. Automatische Erkennung anhand der anliegenden Spannungen.

Das verstehe ich und würde auch Sinn machen, wenn die loadpoint.phases nicht definiert werden.

Aus meiner Sicht macht es aber keinen Sinn den definierten Wert zu überschreiben, schon gar nicht "nach oben" (von 3 auf 1 runter wäre evtl. noch sinnvoll, weil technisch dann gar nicht möglich).

premultiply, 2023-02-06:
`phases` am Loadpoint gibt genau die netzseitig anliegenden Phasen an. Die automatische Erkennung ist somit (immer) richtig.
Eine manuelle Angabe macht nur bei einphasigem Netzanschluss ohne passendes Meter einen Sinn oder um eine Box mit automatischer Phasenumschaltung auf einen der beiden Mode festzulegen.

Hier wäre die Korrekte stelle die Angabe von `phases: 1` am Vehicle.

andig, 2023-02-07:
> phases am Loadpoint gibt genau die netzseitig anliegenden Phasen an.

@premultiply I disagree insofar as that it is also the phases the charger (!) will use for charging. We have not considered the cable scenario. We may of course chose to ignore it.

Optionally using a 1p cable is a perfectly viable scenario that the detection is breaking. Even if user sets `1` on loadpoint (which is correct for the cable) we would change that to `3` which is not what will happen. Also, we don't have an API for setting the value on vehicle level, so we're forcing a wrong update on the user.

premultiply, 2023-02-07:
Well, ok, but the loadpoint setting was never designed (intended) to respect the cable or vehicle.

There is no direct API to switch the setting on vehicle level but it is fine to define the same vehicle twice with different settings to easily switch between them.

andig, 2023-02-07:
Perfect workaround!

@DocBru solved?

@VolkerK62 something for the FAQ?

DocBru, 2023-02-07:
Hi, sorry what is the workaround? I missed it :-)

I do not have any vehicles defined for various reasons (did not detect properly in the past). Can I define the "guest vehicle" somehow as 1phased? That would work than.

DocBru, 2023-02-07:
By the way, at least the docu should be updated, because that was working for me 2 years long and is now not anymore:

`Dieser Parameter ändert nichts am physikalischen Anschluss der Wallbox, sondern dient lediglich dazu, im PV-Modus (in Verbindung mit minCurrent) die minimale Startleistung für die Ladung zu ermitteln.`

andig, 2023-02-07:
You can define 2 different offline vehicles with different phase setting and select that vehicle when you chance the cable.

andig, 2023-02-07:
> Dieser Parameter ändert nichts am physikalischen Anschluss der Wallbox, sondern dient lediglich dazu, im PV-Modus (in Verbindung mit minCurrent) die minimale Startleistung für die Ladung zu ermitteln.

Genauso ist es ja auch.

DocBru, 2023-02-07:
OK will try that, probably only need to do that once after restarting / updating evcc. Will it keep the setting after disconnecting?

DocBru, 2023-02-07:
> > Dieser Parameter ändert nichts am physikalischen Anschluss der Wallbox, sondern dient lediglich dazu, im PV-Modus (in Verbindung mit minCurrent) die minimale Startleistung für die Ladung zu ermitteln.
> 
> Genauso ist es ja auch.

Nein... Der Parameter macht jetzt nichts mehr in Verbindung mit dem minCurrent, er wird einfach ignoriert, da er von einer automatischen Erkennung überschrieben wird..

DocBru, 2023-02-07:
Ok folgendes funktioniert:

```
vehicles:
  - name: 1p
    type: template
    template: offline
    title: 1phasig
    phases: 1
loadpoints:
  - title: go-e # display name for UI
    vehicle: 1p
   ...
```

Die Umschaltung brauche ich nicht unbedingt, da ich das höchstens einmal im Jahr wechsle. Wichtiger ist dass er als default 1phasig nimmt und das geht so

Danke für die Tipps!

andig, 2023-02-07:
Super dass wir eine Lösung gefunden haben ohne den anderen Anwendern die automatische Umschaltung weg nehmen zu müssen :)

closed 2023-02-07 by andig

reach0775, 2023-03-24:
Bei mir läuft evcc seit ca. 2 Tagen nicht mehr (0.114.1 Nightly) 
Im GUI sehe ich immer die Meldung:
charge meter: Get \"http://10.0.0.6/api/status?filter=alw,car,eto,nrg,wh,trx,cards\": dial tcp 10.0.0.6:80: connect: no route to host

Ich habe eine go-e. Das hat aber trotzdem nichts mit diesem Problem hier zu tun, oder?

andig, 2023-03-24:
Kein zusammenhang- an Go-E wurde auch nichts geändert.