evcc-io/evcc #6095: Add 1p3p to Wattpilot implementation

premultiply, 2023-02-06:
Analog go-e-HTTP.

_Originally posted by @thecem in https://github.com/evcc-io/evcc/discussions/6056#discussioncomment-4879678_

andig, 2023-02-06:
Der Wattpilot benutzt ein Websocket API. Wie soll das analog Go-e gemacht werden? Durch wen (@mabunixda)?

mabunixda, 2023-02-06:
Also 1p/3p Umschaltung kann der Wattpilot ja manuell auch, allerdings kann er es eben auch selbst sofern er "Hausintern" mit einem Fronius-WP reden kann/konfiguriert ist. 
Ich denke, dass beides durchaus Sinn machen kann bzw Use Cases dafuer gibt:
* 1p3p durch evcc, weil ich eben mehrere Erzeuger habe und diese Produktion nutzen will
* Wattpilot basiert, weil man hier auf Fronius vertraut und auch nur einen WR hat 

Allerdings ist - soweit ich das verstehe - der 2.Fall aktuell nicht konfigurierbar @andig  ?

premultiply, 2023-02-06:
Das API und die Logik ist doch von den Kommandos her absolut identisch.
Nur den Transportweg HTTP hat man bei der Fronius-Firmware abgeklemmt.

andig, 2023-02-06:
> Allerdings ist - soweit ich das verstehe - der 2.Fall aktuell nicht konfigurierbar @andig ?

Ne, das geht nicht. Wir gehen ja auch immer davon aus, dass das Gegenstück "dumm" ist. Könnten wir Weg 1 per Standard einbauen und zur Not deaktivierbar machen (decorator)?

premultiply, 2023-02-06:
Ich gehe auch davon aus, dass man mit der WS-Implementierung "für den WP" auch genauso den go-e-Charger ansteuern kann.
Im Prinzip ist dann die bisherige HTTP-Implementierung obsolet.

andig, 2023-02-06:
Yo, das ist hier aber OT und ich möchte keine neue Baustelle mit einem nicht supporteten API aufmachen.

mabunixda, 2023-02-06:
Ja prinzpiell sollte das schon gehen, dass evcc hier Wattpilot steuert.

andig, 2023-02-06:
Dann lass uns das mal ausprobieren :)

closed 2023-02-06 by andig