evcc-io/evcc #6100: Docker container not updated?

hyperbart, 2023-02-06:
### Describe the bug

Is it possible there is something wrong with the Docker container build process?

I have a cronjob running every night to update to latest, but since a couple of days/weeks I seem to be stuck on 0.111.1?

### Steps to reproduce

Deploy container EVCC, check version, stays at 0.111.1

### Configuration details

```yaml
N/A
```

### Log details

```text
N/A
```

### What type of operating system are you running?

Linux

Caibuk, 2023-02-06:
You probably need to change the docker repository that you are using from andig/evcc to evcc/evcc. New docker images are only pushed to evcc/evcc since a couple of days (weeks?).

hyperbart, 2023-02-06:
Yeah... Okay, didn't know that one. Thanks @Caibuk .
/me out.

closed 2023-02-06 by hyperbart