evcc-io/evcc #6101: freeze beim Laden

VolkerK62, 2023-02-06:
### Describe the bug

Heute Morgen von 0.112.0. aufs aktuelle Nightly gewechselt. 
Heute Mittag ist mir aufgefallen, dass evcc während des Ladens eingefroren ist.
Ich habe dann ein downgrade von 1.112.4 auf 0.112.0 gemacht.
Danach hat alles wie gewohnt funktioniert.

Jetzt habe ich es jetzt 2-mal getestet.
Update auf aktuelles Nightly.

Erster Versuch (freeze_1.log)
Ladung im PV Modus durch Manipulation von `residualpower` gestartet und nach ein paar Minuten `residualpower` wieder hoch gesetzt. Disable Timer startet. Kurz danach frier evcc ein. Stop des Service hat die Sache dann beendet.

Zweiter Versuch (freeze_2.log)
Log auf trace gestellt. Start wie oben, allerdings kommt der freeze schon viel früher.

[freeze_2.log](https://github.com/evcc-io/evcc/files/10665061/freeze_2.log)
[freeze_1.log](https://github.com/evcc-io/evcc/files/10665062/freeze_1.log)

Bin jetzt wieder auf 0.112.0. zurück. 

### Steps to reproduce

.

### Configuration details

```yaml
network:
  schema: http
  host: evcc.local
  port: 7070

interval: 10s  

sponsortoken: ***

telemetry: true

  
log: debug
levels:
  #site: trace
  #lp-1: trace
  #daheim: trace
  #hyundai: trace
  #main: trace
  #cache: trace
  #db: trace
  #influx: trace
  #mqtt: trace
  #proxy-5200: trace
 
  
########################################################################
meters:
- name: grid
  type: custom
  power:
    source: http
    uri: http://192.168.178.28/status
    jq: .emeters | map(.power) | add
  energy:
    source: http
    uri: http://192.168.178.28/status
    jq: .emeters | map(.total) | add
    scale: 0.001
  currents:
  - source: http
    uri: http://192.168.178.28/emeter/0
    jq: if .power < 0 then - .current else .current end
  - source: http
    uri: http://192.168.178.28/emeter/1
    jq: if .power < 0 then - .current else .current end
  - source: http
    uri: http://192.168.178.28/emeter/2
    jq: if .power < 0 then - .current else .current end
  voltages:
  - source: http
    uri: http://192.168.178.28/emeter/0
    jq: .voltage
  - source: http
    uri: http://192.168.178.28/emeter/1
    jq: .voltage
  - source: http
    uri: http://192.168.178.28/emeter/2
    jq: .voltage
  powers:
  - source: http
    uri: http://192.168.178.28/emeter/0
    jq: .power
  - source: http
    uri: http://192.168.178.28/emeter/1
    jq: .power
  - source: http
    uri: http://192.168.178.28/emeter/2
    jq: .power

- name: pv
  type: template
  template: solarlog
  usage: pv
  host: 192.168.178.23   
  port: 502

 
- name: battery
  type: custom
  power:  # Shelly EM Stromzange 1
    source: http
    uri: http://192.168.178.21/emeter/0
    jq: if .pf==0 then .power else .power / (.pf | fabs) end
  soc:
    source: mqtt
    topic: Haus/Bat_SoC
    timeout: 60m
  capacity: 7.5  
  
    
########################################################################                
chargers:
- name: wallbox
  type: template
  template: daheimladen
  #host: 192.168.178.26
  #port: 502
  stationid: ***
  token: ***
  ##gültig bis 07/2023

 
   
#######################################################################
vehicles:
- name: auto
  type: template
  template: hyundai
  title: (T)Raumschiff
  icon: car      
  user: ***  
  password: ***     
  vin: ***
  capacity: 72
  language: de            
  phases: 3 
  ##onIdentify:
  mode: pv
  targetsoc: 90
  

- name: offline
  type: template
  template: offline
  title: offline
  capacity: 50
  mode: off

  
########################################################################    
loadpoints:
- title: Carport
  charger: wallbox
  vehicle: auto  # Standardfahrzeug
  mode: off
  phases: 1
  mincurrent: 6
  maxcurrent: 20
  resetOnDisconnect: true 
  #enable: #(Standard) 0W ; 1m
  #  threshold: 0
  #  delay: 1ms  
  #disable: #(Standard) 0W ; 3m
  #  threshold: 0    
  #  delay: 5m
  guardduration: 2m #(Standard) 5m
  soc:
    estimate: true  
    poll:
      mode: always
      interval: 240m
      
      
########################################################################  
site:
- title: Haus
  meters:
    grid: grid
    pv: pv
    battery: battery
  bufferSoc: 100 
  prioritySoc: 20
  residualpower: 50  # verbleibende Einspeisung als Puffer für Hausspeicher

########################################################################
tariffs:
  currency: EUR
  grid:
    type: fixed
    price: 0.424  # Preis je kWh Bezug
  feedin:
    type: fixed
    price: 0.1095  # Preis je kWh Einspeisung
  planner:
    type: grünstromindex
    zip: ***

#######################################################################    
influx:
  url: http://localhost:8086
  database: evcc
  
    
mqtt:
  broker: localhost:1883
  topic: evcc
       

#modbusproxy:
#   - port: 5200
#     uri: 192.168.178.26:502
       
```
```

### Log details

```text
.
```

### What type of operating system are you running?

Linux

### Version

evcc 0.112.4 (bbf10833)

andig, 2023-02-06:
Closed in https://github.com/evcc-io/evcc/commit/7faa8bb8e419da00d5008ae0e3be437258d5e7f8. Das ist wirklich maximal hässlich. Ich lasse es jetzt trotzdem beim "schönen" Code; mit diesem Pflaster sollte das hoffentlich geheilt sein.

andig, 2023-02-06:
Nightly in 15min

closed 2023-02-06 by andig

VolkerK62, 2023-02-06:
Bin immer wieder erstaunt, wie du das aus dem log erkennst

VolkerK62, 2023-02-06:
getestet und für gut befunden. Danke