evcc-io/evcc #6102: Einheiten bei Tesla Powerwall 2

Fantus62, 2023-02-06:
Hallo Community,

Die Einheiten für Batteriekapazität  und neuerdings auch des PV-Energiemessers stimmen nicht.  Statt kWh entspricht der jeweils angezeigt Zahlenwert Wh.

evcc dump liefert folgendes Ergebnis:

grid: grid1
-----------
Power: 19W

pv 0: pv2
---------
Power:  -1W
Energy: 466278.6kWh

battery 0: battery3
-------------------
Power:    630W
Soc:      42%
Capacity: 14923.0kWh

Seit meinem Update heute Morgen habe ich demzufolge heute 90.000€ mit Solarstrom gespart :-)

Grüße
Stephan

andig, 2023-02-06:
> Seit meinem Update heute Morgen habe ich demzufolge heute 90.000€ mit Solarstrom gespart :-)

Das klingt doch gut :)

andig, 2023-02-06:
Kannst Du den PR testen? Ich bin noch auf `usage: load` gestossen, bin aber unsicher was die bedeutet? Ist mir korrigiert. Für `grid` ist derzeit keine Energiemenge implementiert- gäbs die auch?

andig, 2023-02-08:
ping @Fantus62

andig, 2023-02-08:
/cc @LeStef101 könntest Du das evtl. auch mal anschauen?

Fantus62, 2023-02-08:
ich hatte direkt auf die E-Mail geantwortet. Ist wohl nicht angekommen.

Leider habe ich keine Ahnung, wie ich den PR teste. Die Änderungen sind offensichtlich nicht Bestandteil des Hauptzweiges. Da sind die Zahlen noch falsch. 

Ich habe auch versucht eine Entwicklungsumgebung unter UBUNTU aufzusetzen. Leider gibt es Fehler beim make.

Habe leider keine Informationen, ob die Powerwall auch eine Energiemenge liefert. Ich habe derweil auf den SMA Home Manager umgestellt, der bei mir auch noch installiert ist.

Fantus62, 2023-02-08:
Meine Auswertung der Ladeenergie ist durch den Fehler komplett daneben. Sonnenenergie 99% Energiepreis 7,7ct/kWh Ersparnis 90.799,00€. Habe schon versucht evcc.db zu löschen, was aber nichts verändert hat. Wie kann ich diese Zähler zurücksetzen?

andig, 2023-02-08:
> ich hatte direkt auf die E-Mail geantwortet. Ist wohl nicht angekommen.

Ne, die scheint bei GH nicht angekommen zu sein. Du kannst in 15min im Nightly testen, ich hab den PR gemerged. Wenn er falsch sein sollte muss ich nacharbeiten ;)

closed 2023-02-08 by andig

Fantus62, 2023-02-08:
Zahler sehen jetzt gut aus:

pvTESLA
-------
Power:  4379W
Energy: 509.5kWh

batteryTESLA
------------
Power:    -3570W
Soc:      83%
Capacity: 14.9kWh

Vielen Dank!

Hast Du noch einen Tip, wie ich die Statistikzähler zurückgesetzt bekomme?

 Vielen

Fantus62, 2023-02-08:
> Hast Du noch einen Tip, wie ich die Statistikzähler zurückgesetzt bekomme?

Hat sich erledigt. evcc.db war woanders als ich dachte.