evcc-io/evcc #6106: Lastabwurfpotenzialzähler

andig, 2023-02-06:
> Jetzt habe ich die Situation, dass der Heizstab mit 3 KW vor sich hin heizt und das Auto angeschlossen wird. Dieses bekommt nun nicht die volle Leistung, weil der Heizstab (im Hausverbrauch) schon 3 KW nimmt. Diese 3 KW sind für mich optional. Gewünscht wäre, dass die Wallbox sich auch aus diesen 3 KW bedienen darf.

Interessanter Anwendungsfall. Das gibts heute nix, würde aber gut in die Struktur passen als eine Art "Lastabwurfpotenzialzähler"?

_Originally posted by @andig in https://github.com/evcc-io/evcc/discussions/3201#discussioncomment-4886360_

andig, 2023-02-06:
Dieses Szenario liesse sich unterstützen:

- site bekommt zusätzliche(n) Zähler für "intelligente" Lasten
- die verfügbare Netzleistung wird um diesen Betrag korrigiert

@webwanze im Prinzip ist das nix anderes als eine Batterie mit capacity=soc=0. Könntest Du mal versuchen, über Plugins zu realisieren.

/cc @premultiply

Webwanze, 2023-02-07:
Hallo,
könnte man wohl als 'Spezialbatterie' fahren. Dann stimmen aber die Werte in der GUI nicht mehr. 
Ähnlicher Effekt, wie wenn man den PV-Ertrag um die 'Intelligente Leistung' nach oben verfälschen würde. Wird auch funktionieren, aber die Anzeige wäre nonsens.

Aber die Idee mit einem zusätzlichen Bucket/Zähler für 'Intelligente Lasten' finde ich gut.

premultiply, 2023-02-07:
Die Idee einen zusätzlichen "anderen" Lasttyp an der Site hinzuzufügen finde ich prinzipiell gut. Kann aber halt nur rein passiv zur Messung zur Ermittlung des Leistungspotenzials dienen.

Alternative wäre einen zweiten Typ Ladepunkt einzuführen der dann auch zusätzlich entsprechend steuern könnte.
Ist natürlich ungleich komplexer aber auch entsprechend flexibler.

Webwanze, 2023-02-07:
Ja, könnte ich dann testen.

Diese 'Intelligenten Lasten' könnte man dann unter einem neuen Zähler saldieren. 
Der Zähler am Netzübergabepunkt hat diese ja eh schon in seinem Saldo drin. Da würde ich auch nichts ändern.
d.h. lediglich bei der Ermittlung der aktuell möglichen Ladeleistung würde der neue Zähler herangezogen werden. 

Die Anzeige würde sich gar nicht verändern. Oder findet es jemand sinnvoll, diese 'Intelligenten Lasten' oder 'Optionalen Lasten' anzuzeigen? Dann müsste man nämlich weitere Werte in der GUI manipulieren damit die Summen wieder stimmen...

andig, 2023-02-07:
@Webwanze kann losgehen :)

Webwanze, 2023-02-07:
@andig 
Wie kriege ich das hier zum Laufen?

andig, 2023-02-09:
Du kannst in 30min das nightly probieren- bin gespannt :)

closed 2023-02-09 by andig

Webwanze, 2023-02-09:
Hallo,

habe nun mal erste Tests unternommen. Schaut gut aus :-)

Ich habe den smart-Zähler in die config eingebaut. Dieser holt sich die Leistung über http von iobroker. Da habe ich aktuell 3000 W fix eingetragen.

```
[site  ] DEBUG 2023/02/09 08:53:39 ----
[lp-1  ] DEBUG 2023/02/09 08:53:39 charge power: 0W
[lp-2  ] DEBUG 2023/02/09 08:53:39 charge power: 0W
[site  ] DEBUG 2023/02/09 08:53:39 pv power: 5026W
[site  ] DEBUG 2023/02/09 08:53:39 battery soc: 76%
[site  ] DEBUG 2023/02/09 08:53:39 battery power: -4420W
[site  ] DEBUG 2023/02/09 08:53:39 grid power: 8W
[site  ] DEBUG 2023/02/09 08:53:39 grid powers: [-3297 336 2968]W
[site  ] DEBUG 2023/02/09 08:53:39 grid currents: [-14.3 1.8 13]A
[site  ] DEBUG 2023/02/09 08:53:39 smart load power: 3000W
[site  ] DEBUG 2023/02/09 08:53:39 site power: -7412W
[lp-1  ] DEBUG 2023/02/09 08:53:39 charger status: A
[site  ] DEBUG 2023/02/09 08:53:49 ----
```

Rechnung für site:
4420 W battery
3000 W smart
- 8 W grid
= 7412 W site
**Stimmt also zu 100%**

[image]
**Die GUI-Werte scheinen mir auch stimmig zu sein.**

Werde nun mal echte Daten vom Heizstab holen. Später ist dann auch der Hausakku voll und das E-Auto wieder da. Dann kann ich nochmal nachsehen.

Grüße,

Thomas

Webwanze, 2023-02-09:
Macht es Sinn, mehrere smart-Zähler zu unterstützen? 
Analog battery:/batteries: unter site?

andig, 2023-02-09:
Ich wusste, dass Du sowas fragen würdest :O. Ja machen wir, ich muss das dafür aber nochmal umbauen. Das neue Pattern wäre "automatisch" mehrere Zähler zuzulassen und dafür wieder die alten Namen `pv`, `battery` zu verwenden. Überlege nur noch ob ich das wirklich jemandem erklären will :O

Webwanze, 2023-02-09:
Gerade wurde das E-Auto angesteckt. 
Laden hat auch wie gewünscht begonnen. Der Heizstab hat daraufhin sofort auf 3W = Ruheleistung runtergeregelt. 

GUI zeigt keine Auffälligkeiten.

Passt so!

```
[site  ] DEBUG 2023/02/09 11:46:53 ----
[lp-1  ] DEBUG 2023/02/09 11:46:53 charge power: 0W
[lp-2  ] DEBUG 2023/02/09 11:46:53 charge power: 0W
[site  ] DEBUG 2023/02/09 11:46:53 pv power: 7483W
[site  ] DEBUG 2023/02/09 11:46:53 battery soc: 100%
[site  ] DEBUG 2023/02/09 11:46:53 battery power: -2350W
[site  ] DEBUG 2023/02/09 11:46:53 grid power: -1619W
[site  ] DEBUG 2023/02/09 11:46:53 grid powers: [-4307 248 2440]W
[site  ] DEBUG 2023/02/09 11:46:53 grid currents: [-19.4 1.4 10.7]A
[site  ] DEBUG 2023/02/09 11:46:53 smart load power: 2967W
[site  ] DEBUG 2023/02/09 11:46:53 site power: -6936W
[lp-1  ] DEBUG 2023/02/09 11:46:53 charger status: B
[lp-1  ] INFO 2023/02/09 11:46:53 car connected
[lp-2  ] INFO 2023/02/09 11:46:53 vehicle updated: VW e-Golf 300 -> unknown
[lp-1  ] INFO 2023/02/09 11:46:53 vehicle updated: unknown -> VW e-Golf 300
[lp-1  ] DEBUG 2023/02/09 11:46:53 set charge mode: pv
[lp-1  ] DEBUG 2023/02/09 11:46:53 set min soc: 0
[lp-1  ] DEBUG 2023/02/09 11:46:53 set target soc: 90
[lp-1  ] DEBUG 2023/02/09 11:46:53 pv timer elapse
[lp-1  ] DEBUG 2023/02/09 11:46:53 pv timer inactive
[script] DEBUG 2023/02/09 11:46:53 /bin/sh -c sudo /usr/local/bin/soc.sh: 59.58
[lp-1  ] DEBUG 2023/02/09 11:46:53 vehicle soc: 59%
[lp-1  ] DEBUG 2023/02/09 11:46:53 pv charge current: 15.1A = 0A + 15.1A (-6936W @ 2p)
[lp-1  ] DEBUG 2023/02/09 11:46:53 site power -6936W <= 0W enable threshold
[lp-1  ] DEBUG 2023/02/09 11:46:53 pv enable in 0s
[lp-1  ] DEBUG 2023/02/09 11:46:53 pv enable timer elapsed
[lp-1  ] DEBUG 2023/02/09 11:46:53 max charge current: 6A
[lp-1  ] DEBUG 2023/02/09 11:46:53 charger enable
[lp-1  ] DEBUG 2023/02/09 11:46:53 wake-up timer: start
[site  ] DEBUG 2023/02/09 11:46:53 ----
[lp-1  ] DEBUG 2023/02/09 11:46:53 charge power: 0W
[lp-2  ] DEBUG 2023/02/09 11:46:53 charge power: 0W
[site  ] DEBUG 2023/02/09 11:46:53 pv power: 7483W
[site  ] DEBUG 2023/02/09 11:46:53 battery soc: 100%
[site  ] DEBUG 2023/02/09 11:46:53 battery power: -2350W
[site  ] DEBUG 2023/02/09 11:46:53 grid power: -1619W
[site  ] DEBUG 2023/02/09 11:46:53 grid powers: [-4307 248 2440]W
[site  ] DEBUG 2023/02/09 11:46:53 grid currents: [-19.4 1.4 10.7]A
[site  ] DEBUG 2023/02/09 11:46:53 smart load power: 2967W
[site  ] DEBUG 2023/02/09 11:46:53 site power: -6936W
[lp-1  ] DEBUG 2023/02/09 11:46:53 charger status: B
[lp-1  ] DEBUG 2023/02/09 11:46:53 next soc poll remaining time: 59s
[lp-1  ] DEBUG 2023/02/09 11:46:53 pv charge current: 15.1A = 0A + 15.1A (-6936W @ 2p)
[lp-1  ] DEBUG 2023/02/09 11:46:53 pv timer reset
[lp-1  ] DEBUG 2023/02/09 11:46:53 pv timer inactive
[lp-1  ] DEBUG 2023/02/09 11:46:53 max charge current: 15A

[site  ] DEBUG 2023/02/09 11:47:03 ----
[lp-1  ] DEBUG 2023/02/09 11:47:03 charge power: 0W
[lp-2  ] DEBUG 2023/02/09 11:47:03 charge power: 0W
[site  ] DEBUG 2023/02/09 11:47:03 pv power: 7500W
[site  ] DEBUG 2023/02/09 11:47:03 battery soc: 100%
[site  ] DEBUG 2023/02/09 11:47:03 battery power: -2330W
[site  ] DEBUG 2023/02/09 11:47:03 grid power: -29W
[site  ] DEBUG 2023/02/09 11:47:03 grid powers: [-960 3567 -2636]W
[site  ] DEBUG 2023/02/09 11:47:03 grid currents: [-6.25 15.7 -11.4]A
[site  ] DEBUG 2023/02/09 11:47:03 smart load power: 3W
[site  ] DEBUG 2023/02/09 11:47:03 site power: -2362W
[lp-2  ] DEBUG 2023/02/09 11:47:03 charge currents: [0 0 0]A
[lp-2  ] DEBUG 2023/02/09 11:47:03 charger status: A

[site  ] DEBUG 2023/02/09 11:47:13 ----
[... 50 lines cut ...]
[site  ] DEBUG 2023/02/09 11:47:43 battery soc: 100%
[site  ] DEBUG 2023/02/09 11:47:43 battery power: -30W
[site  ] DEBUG 2023/02/09 11:47:43 grid power: -72W
[site  ] DEBUG 2023/02/09 11:47:43 grid powers: [-760 3766 -3078]W
[site  ] DEBUG 2023/02/09 11:47:43 grid currents: [-5.55 16.5 -13.3]A
[site  ] DEBUG 2023/02/09 11:47:43 smart load power: 3W
[site  ] DEBUG 2023/02/09 11:47:43 site power: -105W
[lp-2  ] DEBUG 2023/02/09 11:47:43 charge currents: [0 0 0]A
[lp-2  ] DEBUG 2023/02/09 11:47:43 charger status: A
[site  ] DEBUG 2023/02/09 11:47:53 ----
[lp-1  ] DEBUG 2023/02/09 11:47:53 charge power: 7360W
[lp-2  ] DEBUG 2023/02/09 11:47:53 charge power: 0W
[site  ] DEBUG 2023/02/09 11:47:53 pv power: 7507W
[site  ] DEBUG 2023/02/09 11:47:53 battery soc: 100%
[site  ] DEBUG 2023/02/09 11:47:53 battery power: -30W
[site  ] DEBUG 2023/02/09 11:47:53 grid power: -9W
[site  ] DEBUG 2023/02/09 11:47:53 grid powers: [-754 3786 -3040]W
[site  ] DEBUG 2023/02/09 11:47:53 grid currents: [-5.5 16.7 -13.1]A
[site  ] DEBUG 2023/02/09 11:47:53 smart load power: 3W
[site  ] DEBUG 2023/02/09 11:47:53 site power: -42W
[lp-1  ] DEBUG 2023/02/09 11:47:53 charger status: C
[lp-1  ] DEBUG 2023/02/09 11:47:53 next soc poll remaining time: 39s
[lp-1  ] DEBUG 2023/02/09 11:47:53 vehicle soc: 59%
[lp-1  ] DEBUG 2023/02/09 11:47:53 pv charge current: 16.1A = 16A + 0.0915A (-42W @ 2p)
```

andig, 2023-02-09:
Nice :)

andig, 2023-02-09:
/cc @naltatis ggf. könnten wir auch diese Zähler noch anzeigen- auf der Konsumseite wäre ja noch Platz ;)

premultiply, 2023-02-09:
Nur den Namen "Smart Loads" finde ich noch nicht optimal.

Webwanze, 2023-02-09:
Optional Loads / Optional Consumers ?
Surplus Loads / Surplus Consumers ?

Müssen aber ja nicht unbedings Lasten sein. Ich könnte hier auch mitteilen, welcher zusätzliche PV Ertrag gerade abgerufen werden könnte. 
UseCase 1: Zero-Export Anlage und PV abgeregelt. Bringt gerade 1 KW, könnte aber (kann man berechnen) 3 KW bringen. Dann könnte ich die 2 zusätzlich möglichen KWs hier 'anmelden'.
UseCase 2: Ich möchte eine gewisse Leistung aus dem Hausakku erlauben. Steuerung über iobroker etc. z.B. 2 KW

Wie wäre es mit
**Power Reserve** ?
Das würde ja auch für 'Loads' passen.

andig, 2023-02-09:
Die Use cases sind spannend. Eigentlich ist das ja eine Arbeitspunktverschiebung in beliebiger Richtung. Man könnte es also auch `offset` oder `bias` nennen.

premultiply, 2023-02-09:
Wie wäre es dann mit Auxiliary Loads bzw. kurz AuxLoads?

Webwanze, 2023-02-09:
'Loads' würde ich weglassen.
aux
optional
reserve
offset

premultiply, 2023-02-09:
Aux. Kurz und bündig. 😉

andig, 2023-02-09:
`aux` es ist.

VolkerK62, 2023-02-09:
Mal eine Frage:
Ich würde es nutzen wollen, um z.B. die Ladeleistung meines Hausspeicher zu steuern.
PrioritySoc steht auf 100%. Maximale Ladeleistung des Speicher beträgt 1200W. 
Ich würde den Wert bei `aux` auf 500W festsetzen.
Dann würde das Auto diese 500W mit zur Ladung verwenden.
Dadurch bedingt würde der Hausspeicher zurückfahren und nur noch mit 700W laden.
Habe ich das Prinzip richtig verstanden?

Man konfiguriert ganz normal ein Meter und legt in `site` die Verwendung als `aux` fest.
Kann man dieses meter dann auch als "charge"-meter für einen Loadpoint definieren? 
Der wäre dann sozusagen "nachrangig".

mafde2023, 2023-04-02:
Hallo zusammen,

habe genau den gleichen Anwendungsfall. Wie binde ich den Heizstab jetzt genau ein bzw. wie muss ich die Konfig an der Site vornehmen um den AUX Verbrauch einzubinden? My-pv ELWA 3kW Heizstab ist vorhanden.

andig, 2023-04-02:
Bitte die "howtos" in Discussions verlegen.