evcc-io/evcc #6109: Tariff Planner: Request failed with status code 400

thierolm, 2023-02-06:
### Describe the bug

Beim Aufruf der des Ziel-Ladedialoges im UI wird folgende Fehlermeldung in der UI angezeigt:

[image]

`Request failed with status code 400: API request failed https://xxx.xxx.xxx/api//tariff/planner`

Hier ist offensichtlich ein Slash zuviel zwichen `api` und `tariff`.

Mögliche Ursache:
https://github.com/evcc-io/evcc/blob/30e1a4a6309f891b15b580bbb95ff828d647e9e9/assets/js/components/TargetCharge.vue#L254

### Steps to reproduce

1. Ziel-Ladedialog an einem Loadpoint anklicken
2. Fehlermeldung anschauen :-)

### Configuration details

```yaml
-
```

### Log details

```text
-
```

### What type of operating system are you running?

Linux

### Version

0.112.5

thierolm, 2023-02-06:
Da ich nicht der UI (vue) Held bin, habe ich nicht direkt einen PR aufgemacht ...

andig, 2023-02-07:
Der sieht auch falsch aus:

						await api.get(`/loadpoints/${this.id}/target/plan`, opts)

@naltatis könntest Du mal rein schauen? Macht es Sinn im Loadpoint beim API nochmal `loadpoints` zu prefixen oder passiert das schon weiter oben?

naltatis, 2023-02-07:
@thierolm Mit welchem Browser tritt das bei dir auf? Hast du evcc mit Base-Path oder direkt im Root Content laufen? Ich kann den Fehler bei mir nicht nachstellen (hab allerdings noch nen anderen gefunden :D ). Der `/` den du gefunden ist ist auf jeden Fall redundant. Ich hab den im PR rausgenommen. Vielleicht löst das dein Problem ja.

thierolm, 2023-02-07:
Das tritt 
1. bei mir bei der aktuellen Docker Version 0.112.5 auf und
2. auf dem PI von von solarfux, auf dem raspian läuft und evcc über  das apt repo geladen wird. Aktuell ist da noch die 0.112.4 am laufen ... 

Ich nutze Chrome als Browser

Maddin2020, 2023-02-24:
Also ich habe auch den Fehler im Docker drin kann das sein das der da noch gefixxt werden muss. Ich benutze auch Chrome oder Edge. Der Fehler tritt leider bei beiden auf.

andig, 2023-02-24:
Bitte aktuelle Version nutzen

Maddin2020, 2023-02-24:
Leider ist der Fehler noch nicht weg

Maddin2020, 2023-02-24:
[image]

andig, 2023-02-24:
Bitte neues Issue. Hat hiermit nix zuntun.

PR #6116 (2023-02-07): Fixed API paths for target charging
fixes #6109
closed 2023-02-24 by andig