evcc-io/evcc #6111: OCPP Bad Gateway

caol-ila, 2023-02-06:
### Describe the bug

Hi, ich habe EVCC als Add On in Home Assistant installiert und soweit läuft es. Ich habe eine Entratek Wallbox bestellt und möchte diese über OCPP steuern. Sobald ich in der Konfiguration 

```
chargers:
  - name: my_charger
    type: template
    template: ocpp
    meter: false # Benutze den integrierten Zähler (falls vorhanden) # Optional 
```
hinzufüge, startet EVCC nicht mehr richtig und ich bekomme immer 502 Bad Gateway von Home Assistant angezeigt. Wenn ich es wieder aus der Konifguration entferne startet es normal.

### Steps to reproduce

1. EVCC als Add On für Home Assistant installieren
2. Standard Konfiguration von OCPP aus der Doku in die Konfiguration einfügen
3.EVCC in Home Assistant öffnen

### Configuration details

```yaml
network:
  # schema is the HTTP schema
  # setting to `https` does not enable https, it only changes the way URLs are generated
  schema: http
  # host is the hostname or IP address
  # if the host name contains a `.local` suffix, the name will be announced on MDNS
  # docker: MDNS announcements don't work. host must be set to the docker host's name.
  host: bergzentrale
  # port is the listening port for UI and api
  # evcc will listen on all available interfaces
  port: 7070

interval: 10s # control cycle interval

# database configuration for persisting charge sessions and settings
# database:
#   type: sqlite
#   dsn: <path-to-db-file>

# sponsor token enables optional features (request at https://sponsor.evcc.io)
# sponsortoken:

# telemetry enables aggregated statistics
#
# Telemetry allows collecting usage data (grid and green energy, charge power).
# Data is aggregated, no individual charging sessions are tracked. The collected,
# anonymous data can be retrieved using https://api.evcc.io.
#
# See https://github.com/evcc-io/evcc/pull/4343 or details.
#
# For time being, this is only available to sponsors, hence data is associated with
# the sponsor token's identity.
#
# telemetry: true

# log settings
log: info
levels:
  site: debug
  lp-1: debug
  lp-2: debug
  cache: error
  db: error

# modbus proxy for allowing external programs to reuse the evcc modbus connection
# each entry will start a proxy instance at the given port speaking Modbus TCP and
# relaying to the given modbus downstream device (either TCP or RTU, RS485 or TCP)

[... 162 lines cut ...]
  # - type: telegram
  #   token: # bot id
  #   chats:
  #   - # list of chat ids
  # - type: email
  #   uri: smtp://<user>:<password>@<host>:<port>/?fromAddress=<from>&toAddresses=<to>
  # - type: ntfy
  #   uri: https://<host>/<topics>
  #   priority: <priority>
  #   tags: <tags>
```

### Log details

```text
Using config file: /config/evcc.yaml
starting evcc: 'evcc --config /config/evcc.yaml --sqlite /data/evcc.db'
[main  ] INFO 2023/02/07 00:06:33 evcc 0.112.5
[main  ] INFO 2023/02/07 00:06:33 using config file: /config/evcc.yaml
[main  ] INFO 2023/02/07 00:06:33 starting ui and api at :7070
```

### What type of operating system are you running?

Docker container

andig, 2023-02-07:
Please upgrade and restart the HA instance. This is not related to OCPP.

closed 2023-02-07 by andig

caol-ila, 2023-02-08:
@andig I'm on the latest version of ha (2023.2.3). If I comment out the ocpp and use something else instead, it is working well. Es soon as ocpp gets added, it stops working. Could this be related to the Docker config of the plugin an maybe it is not able to open the ocpp port?

andig, 2023-02-08:
It could be anything. Retry without HA and docker.