evcc-io/evcc #6112: loadpoint target time not writable

Matze4711, 2023-02-07:
### Describe the bug

Hi,
I tried to write
evcc/loadpoints/<id>/target/time/<time>: loadpoint target time in RFC3339 / ISO format (writable)
as described [here](https://docs.evcc.io/docs/reference/api/#mqtt-api). 
But there was no change in evcc. Is this API already active? 
Could you give me a concrete example? Maybe I do something wrong. 
Thanks!

### Steps to reproduce

1. publish evcc/loadpoints/1/target/time/1/set 
as mqtt

### Configuration details

```yaml
uri: 192.168.1###### # uri for ui
#network: 192.168.1####### # uri for ui
interval: 3s # control cycle interval

#Sponsor Token
sponsortoken: ##########

# mqtt message broker
mqtt:
  broker: 192.168.1#######
  topic: evcc # root topic for publishing, set empty to disable publishing
  user: #######
  password: "######"

# log settings
log: error
levels:
  core: error
  lp-1: error
  lp-2: error
  sma: error

tariffs:
  currency: EUR # (default EUR)
  grid:
    # static grid price
    type: fixed
    price: 0.32 # [currency]/kWh

  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.065 # [currency]/kWh

chargers:
  - name: WarpLinks
    type: template
    template: tinkerforge-warp
    fw2: true # Firmware v2 installiert
    host: localhost # IP Adresse oder der Hostname des MQTT Brokers
    port: 1883 # MQTT Broker Port # Optional
    user: ######
    password: "######"
    topic: warp/Links # Topic (ohne / am Anfang) # Optional
    timeout: 20s # Akzeptiere keine Daten die älter als dieser Wert sind # Optional 
    
 
  - name: WarpRechts
    type: template
    template: tinkerforge-warp
    fw2: true # Firmware v2 installiert
    host: localhost # IP Adresse oder der Hostname des MQTT Brokers
    port: 1883 # MQTT Broker Port # Optional
    user: ######
    password: "#######"
    topic: warp/Rechts # Topic (ohne / am Anfang) # Optional
    timeout: 20s # Akzeptiere keine Daten die älter als dieser Wert sind # Optional 
    
    
vehicles:
 - type: template
   template: opel
   title: Corsa e # Wird in der Benutzeroberfläche angezeigt # Optional
   user: ######## # Benutzerkonto (bspw. E-Mail Adresse, User Id, etc.)
   password: ###### # Passwort des Benutzerkontos (bei führenden Nullen bitte in einfache Hochkommata setzen)
   vin: ####### # Erforderlich, wenn mehrere Fahrzeuge des Herstellers vorhanden sind # Optional
   capacity: 50 # Akku-Kapazität in kWh # Optional 
   targetSoC: 85
#   minSoC: 40
   name: Corsa

 - type: template
   template: id
   title: VW ID4 # Wird in der Benutzeroberfläche angezeigt # Optional
   user: ######## # Benutzerkonto (bspw. E-Mail Adresse, User Id, etc.)
   password: ####### # Passwort des Benutzerkontos (bei führenden Nullen bitte in einfache Hochkommata setzen)
   vin: ######## # Erforderlich, wenn mehrere Fahrzeuge des Herstellers vorhanden sind # Optional
   capacity: 77 # Akku-Kapazität in kWh # Optional 
   targetSoC: 85   
#   minSoC: 50 
   name: ID4
     
  
meters:

  - name: sma-HM2
    type: template
    template: sma-home-manager
    usage: grid
    host: 192.168.##### # IP-Adresse oder Hostname 
  
  - name: WR6
    type: custom
    power:   
        source: mqtt
        topic:  home-assistant/SMA/STP6/Power
        timeout: 20s # don't accept values older than timeout
        scale: 1 # floating point factor applied to result, e.g. for Wh to kWh conversion
        
  - name: WR8
    type: custom
    power:   
        source: mqtt
        topic:  home-assistant/SMA/STP8/Power
        timeout: 20s # don't accept values older than timeout
        scale: 1 # floating point factor applied to result, e.g. for Wh to kWh conversion           
  
  - name: GrowattHuette
    type: custom
    power:   
        source: mqtt
        topic:  home-assistant/Growatt/Huette/Power
        timeout: 20s # don't accmept values older than timeout
        scale: 1 # floating point factor applied to result, e.g. for Wh to kWh conversion   
   
    
    

site:
- title: Zuhause # display name for UI#
  residualPower: 100
  meters: 
    grid: sma-HM2
    pvs: 
    - WR8
    - WR6
    - GrowattHuette
      
loadpoints:
  - title: Garage
    charger: WarpLinks
    mode: pv
    maxCurrent: 32
    vehicle: Corsa
    resetOnDisconnect: true
    phases: 1
    soc:
      estimate: true
    enable:
      threshold: 0
      delay: 70s      

  - title: Einfahrt 
    charger: WarpRechts
    mode: pv
    maxCurrent: 16
    vehicle: ID4
    resetOnDisconnect: true    
    phases: 3
    soc:
      estimate: true    
    enable:
      threshold: 0
      delay: 1m
```

### Log details

```text
Using config file: /config/evcc.yaml
starting evcc: 'evcc --config /config/evcc.yaml --sqlite /data/evcc.db'
[main  ] INFO 2023/02/07 07:22:48 evcc 0.112.5
[main  ] INFO 2023/02/07 07:22:48 using config file: /config/evcc.yaml
```

### What type of operating system are you running?

Docker container

### Version

0.112.5

andig, 2023-02-07:
> evcc/loadpoints//target/time

wrong path, missing loadpoint id

Matze4711, 2023-02-07:
@andig:
Could you please post an mqtt example to set the time or target soc? 
I tried 
evcc/loadpoints/1/target/time/1/set

Matze4711, 2023-02-07:
@andig:
Thanks again.
I tried over MQTT these:
- evcc/loadpoints/2/target/soc
 with payload 90
- evcc/loadpoints/2/target/soc/set
 with payload 90
- evcc/loadpoints/2/target/soc/90
with no payload

None of them worked. 
Only this one:
evcc/loadpoints/2/targetSoc/set
with payload 85 (or other value)

I would like to set the target time and soc over MQTT API.

andig, 2023-02-08:
As I said: just check what the ui does

andig, 2023-02-08:
Actually, sorry! I realise that mqtt is not en par with REST

andig, 2023-02-08:
@naltatis restbhas a hierarchical structure now that doesn‘t match mqtt where the setter is always adjacent to the getter. Fix the docs (plus add missing mqtt endpoints) or do something more complicated?

andig, 2023-02-08:
The mqtt docs are wrong:

```
evcc/loadpoints/<id>/target/soc/<soc>: loadpoint target SoC in % (writable)
evcc/loadpoints/<id>/target/energy/<energy>: loadpoint target energy in kWh (writable)
evcc/loadpoints/<id>/target/time/<time>: loadpoint target time in RFC3339 / ISO format (writable)

```
should be

```
evcc/loadpoints/<id>/targetSoc: loadpoint target SoC in % (writable)
evcc/loadpoints/<id>/targetEnergy: loadpoint target energy in kWh (writable)
evcc/loadpoints/<id>/targetTime: loadpoint target time in RFC3339 / ISO format (writable)
```

andig, 2023-02-08:
Fixed docs in https://github.com/evcc-io/docs/commit/050fd7b3fd6851d0776d55388c904186ba71383f

closed 2023-02-08 by andig