evcc-io/evcc #6113: Value Energy wrong for Shelly EM with template shelly-1pm

SwaY88LA, 2023-02-07:
### Describe the bug

See discussion https://github.com/evcc-io/evcc/discussions/6089

Using a Shelly EM (the simple 2 Channel version) with the template shelly-1pm will give the correct "power" value, but the wrong "energy" value

[image]

If I use the template shelly-3em the values "power" and "energy" are correct, but of course there are errors because of missing values, the simple EM can not provide compared to 3EM

[image]

Shelly Output:

Shelly Pro 1PM http://192.168.178.64/rpc/shelly.getstatus
[Shelly1PM_status.txt](https://github.com/evcc-io/evcc/files/10672556/Shelly1PM_status.txt)

Shelly EM http://192.168.178.65/status
[ShellyEM_status.txt](https://github.com/evcc-io/evcc/files/10672557/ShellyEM_status.txt)

Solution could be a shelly-em template like shelly-3em without the values for "Current L1...L3", "Voltage L1...L3" and "Power L1...L3"?

### Steps to reproduce

1. create loadpoint using Shelly EM as meter
2. use shelly-1pm template -> get wrong/no energy value
3. use shelly-3em template -> get correct energy value, but of course errors for missing values

### Configuration details

```yaml
uri: 192.168.178.70:7070
interval: 10s

# log settings
log: error
levels:
  core: debug
  lp-1: debug
  lp-2: debug

# meter definitions
meters:
- name: grid
  type: custom
  power:
    source: modbus
    uri: 192.168.178.61:502
    id: 1
    register:
      address: 40073
      type: holding
      decode: int32s
- name: pv
  type: custom
  power:
    source: modbus
    uri: 192.168.178.61:502
    id: 1
    register:
      address: 40067
      type: holding
      decode: int32s
- name: battery
  type: custom
  power:
    source: modbus
    uri: 192.168.178.61:502
    id: 1
    register:
      address: 40069
      type: holding
      decode: int32s
    scale: -1
  soc:
    source: modbus
    uri: 192.168.178.61:502
    id: 1
    register:
      address: 40082
      type: holding
      decode: uint16
- name: wallbox
  type: sma
  uri: 192.168.178.63
- name: shelly_m_heizstab
  type: template
  template: shelly-1pm
  usage: charge
  host: 192.168.178.65
#  user: ***
#  password: ***
#  channel: 0

# charger definitions
chargers:
- name: wallbe
  type: wallbe
  uri: 192.168.178.62:502
- name: shelly_s_heizstab
  type: template
  template: shelly
  host: 192.168.178.64
  standbypower: -5

# vehicle definitions
vehicles:
- name: hyundai
  type: hyundai
  title: Ioniq
  capacity: 38
  phases: 1
  user: ***
  password: ***
  onIdentify:
    mode: pv

- name: renault
  type: renault
  title: Twingo
  user: ***
  password: ***
  capacity: 22
  onIdentify:
    mode: pv
    targetSoC: 80

- name: heizen
  type: custom
  title: Heizstab
  phases: 1
  soc:
    source: js
    script: 50;

# site definition
site:
- title: Home
  meters:
    grid: grid
    pvs:
    - pv
    batteries:
    - battery
  prioritySoC: 30

# loadpoint definition
loadpoints:
- title: Garage
  charger: wallbe
  meter: wallbox
  vehicles:
  - hyundai
  - renault
  mode: pv
  resetOnDisconnect: true
  soc:
    poll:
      mode: connected
      interval: 60m
    estimate: true
  phases: 1
  enable:
    delay: 1m
    threshold: 0
  disable:
    delay: 5m
    threshold: 0
  guardduration: 5m
  mincurrent: 6
  maxcurrent: 16
- title: Heizstab
  charger: shelly_s_heizstab
  meter: shelly_m_heizstab
  vehicles: heizen
  mode: pv
  resetOnDisconnect: true
  phases: 1
  mincurrent: 13
  maxcurrent: 14
  enable:
    threshold: 0
    delay: 30s
  disable:
    delay: 15s
    threshold: 0

# tariff definition
tariffs:
  currency: EUR
  grid:
    type: fixed
    price: 0.2923
  feedin:
    type: fixed
    price: 0.07
```

### Log details

```text
[main  ] INFO 2023/02/07 08:25:37 evcc 0.112.4
[main  ] INFO 2023/02/07 08:25:37 using config file: /etc/evcc.yaml
[main  ] WARN 2023/02/07 08:25:37 `uri` is deprecated and will be ignored. Use `network` instead.
[main  ] INFO 2023/02/07 08:25:37 starting ui and api at :7070
[db    ] INFO 2023/02/07 08:25:37 using sqlite database: /root/.evcc/evcc.db
[lp-1  ] WARN 2023/02/07 08:25:41 poll mode '{connected 1h0m0s}' may deplete your battery or lead to API misuse. USE AT YOUR OWN RISK.
[lp-1  ] WARN 2023/02/07 08:25:41 vehicles option is deprecated
[lp-2  ] WARN 2023/02/07 08:25:41 vehicles option is deprecated
[site  ] WARN 2023/02/07 08:25:41 vehicle 'Heizstab' does not support automatic detection
[site  ] INFO 2023/02/07 08:25:41 site config:
[site  ] INFO 2023/02/07 08:25:41   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2023/02/07 08:25:41     grid:      power ✓ energy ✗ currents ✗
[site  ] INFO 2023/02/07 08:25:41     pv 1:      power ✓ energy ✗ currents ✗
[site  ] INFO 2023/02/07 08:25:41     battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✗
[site  ] INFO 2023/02/07 08:25:41   vehicles:
[site  ] INFO 2023/02/07 08:25:41     vehicle 1: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
[site  ] INFO 2023/02/07 08:25:41     vehicle 2: range ✓ finish ✓ status ✓ climate ✗ wakeup ✓
[site  ] INFO 2023/02/07 08:25:41     vehicle 3: range ✓ finish ✓ status ✓ climate ✓ wakeup ✓
[lp-1  ] INFO 2023/02/07 08:25:41 loadpoint 1:
[lp-1  ] INFO 2023/02/07 08:25:41   mode:        pv
[lp-1  ] INFO 2023/02/07 08:25:41   charger:     power ✗ energy ✗ currents ✗ phases ✗ wakeup ✗
[lp-1  ] INFO 2023/02/07 08:25:41   meters:      charge ✓
[lp-1  ] INFO 2023/02/07 08:25:41     charge:    power ✓ energy ✓ currents ✓
[lp-2  ] INFO 2023/02/07 08:25:41 loadpoint 2:
[lp-2  ] INFO 2023/02/07 08:25:41   mode:        pv
[lp-2  ] INFO 2023/02/07 08:25:41   charger:     power ✓ energy ✓ currents ✗ phases ✗ wakeup ✗
[lp-2  ] INFO 2023/02/07 08:25:41   meters:      charge ✓
[lp-2  ] INFO 2023/02/07 08:25:41     charge:    power ✓ energy ✓ currents ✗
[lp-1  ] DEBUG 2023/02/07 08:25:41 phase timer inactive
[lp-1  ] DEBUG 2023/02/07 08:25:41 pv timer inactive
[lp-2  ] DEBUG 2023/02/07 08:25:41 phase timer inactive
[lp-2  ] DEBUG 2023/02/07 08:25:41 pv timer inactive
[site  ] DEBUG 2023/02/07 08:25:41 ----
[lp-1  ] DEBUG 2023/02/07 08:25:41 charge power: 6W
[main  ] FATAL 2023/02/07 08:25:41 could not open port- check that evcc is not already running (listen tcp :7070: bind: address already in use)
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.112.4

SwaY88LA, 2023-02-07:
This works for me as solution:

```
meters:
- name: shelly_m_heizstab
  type: custom
  power:  
    source: http
    uri: http://xxx.xxx.xxx.xxx/emeter/0
    jq: .power
  energy:  
    source: http
    uri: http://xxx.xxx.xxx.xxx/emeter/0
    jq: .total
    scale: 0.001
```

andig, 2023-02-07:
/cc @thierolm

thierolm, 2023-02-08:
@SwaY88LA in general the template shelly-1pm was named in that way, bec. it was NOT created for a shelly EM ...

But could U please provide me a Trace log (`evcc -l trace`), which shows the JSON response of your shelly EM device, so that I can check what we have to add to make it working?

thierolm, 2023-02-08:
Oh, I see U added it already, so please stay tuned .. :-)

thierolm, 2023-02-08:
Could U pls use usage *pv* instead of *charge* in your meter definition:
```yaml
- name: shelly_m_heizstab
  type: template
  template: shelly-1pm
  usage: pv
  host: 192.168.178.65
```

SwaY88LA, 2023-02-08:
Hi thierolm

I used the template shelly-1pm, because it's listed for Shelly EM in the documentation:

[image]

I used `usage: heizstab` at first, didn't make any difference if I use heizstab, charge or nothing at all

Now I use this which works fine for me and shows correct values for power and energy

```
meters:
- name: shelly_m_heizstab
  type: custom
  power:  
    source: http
    uri: http://xxx.xxx.xxx.xxx/emeter/0
    jq: .power
  energy:  
    source: http
    uri: http://xxx.xxx.xxx.xxx/emeter/0
    jq: .total
    scale: 0.001
```

thierolm, 2023-02-08:
Nice! But give me a favor and test the usage: pv setting with your EM.

A template configuration is better and more sequre, compared to a custom configuration.

An I would like to know, whether the template works for EM or not, bec I do not own such a device ...

andig, 2023-02-08:
@thierolm the template has a usage parameter, but that will only act as a filter. It does not change how the Shelly is actually configured:

```yaml
template: shelly-1pm
products:
  - brand: Shelly
    description:
      generic: 1PM, EM
group: switchsockets
params:
  - name: usage
    choice: ["pv"]
  - name: host
    default: 192.0.2.2
  - name: user
  - name: password
    mask: true
  - name: channel
    default: 0
render: |
  type: shelly
  uri: http://{{ .host }}  # shelly device ip address (local)
  {{ if ne .user "" }}user: {{ .user }}{{ end }}
  {{ if ne .password "" }}password: {{ .password }}{{ end }}
  channel: {{ .channel }}  # shelly device relay channel
```

andig, 2023-02-08:
The 3em is different. Seems the 1em should be like a 3em but only 1p?

SwaY88LA, 2023-02-09:
Here's the result for template 1pm and usage pv:

[image]

No difference to any other usage. But I have no idea where the 0.4kWh come from, I can't find a value like that in any of my Shelly devices - and I only have the Pro 1PM and the EM.

Yes, 1em (or better 2em, because it has 2 channels) is more similiar to 3em. But it doesn't deliver current, only voltage, power and energy. 
And if I saw it correctly your 3em template adds all totals together and outputs only one combined value for output. It think with the simple EM the channel should be specified and the values total should not be added together.

That  works, if you want to use both channels:

```
- name: shelly_m_heizstab
  type: custom
  power:  
    source: http
    uri: http://xxx.xxx.xxx.xxx/emeter/0
    jq: .power
  energy:  
    source: http
    uri: http://xxx.xxx.xxx.xxx/emeter/0
    jq: .total
    scale: 0.001
- name: shelly_em_2
  type: custom
  power:  
    source: http
    uri: http://xxx.xxx.xxx.xxx/emeter/1
    jq: .power
  energy:  
    source: http
    uri: http://xxx.xxx.xxx.xxx/emeter/1
    jq: .total
    scale: 0.001
```

[image]

thierolm, 2023-02-09:
Could U provide a trace log, to have the jsons for the above tests? I will check afterwards ...

VolkerK62, 2023-02-11:
This ist the output of my Shelly EM (only one channel in use)
```
[http  ] TRACE 2023/02/11 05:19:22 GET http://192.168.178.21/emeter/0
[http  ] TRACE 2023/02/11 05:19:22 {"power":111.88,"reactive":100.42,"pf":0.74,"voltage":235.27,"is_valid":true,"total":401357.6,"total_returned":652286.4}

```

thierolm, 2023-02-11:
Thx Volker, die Customer Config Ausgabe nutzt mir nur nichts, da ich das komplette Status JSON benötige.
Kannst du mit deinem EM mal mit dem 1PM Meter Template einen Trace fahren?
Folgende config:
```yaml
- name: shelly_em_test
  type: template
  template: shelly-1pm
  usage: pv
  host: 192.168.178.21
```
```

VolkerK62, 2023-02-11:
ah, das ist ei Unterschied ;-)
Keine Ahnung, warum die Abfrage 2x kommt

```
test
----
[shelly] TRACE 2023/02/11 11:16:06 GET http://192.168.178.21/status
[shelly] TRACE 2023/02/11 11:16:06 {"wifi_sta":{"connected":true,"ssid":"KAS1","ip":"192.168.178.21","rssi":-55},"cloud":{"enabled":false,"connected":false},"mqtt":{"connected":false},"time":"11:16","unixtime":1676110566,"serial":21580,"has_update":false,"mac":"C45BBE77DE81","cfg_changed_cnt":0,"actions_stats":{"skipped":0},"relays":[{"ison":false,"has_timer":false,"timer_started":0,"timer_duration":0,"timer_remaining":0,"overpower":false,"is_valid":true,"source":"input"}],"emeters":[{"power":-620.34,"reactive":714.48,"pf":-0.66,"voltage":235.68,"is_valid":true,"total":401472.9,"total_returned":653673.7},{"power":0.00,"reactive":0.00,"pf":0.00,"voltage":235.68,"is_valid":true,"total":173411.3,"total_returned":294.2}],"update":{"status":"idle","has_update":false,"new_version":"20221027-105518/v1.12.1-ga9117d3","old_version":"20221027-105518/v1.12.1-ga9117d3"},"ram_total":51072,"ram_free":35844,"fs_size":233681,"fs_free":156373,"uptime":2226139}
[shelly] TRACE 2023/02/11 11:16:06 GET http://192.168.178.21/status
[shelly] TRACE 2023/02/11 11:16:06 {"wifi_sta":{"connected":true,"ssid":"KAS1","ip":"192.168.178.21","rssi":-55},"cloud":{"enabled":false,"connected":false},"mqtt":{"connected":false},"time":"11:16","unixtime":1676110566,"serial":21580,"has_update":false,"mac":"C45BBE77DE81","cfg_changed_cnt":0,"actions_stats":{"skipped":0},"relays":[{"ison":false,"has_timer":false,"timer_started":0,"timer_duration":0,"timer_remaining":0,"overpower":false,"is_valid":true,"source":"input"}],"emeters":[{"power":-620.34,"reactive":714.48,"pf":-0.66,"voltage":235.68,"is_valid":true,"total":401472.9,"total_returned":653673.7},{"power":0.00,"reactive":0.00,"pf":0.00,"voltage":235.68,"is_valid":true,"total":173411.3,"total_returned":294.2}],"update":{"status":"idle","has_update":false,"new_version":"20221027-105518/v1.12.1-ga9117d3","old_version":"20221027-105518/v1.12.1-ga9117d3"},"ram_total":51072,"ram_free":35660,"fs_size":233681,"fs_free":156373,"uptime":2226140}
Power:  -620W
Energy: 6.7kWh
```

thierolm, 2023-02-11:
Also Power stimmt , ist denn der Energy-Wert korrekt?

VolkerK62, 2023-02-11:
Kann ich dir leider nicht sagen, weil ich darauf keinen Fokus habe. Ich messe damit lediglich die Lade-/Entladeleistung meiner Hausbatterie.

thierolm, 2023-02-11:
Alles klar, ich glaube ich hab' den Fehler gefunden:
Der Shelly 1PM liefert Watt-Minuten und der EM Wattstunden ... ich teile immer durch 60 ... ich baue asap einen Fix für die Energy ...
[image]

[image]

VolkerK62, 2023-02-11:
Ich weiß nicht, wann in 2022 ich den Shelly zu dem Zweck in Gebrauch genommen habe, aber wenn ich in das (dürftige) Management meiner Hausbatterie schaue, dann habe ich im letzten Jahr 1165 kWh aus der Batterie bezogen. Das müsste ja dann "total" sein (wenn geladen wird ist die Anzeige negativ, wenn entladen wird positiv). Insofern kann der Wert oben eigentlich nicht stimmen. Oder gibt es hier evtl eine Verfälschung durch Wirk-/Schein-/Blindleistung?

thierolm, 2023-02-11:
@VolkerK62 : 
Nein, der falsche Wert liegt daran, dass wie beom 1PM der Watt-MINUTEN liefert, der Watt-STUNDEN Wert vom EM durch 60 dividiert wird ...

Die Shelly Jungs sind da leider nicht so auf EINEN Standard aus ...  

Kannst du mir noch einen Gefallen tun und die JSON Ausgabe von `http://192.168.178.21/shelly` von deinem EM zukommen lassen?
Wichtig wäre der `type`...
Bei meinem 1 PM sieht das wie folgt aus:
```json
{"type":"SHSW-PM","mac":"84CCA8B10D73","auth":true,"fw":"20220209-094317/v1.11.8-g8c7bb8d","discoverable":true,"longid":1,"num_outputs":1,"num_meters":1}
```

andig, 2023-02-11:
@thierolm der EM scheint ja supereinfach zu sein. Sollen wir den nicht in ein Template mit JQ packen anstatt da Go Code zu schreiben?

thierolm, 2023-02-11:
Klönnen wir, aber wenn ich den `type` noch in der connection mitgebe, kann ich bei der Berechnung der TotalEnergy einfach eine Fallunterscheidung haben.
Dann hätten wir alles aus einem Guss ...

andig, 2023-02-11:
Ich fänds einfacher und klarer- falls möglich- alle einfachen Fälle über Templates zu machen. Viel weniger Code und leichter zu verstehen. Überblicke den Gerätezoo aber nicht genug um zu beurteilen, ob das möglich ist. Caching wäre ein weiterer Nachteil.

andig, 2023-02-11:
Test cases wäre gut :)

thierolm, 2023-02-11:
> Test cases wäre gut :)

jepp, ich bin schon am Sammeln ...

VolkerK62, 2023-02-11:
das ist einfach
`{"type":"SHEM","mac":"C45BBE77DE81","auth":false,"fw":"20221027-105518/v1.12.1-ga9117d3","discoverable":true,"longid":1,"num_outputs":1,"num_meters":0,"num_emeters":2,"report_period":1}`

PR #6204 (2023-02-11): Shelly: Correct Total Energy 4 gen1 EM devices
Fixes #6113

including new Shelly JSON tests.

PR tested succesful with

1. Shelly 1
2. Shelly 1PM
3. Shelly Plus 1PM
closed 2023-02-11 by andig