evcc-io/evcc #6115: Preisberechnung und Anzeige Batteriekapazität

StevieC121176, 2023-02-07:
### Describe the bug

bezieht sich auf : #6048 

Bei der Preisanzeige beim Hausverbrauch wird mir 8ct angezeigt, obwohl Strom ins Netz gespeist wird. Sobald eingespeist wird sollte der Hausverbrauch doch 0ct sein oder?

Wenn ich mit der Maus über den aktuellen SOC der Batterie gehe, wird mir nicht die Kapazität sondern ein Text angezeigt. 

sh. Foto

[image]

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: info
levels:
  cache: error

# unique installation id
plant: *****

interval: 10s # control cycle interval

sponsortoken: *****

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: true

meters:
- type: template
  template: kostal-plenticore 
  id: 71  
  host: 192.168.178.117  
  port: 1502  
  usage: pv  
  modbus: tcpip  
  name: pv1
- type: template
  template: kostal-plenticore 
  id: 71  
  host: 192.168.178.117  
  port: 1502  
  usage: battery  
  modbus: tcpip 
  capacity: 12.8 
  name: battery2
- type: template
  template: kostal-ksem 
  id: 71  
  host: 192.168.178.116  
  port: 502  
  usage: grid  
  modbus: tcpip  
  name: grid3

chargers:
- type: template
  template: daheimladen 
  token: *****
  stationid: *****  
 #host: 192.168.178.74  
  name: wallbox9
- type: template
  template: homematic 
  host: 192.168.178.68  
  port: 2010  
  device: '****'  
  standbypower: 15  
  user: *****
  password: *****
  meterchannel: 6  
  switchchannel: 3  
  name: wallbox10

vehicles:
- type: template
  template: enyaq
  title: Skoda Enyaq 80iV
  user: *****
  password: *****
  vin: *****
  capacity: 82
  phases: 3
  cache: 15
  mode: off
  minSoC: 30
  targetSoC: 99
  minCurrent: 6
  maxCurrent: 16
  identifiers:
  - ****
  name: ev6
  icon: car
- type: template
  template: skoda 
  title: Skoda Citigo iV  
  user: *****
  password: *****
  vin: *****
  capacity: 38  
  phases: 2  
  cache: 15  
  mode: off  
  minSoC: 30
  targetSoC: 99  
  minCurrent: 6  
  maxCurrent: 16  
  identifiers:
  - **** 
  name: ev7
  icon: car
- type: template
  template: offline 
  title: E-Bike
  capacity: 0.75
  phases: 1  
  mode: off  
  minCurrent: 0.2
  maxCurrent: 0.2
  name: ev8
  icon: bike

loadpoints:
- title: Garage
  charger: wallbox9
  mode: off
  phases: 3
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: false
  soc:
    poll:
      mode: charging
      interval: 60m
    estimate: true
- title: Fahrrad
  charger: wallbox10
  vehicle: ev8
  mode: off
  phases: 1
  mincurrent: 6
  maxcurrent: 16
  resetOnDisconnect: false

site:
  title: Mein Zuhause
  meters:
    grid: grid3
    pvs: pv1
    batteries:
    - battery2
  # capacity: 12 
  # residualPower: 100
#  bufferSoC: 90
#  prioritySoC: 80
#  bufferSoC: 100
#  prioritySoC: 100

influx:
  url: http://192.168.178.80:8086
  database: evcc
  user: *****
  password: *****

mqtt:
  broker: 192.168.178.77:1883
  topic: evcc # root topic for publishing, set empty to disable publishing
  # clientid: foo
  user: *****
  password: *****

tariffs:
  currency: EUR # (default EUR)
  grid:
    # static grid price
    type: fixed
    price: 0.234 # [currency]/kWh

  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.082 # [currency]/kWh

  planner:
    type: grünstromindex
    zip: "*****"
```

### Log details

```text
.
```

### What type of operating system are you running?

Linux

### Version

0.112.5 (4ff0b9a5)

Robbe64, 2023-02-07:
Die 8,2 ct sollen sicher die 8,2 ct sein die dir durch das Nicht-Einspeisen durch die Lappen geht.

Den Fehlerhaften Tooltip kann ich bestätigen.

StevieC121176, 2023-02-07:
> Die 8,2 ct sollen sicher die 8,2 ct sein die dir durch das Nicht-Einspeisen durch die Lappen geht.
> 
> Den Fehlerhaften Tooltip kann ich bestätigen.

Mir geht doch nichts durch die Lappen. Ich speise ein, dafür gibts die 8,2ct. Das wird auch angezeigt. 
Ich verbrauche Strom ohne Netzbezug, damit kostet er mich 0 ct.
Ich verbrauche Strom mit Netzbezug, dann kostet er mich die 23,4ct - den Sonnenanteil

VolkerK62, 2023-02-07:
>Ich verbrauche Strom ohne Netzbezug, damit kostet er mich 0 ct.

Wenn du den Strom nicht im Haus verbrauchen würdest, würde er eingespeist werden und dafür würde es dann 8ct geben.
Geht schon in Richtung Philosophie ;-)

naltatis, 2023-02-07:
> Die 8,2 ct sollen sicher die 8,2 ct sein die dir durch das Nicht-Einspeisen durch die Lappen geht.

Ja, nach der Logik funktioniert auch die Ersparnisberechnung. Dort betrachten wir den PV-Strom/Batteriestrom nicht als "free" sondern rechnen ihn mit der Entgangenen Einspeisevergütung. Du kannst die Einspeisevergütung in der Konfiguration natürlich auf 0 drehen um den gewünschten Effekt zu bekommen. Die wird für nichts anderes verwendet.

PR #6117 (2023-02-07): Add missing battery tooltip
fixes #6115

Die Übersetzung ist beim Merge durchgerutscht.
closed 2023-02-07 by andig