evcc-io/evcc #6131: telemetry upload error 403

VolkerK62, 2023-02-08:
### Describe the bug

Wenn geladen wird, bekomme ich einen telemetry upload error angezeigt. 
Keine negativen Auswirkungen spürbar.

### Steps to reproduce

.

### Configuration details

```yaml
.
```

### Log details

```text
Feb 08 09:38:39 raspberrypi evcc[31588]: [site  ] DEBUG 2023/02/08 09:38:39 telemetry: charge: Δ656/780Wh @ 2500W
Feb 08 09:38:39 raspberrypi evcc[31588]: [site  ] ERROR 2023/02/08 09:38:39 telemetry: upload failed: unexpected status: 403
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.112.4 (7faa8bb8)

andig, 2023-02-08:
Temporär oder permanent?

VolkerK62, 2023-02-08:
alle 10 Sek (interval). Ich habe telemetry jetzt übers UI deaktiviert.

apunkt, 2023-02-08:
Hatte ich gestern für ein paar Stunden, dann wars wieder ok.

emcepe, 2023-02-08:
Hatte ich heute morgen für längere Zeit. Hab evcc dann neu gestartet, danach war's weg.

VolkerK62, 2023-02-10:
läuft jetzt wieder ohne Fehler. Einfach so. Kein update, keine Änderung.

closed 2023-02-10 by VolkerK62

teslajoshi, 2023-02-10:
Gleiches Problem bei mir. Alle 10 Sekunden, nur wenn das Auto eingesteckt ist.

teslajoshi, 2023-02-10:
> Gleiches Problem bei mir. Alle 10 Sekunden, nur wenn das Auto eingesteckt ist.

Update: Nach Neustart wieder ok!

weindler, 2023-03-27:
Hatte heute auch den selben Fehler

telemtry: upload failed. Post\"https://api.evcc.io/v1/charge\":context deadline exceeded

stritti, 2023-06-20:
Kommt bei mir heute auch alle paar Sekunden, @andig .

CarstenPV, 2023-06-23:
Ist bei mir auch immer öfter dabei. 
Ist es eigentlich nicht so, dass der telemetry Eintrag unnötig ist (veraltet)?
Kann man diesen dann nicht löschen?
Wird das nicht durch den Sponsor Token ersetzt?

andig, 2023-06-23:
Es braucht bitte ein trace log für `site` (evtl. auch `http`). Ich würde gerne die Serverantwort für den 403 sehen. Wenn wir die verstanden haben können wir auch das Logging dazu verstecken.

CarstenPV, 2023-06-24:
Sorry, wenn ich nerve, ihr macht das halt jeden Tag.
Was ich noch weiß: in der evcc.yaml ist site: log drin.
Wie ist jetzt der Befehl dazu?
sudo journalctl -u evcc -S "2023-06-24 07:00:00" -U "2023-06-24 09:22:00" -q > ~/evcc.log für heute morgen
[evcc.log](https://github.com/evcc-io/evcc/files/11855890/evcc.log)
[trace.meter_log.txt](https://github.com/evcc-io/evcc/files/11855892/trace.meter_log.txt)

CarstenPV, 2023-06-24:
Dazu ist gestern Abend noch jemand in die Zoe hinten drauf gefahren, sieht nach nichts aus, aber man weiß ja nie.

andig, 2023-06-24:
Entweder:

    log: trace

oder

```
levels:
  site: trace
```

CarstenPV, 2023-06-24:
Okay dann leider nächste Mal, weil durch einen Neustart funktioniert alles ohne Fehle Meldung. 95 % jetzt, versuche nochmal abzukoppeln und neu

stritti, 2023-07-03:
Bei mir kommt immer folgendes im log:

```sh
Jul 03 17:07:41 smarthome-pi evcc[21343]: [site  ] DEBUG 2023/07/03 17:07:41 static grid tariff
Jul 03 17:07:41 smarthome-pi evcc[21343]: [site  ] DEBUG 2023/07/03 17:07:41 pv power: 3121W
Jul 03 17:07:42 smarthome-pi evcc[21343]: [site  ] DEBUG 2023/07/03 17:07:42 battery soc: 89%
Jul 03 17:07:42 smarthome-pi evcc[21343]: [site  ] DEBUG 2023/07/03 17:07:42 battery power: 1911W
Jul 03 17:07:42 smarthome-pi evcc[21343]: [site  ] DEBUG 2023/07/03 17:07:42 grid power: 150W
Jul 03 17:07:42 smarthome-pi evcc[21343]: [site  ] DEBUG 2023/07/03 17:07:42 site power: 2161W
Jul 03 17:07:42 smarthome-pi evcc[21343]: [site  ] DEBUG 2023/07/03 17:07:42 telemetry: charge: Δ632/639Wh @ 4107W
Jul 03 17:07:42 smarthome-pi evcc[21343]: [site  ] TRACE 2023/07/03 17:07:42 POST https://api.evcc.io/v1/charge
Jul 03 17:07:43 smarthome-pi evcc[21343]: [site  ] TRACE 2023/07/03 17:07:43 {"instanceId":"33aac90af7a06fbb1afb8eeb9d57c0798931c2c2433c8943a6a7898f9018aaa8","chargePower":4107.4,"greenPower":3988.428779000127,"chargeEnergy":0.638986454133721,"greenEnergy":0.6315438040096244}
Jul 03 17:07:43 smarthome-pi evcc[21343]: [site  ] ERROR 2023/07/03 17:07:43 telemetry: upload failed: unexpected status: 403 (Forbidden)
Jul 03 17:07:51 smarthome-pi evcc[21343]: [site  ] DEBUG 2023/07/03 17:07:51 ----
Jul 03 17:07:51 smarthome-pi evcc[21343]: [site  ] DEBUG 2023/07/03 17:07:51 static grid tariff
Jul 03 17:07:51 smarthome-pi evcc[21343]: [site  ] DEBUG 2023/07/03 17:07:51 pv power: 2500W
Jul 03 17:07:52 smarthome-pi evcc[21343]: [site  ] DEBUG 2023/07/03 17:07:52 battery soc: 89%
Jul 03 17:07:52 smarthome-pi evcc[21343]: [site  ] DEBUG 2023/07/03 17:07:52 battery power: 2883W
Jul 03 17:07:52 smarthome-pi evcc[21343]: [site  ] DEBUG 2023/07/03 17:07:52 grid power: 25W
Jul 03 17:07:52 smarthome-pi evcc[21343]: [site  ] DEBUG 2023/07/03 17:07:52 site power: 3008W
Jul 03 17:07:53 smarthome-pi evcc[21343]: [site  ] DEBUG 2023/07/03 17:07:53 telemetry: charge: Δ644/651Wh @ 4115W
Jul 03 17:07:53 smarthome-pi evcc[21343]: [site  ] TRACE 2023/07/03 17:07:53 POST https://api.evcc.io/v1/charge
Jul 03 17:07:53 smarthome-pi evcc[21343]: [site  ] TRACE 2023/07/03 17:07:53 {"instanceId":"33aac90af7a06fbb1afb8eeb9d57c0798931c2c2433c8943a6a7898f9018aaa8","chargePower":4115.1,"greenPower":4096.379780020223,"chargeEnergy":0.6510922154944131,"greenEnergy":0.6435944944083241}
Jul 03 17:07:53 smarthome-pi evcc[21343]: [site  ] ERROR 2023/07/03 17:07:53 telemetry: upload failed: unexpected status: 403 (Forbidden)
```