evcc-io/evcc #6138: Add usage PV to SDM meters

premultiply, 2023-02-08:
@VolkerK62 usage gibt es bei den SDMs aktuell nicht.
Müsste tatsächliche für den Fall PV aber rein.

_Originally posted by @premultiply in https://github.com/evcc-io/evcc/discussions/6070#discussioncomment-4904558_

andig, 2023-02-08:
Ändert der was? Müssen wir das Vorzeichen drehen?

andig, 2023-02-08:
Der SDM hat überhaupt keine Usage, kann also überall eingesetzt werden.

premultiply, 2023-02-08:
Das ist richtig, funktioniert aber für den Anwendungsfall PV und Batterie nicht, da dann die Vorzeichen aus Sicht von evcc falsch rum sind.
Wir arbeiten ja bei PV- und Batterieexport mit positiven Vorzeichen.
Ausserdem muss dabei für die Energie das Export statt des Import-Registers verwendet werden.

andig, 2023-02-09:
Bzgl. Export hast Du natürlich recht- wobei wir die bei PV ja nicht nutzen. Das heisst aber im Klartext, dass wir bei allen regulären Zählern PV eine Sonderbehandlung für `Power` geben müssen. Betrifft das noch weitere Register? Ströme???

premultiply, 2023-02-09:
Ja, die Phasenströme bzw. Phasenleistungen auch.

Die "offizielle" Einbaurichtung für Zählpunkte ist immer am Netzanschluss orientiert.
Energiefluss aus Richtung Netz: positiv, Import;
Energiefluss in Richtung Netz: negativ, Export.

Bei allen internen Messpunkten von WR ist das allerdings (wie bei evcc) andersrum, da hier nur die "überwiegende" Energierichtung betrachtet wird bzw. stärker die Erwartungshaltung des Anwenders bedient wird.

closed 2023-02-16 by andig