evcc-io/evcc #6162: Nightly: systemd update error

andig, 2023-02-09:
This is the error on todays update:
```
Vorbereitung zum Entpacken von .../evcc_0.112.5+1675930264_armhf.deb ...
Warning: evcc's home directory is incorrect (116)
but can't be changed because another process (31452) is using it.
Stop offending process(es), then restart installation
dpkg: Fehler beim Bearbeiten des Archivs /var/cache/apt/archives/evcc_0.112.5+1675930264_armhf.deb (--unpack):
 ?neues evcc-Skript des Paketes pre-installation?-Unterprozess gab den Fehlerwert 1 zur?ck
Fehler traten auf beim Bearbeiten von:
 /var/cache/apt/archives/evcc_0.112.5+1675930264_armhf.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)
```
process (31452) was the previous EVCC running process.

_Originally posted by @Chris591 in https://github.com/evcc-io/evcc/issues/4901#issuecomment-1424061757_

pauxus, 2023-02-09:
I still thin we are dealing with corner cases here (earlier created folder, but with root), but we can handle them better.

We can stop the service earlier (i.e. always manually with restart flag). PR on the way.

andig, 2023-02-09:
> There is another issue caused by this, which needs to be addressed: https://github.com/evcc-io/evcc/discussions/6140#discussioncomment-4906057

> /dev/tty... devices can not be accessed by the evcc user.

commit 31096506 (2023-02-09): chore: fix issues with service user change (#6164)

* Fix #6162: Always stop before preinstall

Also: if creating a new user, chown the home directory just to be on the safe side (if /var/lib/evcc already exists but belongs to someone else)

* Add the evcc user to the dialout group

This allows evcc to access /dev/tty.

Fix #6140
closed 2023-02-09 by andig