evcc-io/evcc #6178: UI hidden feature max power falsch

VolkerK62, 2023-02-10:
Bei der Anzeige der Leistung wird die aktuelle Phasen-Einstellung nicht beachtet.
Einstellung ist 1p.
MinCurrent 6A = 1,4 kW ist richtig
MaxCurrent 20A = 13,8 kW ist falsch, weil 3p

[image]

andig, 2023-02-10:
@naltatis ein API dafür? Sonst Leistungen weg lassen?

TTXaser2, 2023-03-08:
> Bei der Anzeige der Leistung wird die aktuelle Phasen-Einstellung nicht beachtet. Einstellung ist 1p. MinCurrent 6A = 1,4 kW ist richtig MaxCurrent 20A = 13,8 kW ist falsch, weil 3p
> 
> [image]

@VolkerK62  Kurze Nachfrage: Wie kommst Du auf den Dialog?

VolkerK62, 2023-03-08:
Gibt es nur in Nightly-Versionen. Lässt sich einschalten über Einstellungen -> experimentelle UI-Feature

VolkerK62, 2023-03-09:
Sorry, habe gerade erst die Mail entdeckt.

Das wird mit im Browser bei /api/state angezeigt

<html><body>

phasesConfigured | null
-- | --

</body>
</html>

andig, 2023-03-09:
@VolkerK62 was sagen denn die anderen Werte bei Dir:

	phasesConfigured = "phasesConfigured" // configured phases (1/3, 0 for auto on 1p3p chargers, nil for plain chargers)
	phasesEnabled    = "phasesEnabled"    // enabled phases (1/3)
	phasesActive     = "phasesActive"     // active phases as used by vehicle (1/2/3)

VolkerK62, 2023-03-09:
<html><body>

phasesActive | 1
-- | --

</body>
</html>
<html><body>

phasesEnabled | 1
-- | --

</body>
</html>

naltatis, 2023-03-09:
> phasesActive die letztlich erwartbar vom Fahrzeug genutzten. Ich denke den Wert müssten wir nehmen.

Ne, hie gehts ja um die potentielle Bandbreite des Ladepunktes. Hießt bei einem einphasigen LP stehen da andere Werte als bei einem dreiphasigen. Bei LPs mit Phasenumschaltung ist der Wert abhängig davon ob auto, 1p, 3p konfiguriert ist. @andig zum Verständnis: heißt das, dass `phasesConfigured` nur bei Ladepunkten mit Phasenumschaltung gesetzt ist? Oder was meint `nil for plain chargers`?

[image]
[image]
[image]

Hier ein Beispiel wie die Werte bei einem 1p/3p charger aussehen.

andig, 2023-03-09:
Genau. Wenn Du die statische Config (unabhängig vom Auto) willst dann wäre das `phasesEnabled`. Ich kann mich grad nicht erinnern, wie der Case "1p3p" charger als 1p anschlossen gelöst war :O

premultiply, 2023-03-09:
> Ich kann mich grad nicht erinnern, wie der Case "1p3p" charger als 1p anschlossen gelöst war :O

Ich meine da muss man am LP `phases: 1` setzen da `phases: 0` (Automatik) immer von drei netzseitig anliegenden Phasen ausgeht.

StevieC121176, 2023-03-10:
Meine Wallbox läuft gerade 3phasig. Die Anzeige ist aber analog @VolkerK62 

[image]
[image]

naltatis, 2023-03-11:
So ganz verstehe ich es leider immer noch nicht. Hier mein Zustand wenn gerade kein Fahrzeug lädt:

```
 "phasesActive": 3,
 "phasesConfigured": 3,
 "phasesEnabled": 0,
```

Die WB kann 1p3p. Das Umschalten via evcc ändert "nur" den `phasesConfigured` Wert (`0/1/3`).

Heißt das, dass der `phasesConfigured` Wert bei 1p/3p WBs relevant ist und bei "normalen" WBs bekomme ich die Information aus `phasesEnabled`? Könnte dieser Wert nicht generell befüllt werden?

andig, 2023-03-12:
@StevieC121176 wo kommt das denn her? Schaltest Du da was per api um? NaN sollte es nicht sein!

> phasesConfigured = NaN

andig, 2023-03-12:
@naltatis @premultiply  die Phasenbenennnung erscheint inkonsistent. Das müssen wir nochmal grundsätzlich angehen.

StevieC121176, 2023-03-12:
> @StevieC121176 wo kommt das denn her? Schaltest Du da was per api um? NaN sollte es nicht sein! Kann Dein Charger 1p3p?
> 
> > phasesConfigured = NaN

Nein, ich nutze mqtt nur lesend (zum verhindern der Batterieentladung während der Fahrzeugladung), auch ansonsten kein Schalten per API. Ich weiss nicht ob es daran liegen kann, ich nutze den evcc modbusproxy um meinen charger im iobroker zu visualisieren.
Mein Charger (Daheimladen) kann kein 1p3p, daher nutze ich das manuell per Lasttrennschalter

TheLightingFactory, 2023-03-14:
> Gibt es nur in Nightly-Versionen. Lässt sich einschalten über Einstellungen -> experimentelle UI-Feature

I can't get this option visible in the Settings menu, even when using the dark theme using the latest version. Running the Windows version to for testing.

closed 2023-03-15 by andig

StevieC121176, 2023-03-16:
> Bei mir lädt gerade der Citigo meiner Frau:
> 
> ```
> phasesConfigured = NaN
> PhasesEnabled = 3
> phasesActive = 2
> ```
> 
> Anzeige in evcc bei 6A - 1,4 kW und bei 16A - 11 kW
> 
> Wenn kein Fahrzeug angeschlossen ist:
> 
> ```
> phasesConfigured = null
> PhasesEnabled = 3
> phasesActive = 3
> ```
> 
> Anzeige in evcc bei 6A - 1,4 kW und bei 16A - 11 kW
> 
> konfiguriert in der yaml ist `phases: 3` bei diesem Loadpoint. automatisch umschalten 1p3p kann meine Wallbox (Daheimladen) nicht, dafür habe ich einen Lasttrennschalter installiert

gleiches Verhalten auch mit der neuen Version `evcc version 0.114.1 (e7e12a90)`

premultiply, 2023-03-16:
phases: 3 -> phasesEnabled = 3

StevieC121176, 2023-03-16:
ja beim Umschalten auf 1 Phasig wird jetzt auch `PhasesEnabled` = 1 angezeigt.

PR #6829 (2023-03-15): Publish phasesEnabled
Publish `phasesEnabled` on phase reconfiguration to reflect the enabled phases seen on the charge connector.

Replaces #6814
May fix #6178
PR #9728 (2023-09-06): Improve min/maxcurrent limit UI
- add lower min current options (0.125, 0.25, 0.5). No free form number input, they are messy on mobile. fixes #9595
- existing non-integer values (e.g. set via API) will now show correctly in settings modal. fixes #9595
- fixed phase-based min/max power estimation based on `phasesConfigured` and `phasesActive` fixes #6178

**Note:** 🧪 Feature remains experimental until we replace yaml config of this value with db persistance.

lower min current options
[image]

add existing non-int values
[image]