evcc-io/evcc #6185: UI BatterieSoc immer "grau"

VolkerK62, 2023-02-10:
Der Batterie Soc wird immer grau dargestellt.
Könnte man das so anpassen, das die Darstellung identisch ist mit der Lade-/Entladeleistung.
Also wenn "aktiv" dann nicht grau.

[image]

naltatis, 2023-02-11:
@VolkerK62 Die Farbgebung habe ich mit dem Einführen der zusätzliche Preis/CO2 Daten bewusst geändert. Die Idee ist, dass die "Zusatzwerte" etwas zurückgenommen dargestellt werden (nicht bold und schwächere Farbe). Der Batterie SoC ist jetzt auch teil dieser "Zusatzspalte". Vorher klebte er ja immer mit nem "/" Trenner direkt an der Leistung.

[image]

[image]

Ich glaube, dass es komisch aussieht wenn wir das jetzt nur für den Batterie SoC wieder zurück ändern.

naltatis, 2023-02-11:
Ich hab mal hier die Alternative gemacht. Also dass immer alle Werte einer Zeile die gleiche Farbe haben:

[image]

[image]

Kann ich mir auch vorstellen. Gerade wenn man keine Zusatzwerte aktiviert hat (wie in deinem Batteriefall) ist das runder.

VolkerK62, 2023-02-11:
danke für die Erklärung, wie es dazu kommt.
das untere Beispiel gefällt mir, das ist stimmig.

naltatis, 2023-02-11:
Ok, mit etwas abstand gefällt mir das jetzt auch besser. Der normal/bold Unterschied ist schon ausreichend um die Werte etwas voneinander abzuheben.

closed 2023-02-11 by andig

gvonhof, 2024-11-01:
Hallo, kann man die Farbe des Netzbezugs nicht in rot darstellen, analog der Darstellung in openWB?

Günther

naltatis, 2024-11-01:
> Hallo, kann man die Farbe des Netzbezugs nicht in rot darstellen, analog der Darstellung in openWB?

Könnte man. Wir haben uns aber für Schwarz bzw. Grau (Darkmode) entschieden 😄

PR #6201 (2023-02-11): Adjust color of active extra-value (soc, co2, price)
fixes #6185