evcc-io/evcc #6190: E3DC external source is not displayed in EVCC

JanDragon, 2023-02-10:
### Describe the bug

I have 2 other inverters connected to my E3DC. These are registered in E3DC as an external source. So E3DC can retrieve all data regularly and treats them like built-in inverters. Only the external source is not displayed and used in EVCC. Here only the built-in inverter is used. So I can not use the full capacity of my plant.

### Steps to reproduce

Continue connecting inverters and register them as an external source at E3DC. Now they are not displayed / used in EVCC.

### Configuration details

```yaml
-
```

### Log details

```text
[main  ] INFO 2023/02/10 23:51:41 evcc 0.112.5
[main  ] INFO 2023/02/10 23:51:41 using config file: /etc/evcc.yaml
[main  ] WARN 2023/02/10 23:51:41 `uri` is deprecated and will be ignored. Use `network` instead.
[main  ] INFO 2023/02/10 23:51:41 starting ui and api at :7070
[db    ] INFO 2023/02/10 23:51:42 using sqlite database: /home/pi/.evcc/evcc.db
[site  ] INFO 2023/02/10 23:51:42 site config:
[site  ] INFO 2023/02/10 23:51:42   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2023/02/10 23:51:42     grid:      power ✓ energy ✗ currents ✗
[site  ] INFO 2023/02/10 23:51:42     pv 1:      power ✓ energy ✗ currents ✗
[site  ] INFO 2023/02/10 23:51:42     battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✗
[lp-1  ] INFO 2023/02/10 23:51:42 loadpoint 1:
[lp-1  ] INFO 2023/02/10 23:51:42   mode:        pv
[lp-1  ] INFO 2023/02/10 23:51:42   charger:     power ✓ energy ✓ currents ✓ phases ✗ wakeup ✗
[lp-1  ] INFO 2023/02/10 23:51:42   meters:      charge ✓
[lp-1  ] INFO 2023/02/10 23:51:42     charge:    power ✓ energy ✓ currents ✓
[lp-2  ] INFO 2023/02/10 23:51:42 loadpoint 2:
[lp-2  ] INFO 2023/02/10 23:51:42   mode:        pv
[lp-2  ] INFO 2023/02/10 23:51:42   charger:     power ✓ energy ✓ currents ✓ phases ✗ wakeup ✗
[lp-2  ] INFO 2023/02/10 23:51:42   meters:      charge ✓
[lp-2  ] INFO 2023/02/10 23:51:42     charge:    power ✓ energy ✓ currents ✓
[lp-1  ] DEBUG 2023/02/10 23:51:42 phase timer inactive
[lp-1  ] DEBUG 2023/02/10 23:51:42 pv timer inactive
[lp-2  ] DEBUG 2023/02/10 23:51:42 phase timer inactive
[lp-2  ] DEBUG 2023/02/10 23:51:42 pv timer inactive
[site  ] DEBUG 2023/02/10 23:51:42 ----
[main  ] FATAL 2023/02/10 23:51:42 could not open port- check that evcc is not already running (listen tcp :7070: bind: address already in use)
[server] FATAL 2023/02/10 23:51:42 unlinkat /tmp/evcc: operation not permitted
```

### What type of operating system are you running?

Linux

### Version

evcc version 0.112.5

premultiply, 2023-02-11:
[image]

premultiply, 2023-02-11:
40075 has to be added to pv power in template.

https://github.com/evcc-io/evcc/blob/master/templates/definition/meter/e3dc.yaml

andig, 2023-02-18:
Closed in https://github.com/evcc-io/evcc/commit/16537c4769bedabdb1ae058fcafa3cfc6036553a

closed 2023-02-18 by andig