evcc-io/evcc #6193: http plugin: Schreiben von Werten über GET Request

menkej, 2023-02-11:
**Is your feature request related to a problem? Please describe.**
Ich nutze aktuell das SimpleApi von ioBroker zur Erstellung einer "manuellen Wallbox"/Visualisierung des Wärmepumpe (ohne Steuerung aktuell). Leider ist hier das Schreiben nur über GET möglich. Daher funktioniert das %v von hier https://docs.evcc.io/docs/reference/plugins#http-lesenschreiben nicht. Ein Unterstützter Aufruf hätte folgende Form: `http://ipaddress:8087/set/stateID?value=HIERISTMEINWERT`

Beim Schreiben des Tickets habe ich alternativ noch dies (suboptimal dokumentierte) Möglichkeit im SimpleApi gefunden: `http://datenhaufen:8087/setValueFromBody?stateID1`. Dies wollte ich dann eben testen. In der EVCC Doku steht [hier](https://docs.evcc.io/docs/reference/plugins#http-lesenschreiben) nur (etwas dünn) `body: %v # only applicable for PUT or POST requests`.

```
chargers:
  - name: heizung
    type: custom
    enable: # set charger enabled state (true/false or 0/1)
      source: http
      # uri: http://datenhaufen.fritz.box:8087/toggle/0_userdata.0.StromWärmepumpe.LadenAktivieren
      uri: http://datenhaufen.fritz.box:8087/setValueFromBody?0_userdata.0.StromWärmepumpe.LadenAktivieren
      #jq: .val
      body: %v
      timeout: 30s # error if no update received in 30 seconds
```
Hier bekomme ich direkt Schimpfe von VSCode: `Plain value cannot start with directive indicator character %`

Da müssen wir wohl auch noch mal an die EVCC Doku ran. Mache ich gern, wenn ich's "richtig" weiß. ;-)

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
Nutzen des toggle-Request im SimpleApi funktioniert, ist aber nicht robust. Umweg über mqtt wäre in meinem Fall möglich. Ist aber ein Umweg...

andig, 2023-02-11:
Bitte probieren ;)

andig, 2023-02-11:
> Hier bekomme ich direkt Schimpfe von VSCode: Plain value cannot start with directive indicator character %

Der Parameter heisst immer so wie der Setter, hier also `enable`. Den musst/darfst Du jetzt in Golang Templates verwenden, also etwa

    body: "{{ .enable }}"

Das Gleiche geht mit dem PR dann auch in der URI. Weiterhin stehen Dir alle Templatefunktionen als auch Sprig zur Verfügung. 

PR zur Doku wäre Klasse- da ist wirklich unglaubliche Flexibilität drin.

andig, 2023-02-11:
@menkej sehe Du bist auch Entwickler- hast Du Lust Dich mal bei Slack zu melden? Wir kommen sogar aus der gleichen Ecke ;)

closed 2023-02-11 by andig

menkej, 2023-02-11:
> @menkej sehe Du bist auch Entwickler- hast Du Lust Dich mal bei Slack zu melden? Wir kommen sogar aus der gleichen Ecke ;)

Da bin ich doch längst unterwegs - Stichwort Aiways U5. ;-)