evcc-io/evcc #6198: UI Anzeige beim Beenden des Ladevorgangs

VolkerK62, 2023-02-11:
Es ist nur ein "Schönheitswunsch", Funktionalität ist fehlerfrei.
Disable Timer wird angezeigt. In dem Moment wo er abgelaufen ist, wechselt die Anzeige wieder auf "Ladevorgang aktiv". Dann dauert es 2-3 Zyklen, bis die angezeigte Leistung auf 0 geht und die Anzeige auf "verbunden" wechselt.
Da frage ich mich immer, har er jetzt aufgehört, oder wurde der Timer in letzter Sekunde unterbrochen.
Wäre es möglich in dem Fall "Ladevorgang wird beendet" anzuzeigen.

naltatis, 2023-02-11:
Aktuell liefert uns die UI nicht die relevanten informationen dafür. Der Timer ist abgelaufen, aber der Charger Status passt passt noch nicht bzw. ist träger. @andig hast du ne Idee wie man diesen Zwischenzustand abbilden/signalisieren kann? Ich glaube beim Starten eines Ladevorgangs (enable Timer) gibts das auch.

VolkerK62, 2023-02-11:
Ja, es liegt an der Trägheit. Könnte man das "charger disable" Kommando dafür nutzen? Solange bis chargePower auf 0 ist?

andig, 2023-02-11:
Beim starten ist nicht klar, ob das Auto wirklich Leistung braucht. Beim Stoppen hängen wir an Trägheit von WB und Auto. Selbst wenn wir da einmalig „verbunden“ annehmen würden könnte das beim nächsten Update wieder auf „lädt“ springen für 1,2 Zyklen. Ich seh leider nicht was man da sinnvoll machen könnte.

VolkerK62, 2023-02-11:
OK. Wie gesagt, war eher ein Schönheitswunsch. Evtl fällt es mir gar nicht mehr auf, wenn DaheimLaden lokal als Modbus läuft. Ich vermute, dann ist die Trägheit kleiner.

andig, 2023-02-11:
Da kannste von ausgehen…

closed 2023-02-11 by andig