evcc-io/evcc #6212: Klärung Verhalten von BufferSoc

VolkerK62, 2023-02-12:
Durch mehrere Diskussionen und einen Selbstversuch habe ich folgendes herausgefunden:
Ausgangssituation: Auto ist angesteckt, Modus PV, es wird nicht geladen, BatterySoc > bufferSoc

Solange die Hausbatterie geladen wird, startet die Fahrzeugladung nicht.
Wird die Hausbatterie nicht mehr geladen bzw. entladen (Kipppunkt konnte ich nicht testen), dann startet die Fahrzeugladung.

Soll das so sein?
Wäre es nicht umgekehrt, richtiger?

manny4566, 2023-02-12:
Ich habe mir auch nochmal den Kopf zerbrochen und kam auf die Idee, dass doch evcc gar nicht steuern kann, ob die Batterie entladen werden soll oder nicht. Wie auch?
Der Wechselrichter steuert doch gegen, wenn im Hausnetz entsprechend Verbrach ist. Und der WR kann doch gar nicht wissen, dass dort 5 Backoffen gerade laufen oder eben ein Auto über evcc geladen wird.

Ergo müsste es eine Funktion geben, die es erlaubt zu sagen, dass evcc z.b. über Modbus und einem HoldingRegister dem WR sagt er soll die Batterie nicht entladen.
Z.b. in dem man den EntladeSOC der Batterie z.b. auf 100% stellt und somit weiß der Wechselrichter ich entlade die Batterie jetzt nicht. Nach dem Auto laden wird er runter gestellt z.b. zurück auf 20% und der WR nutzt die Batterie wieder für den Hausverbrauch.

Ich bezweifle, dass das bufferSoc macht oder?
Der Wert liest doch nur den Soc der Batterie aus und startet das laden des Autos oder nicht je nach definierten bufferSoc . Was aber der WR im Hintergrund macht, sprich Batterie ggf. zu entladen, kann evcc stand jetzt nicht beeinflussen.

thebrainkafka, 2023-02-12:
> Durch mehrere Diskussionen und einen Selbstversuch habe ich folgendes herausgefunden: Ausgangssituation: Auto ist angesteckt, Modus PV, es wird nicht geladen, BatterySoc > bufferSoc
>
> Solange die [...]

Erwartet hätte ich auch eine Ladung immer wenn PrioritySoc und BufferSoc erreicht/ überschritten.

VolkerK62, 2023-02-12:
dupe #6213

closed 2023-02-12 by VolkerK62